polarofficial/polar-ble-sdk #181: Unable to read file Polar H10

awilliamsra, 2021-08-23:
**Platform on which you observed the bug:** 
- [ ] Android 
- [X] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [X] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
When trying to read a large recording from the Polar H10 device ( 8+ hrs ) the Polar SDK will stop and throw a wait Timeout error, I can see the data being streamed for about 10 seconds it will then stop and throw the error listed above. When I try and read the file from the device after this a different error will be thrown - response Error with code 303. 

**How to Reproduce**
Perform long recording then try and read recording - it should be noted that I have only seen this once but I was unable to read the file from the device.

**Expected behavior**
File should be read.

**Screenshots and logs**

JOikarinen, 2021-08-24:
@awilliamsra thank you for the report. I will try to reproduce. 303 error is for "air packet lost", which may hint that something is interfering the BLE communication.

Mezah, 2021-09-20:
I am also facing this problem, in my case I tried to create a timer that request recording status and based on the existence of a file I stop read and start a new record.
@awilliamsra Did you manage to find out what was the problem

JOikarinen, 2021-09-27:
@awilliamsra  and @Mezah, I've tried to reproduce the problem by doing long recording on H10 and then reading it out. My Steps:
- start recording with [example app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-ios/iosBleSdkTestApp)
- stop recording with [example app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-ios/iosBleSdkTestApp) ( ~ 24 hours since recording start)
- list recordings
- read out the recording

Observation:
- reading out the 24 hour recording took around 3min
```
2021-09-27 08:32:28.853281+0300 iosBleSdkTestApp[386:18256] Polar SDK log:  [BLE] PS-FTP new request operation
...
2021-09-27 08:35:34.838348+0300 iosBleSdkTestApp[386:18239] exercise data count: 84510 samples:
```
- recording was read out successfully

Is it possible you could share the logs from the moment you are reading the big file from H10?

awilliamsra, 2021-11-05:
@JOikarinen - I don't have any at the moment. But if it appears again I will post them here !

awilliamsra, 2021-11-10:
So I tried to read a large recording over 9+hr's - the read seemed to timeout then all subsequent attempts to list the exercises caused a 303 error. 

[polar_303_log.txt](https://github.com/polarofficial/polar-ble-sdk/files/7511890/polar_303_log.txt)

JOikarinen, 2021-11-22:
@awilliamsra thanks for the good logs and sorry for delayed answer. 

This is what I see in the logs:
- device connected: `20:53:40.889426+0100	 DEVICE CONNECTED:..`
- recoding stop: `20:53:42.549307+0100	Polar SDK log:  MTU send  HEX 02 0F 80`
- record listing completed: `20:53:42.735833+0100	 entry: 2021-08-22 19:53:42 ... `
- record read start: `20:53:42.736356+0100	Polar SDK log:  MTU send  HEX 02 31 00 ... `
- first data packet received `20:53:42.816696+0100	Polar SDK log:  MTU in  HEX 06 ... `
- ... multiple data packets is sent by H10 ...
-  data packet received: `20:53:53.559039+0100	Polar SDK log:  MTU in  HEX 77 05 B2 ... `
- ... more data expected still from H10. Polar SDK is waiting ...
- finally the waiting of the data timeout's: `20:54:23.565434+0100	Polar SDK log:  [BLE][ERROR] PS-FTP request interrupted error: waitTimeout`
- device is connected again: `20:55:26.557061+0100	DEVICE CONNECTED:`
- record listing started: `20:55:27.945332+0100	Polar SDK log:  MTU send  HEX 02 05 ...`
- Polar H10 is answering with the data what seems to be missing part of data for previous record read: `20:55:28.064008+0100	Polar SDK log:  MTU in  HEX 83 04 E3 ...`

The problem you have described seems very obvious in the log. It seems that for some reason H10 just stops sending the data or at least it is not received in SDK. Future more, there is weird behaviour, the missing part of data seems to be received when H10 is connected again. The root cause of the issue still unclear. I will try to reproduce this to get a bit more light on this mysterious issue.

JOikarinen, 2021-11-23:
I retested this case again. I recorded long exercise (>12h). Started the recording and read out the recording with the [example app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-ios/iosBleSdkTestApp). The outcome is that data is successfully read every time I attempt to read it. 

Questions @awilliamsra:
- which [RecordingInterval](https://github.com/polarofficial/polar-ble-sdk/blob/dcad1148853a12e23ad52ed045313cbf45753cf2/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L18) and [SampleType](https://github.com/polarofficial/polar-ble-sdk/blob/dcad1148853a12e23ad52ed045313cbf45753cf2/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L24) parameters you use for [startRecording](https://github.com/polarofficial/polar-ble-sdk/blob/dcad1148853a12e23ad52ed045313cbf45753cf2/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L194)? 
- in your log I see the `Background Task XXX, was created over 30 seconds ago.` which make me wonder the root cause of those. Do you have idea?
- if you have possibility could you try what happens if you try to read the long exercise using the [example application](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-ios/iosBleSdkTestApp). Steps would be something like this:
    - install the example app
    - connect to H10 using either auto connect (i.e. connects automatically to nearest Polar sensor) or with connect button. If using connect button then DeviceId need to define in [source code](https://github.com/polarofficial/polar-ble-sdk/blob/dcad1148853a12e23ad52ed045313cbf45753cf2/examples/example-ios/iosBleSdkTestApp/iosBleSdkTestApp/PolarBleSdkManager.swift#L15). 
    - press list exercises and then read exercise

smyffanon, 2023-04-29:
Hello, this has happened 2 times in the last month.

with the newest version of the software the logcat is happening much faster until I hit the error, now hopefully we can make this issue go away

[polar_read_error.txt](https://github.com/polarofficial/polar-ble-sdk/files/11359005/polar_read_error.txt)
[polar_read_error2.txt](https://github.com/polarofficial/polar-ble-sdk/files/11359006/polar_read_error2.txt)

smyffanon, 2023-04-29:
Occasionally I have this happen and the 2nd or 3rd time the read succeeds.
1 or 2 times a month I can re try reading 30 or 40 times and reading keeps failing

I have 2 polar h10s.

Next time this issue happens I could send you the unit, although I suspect with a few days in the mail, when you get it the read will succeed.

smyffanon, 2023-04-29:
two times this happened in the last month
[logcat.txt](https://github.com/polarofficial/polar-ble-sdk/files/11359046/logcat.txt)
[polar_read_error2.txt](https://github.com/polarofficial/polar-ble-sdk/files/11359047/polar_read_error2.txt)

orestesgaolin, 2023-12-04:
Wondering if this issue is being worked on. We're experiencing the same (or very similar) problem with reading recordings from Verity Sense. Unfortunately I wasn't able to reproduce it on any of my devices, but few of our users encounter this in the wild.