polarofficial/polar-ble-sdk #185: PolarErrors not found

avik-efcom, 2021-09-06:
***Platform your question concerns:*** 
- [ ] Android 
- [X] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [X] Device is not relevant for this question

**Description:**
Hi there,
I try to get some information about PolarErrors but it seems there is no link on documentation for this.
Can you fix it? or send me the PolarErrors list on IOS?
[image]

JOikarinen, 2021-09-10:
Hi @avik-efcom, good point.

Fixed the documentation problem in 62c8af2 and 1585e1de6572d59fe6378830b0764375c912b676.

Are you facing some particular error which you like to get more information?

avik-efcom, 2021-09-12:
I try to understand the error "ServiceNotFound" when I ask from the exercise list, but I find it my problem cause I not ask for all SDK device permission at the start

[https://github.com/polarofficial/polar-ble-sdk/blob/1585e1de6572d59fe6378830b0764375c912b676/examples/example-ios/iosBleSdkTestApp/iosBleSdkTestApp/PolarBleSdkManager.swift#L12](url)

Thanks

closed 2021-09-12 by avik-efcom