polarofficial/polar-ble-sdk #186: Failed with error: ERROR_ALREADY_IN_STATE

sach16795, 2021-09-09:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
I have created a foreground service that will run continuously. In the logs I can see at times if the service process is killed, it tries to restart itself. In the onCreate method, I try to initialize the Polar device and establish the connection. However, once I start to read the accelerometer stream, the following error is shown. Not sure why this would happen. In the logs printed earlier, it shows that the device is ready for the accelerometer stream. Additionally, no documentation was found for the ERROR_ALREADY_IN_STATE issue. Please help.

**How to Reproduce**
1. Create foreground service with start_sticky. 
2. Start accelerometer stream.
3. Wait for the process to be killed.
4. On the restart following issue is encountered. 

**Expected behavior**
Since the connection is established with the device. Accelerometer readings should be possible even on service restart. 

**Screenshots and logs**
22:18:09.085 INFO: Trying to connect to device A17B5529
22:18:09.090 DEBUG: BDScanCallback/commandState state:IDLE action: CLIENT_START_SCAN
22:18:09.091 DEBUG: BDScanCallback/commandState state:IDLE action: EXIT
22:18:09.092 DEBUG: BDScanCallback/commandState state:SCANNING action: ENTRY
22:18:09.092 DEBUG: BDScanCallback/timestamps left: 0
22:18:09.093 DEBUG: BDScanCallback/Scan started -->
22:18:09.113 DEBUG: BDScanCallback/Scan started <--
22:18:09.114 INFO: onStartCommand exited
22:18:09.649 DEBUG: BDScanCallback/commandState state:SCANNING action: CLIENT_REMOVED
22:18:09.650 DEBUG: BDScanCallback/commandState state:SCANNING action: EXIT
22:18:09.650 DEBUG: BDScanCallback/Stop scanning
22:18:09.657 DEBUG: BDScanCallback/commandState state:IDLE action: ENTRY
22:18:09.665 DEBUG: ConnectionHandler/state: CONNECTING action: ENTRY
22:18:09.665 DEBUG: BDScanCallback/commandState state:IDLE action: ADMIN_STOP_SCAN
22:18:09.666 DEBUG: BDScanCallback/commandState state:IDLE action: EXIT
22:18:09.666 DEBUG: BDScanCallback/commandState state:STOPPED action: ENTRY
22:18:09.666 DEBUG: ConnectionHandler/ Session update from: SESSION_CLOSED to: SESSION_OPENING
22:18:09.667 DEBUG: BDScanCallback/commandState state:STOPPED action: CLIENT_REMOVED
22:18:09.668 INFO: CONNECTING: A17B5529
22:18:09.677 DEBUG: connect search complete
22:18:09.683 DEBUG: ConnectionHandler/state: CONNECTING action: ADVERTISEMENT_HEAD_RECEIVED
22:18:09.685 DEBUG: ConnectionHandler/state: CONNECTING action: ADVERTISEMENT_HEAD_RECEIVED
22:18:09.689 DEBUG: ConnectionHandler/state: CONNECTING action: ADVERTISEMENT_HEAD_RECEIVED
22:18:09.690 DEBUG: ConnectionHandler/state: CONNECTING action: ADVERTISEMENT_HEAD_RECEIVED
22:18:09.691 DEBUG: BDGattCallback/GATT state changed device newState: 2 status: 0
22:18:09.694 DEBUG: ConnectionHandler/state: CONNECTING action: ADVERTISEMENT_HEAD_RECEIVED
22:18:09.696 DEBUG: ConnectionHandler/state: CONNECTING action: ADVERTISEMENT_HEAD_RECEIVED
22:18:09.699 DEBUG: BDGattCallback/onMtuChanged status: 0
22:18:09.699 DEBUG: ConnectionHandler/state: CONNECTING action: ADVERTISEMENT_HEAD_RECEIVED
22:18:09.700 DEBUG: BDDeviceSessionImpl/handleMtuChanged status: 0 mtu: 232
22:18:09.701 DEBUG: BDGattCallback/Services discovered authentication is not needed
22:18:09.703 DEBUG: ConnectionHandler/state: CONNECTING action: DEVICE_CONNECTED
22:18:09.704 DEBUG: ConnectionHandler/ Session update from: SESSION_OPENING to: SESSION_OPEN
22:18:09.704 DEBUG: BDGattCallback/ phy updated tx: 2 rx: 2 status: 0
22:18:09.705 DEBUG: BDScanCallback/commandState state:STOPPED action: CLIENT_REMOVED
22:18:09.708 DEBUG: ConnectionHandler/state: CONNECTING action: EXIT
22:18:09.709 DEBUG: BDScanCallback/commandState state:STOPPED action: ADMIN_START_SCAN
22:18:09.709 DEBUG: BDScanCallback/commandState state:STOPPED action: EXIT
22:18:09.709 DEBUG: BDScanCallback/commandState state:IDLE action: ENTRY
22:18:09.800 INFO: Sign in succeeded
22:18:10.313 DEBUG: BDDeviceSessionImpl/ SERVICE: 00001800-0000-1000-8000-00805f9b34fb
22:18:10.316 DEBUG: BDDeviceSessionImpl/No client found for SERVICE: 00001800-0000-1000-8000-00805f9b34fb chrs: 4
22:18:10.318 DEBUG: BDDeviceSessionImpl/ SERVICE: 00001801-0000-1000-8000-00805f9b34fb
[... 32 lines cut ...]
22:18:10.588 DEBUG: BDDeviceSessionImpl/onDescriptorWrite uuid: 00002a19-0000-1000-8000-00805f9b34fb status: 0
22:18:10.660 DEBUG: BDDeviceSessionImpl/onDescriptorWrite uuid: fb005c81-02e7-f387-1cad-8acd2d8df0c8 status: 0
22:18:10.709 DEBUG: BDDeviceSessionImpl/onDescriptorWrite uuid: fb005c82-02e7-f387-1cad-8acd2d8df0c8 status: 0
22:18:10.758 INFO: Streaming feature PPG is ready
22:18:10.760 INFO: Streaming feature ACC is ready
22:18:10.761 INFO: Streaming feature GYRO is ready
22:18:10.762 INFO: Streaming feature MAGNETOMETER is ready
22:18:10.763 INFO: Streaming feature PPI is ready
22:18:10.802 INFO: BATTERY LEVEL: 60
22:18:10.878 INFO: uuid: 00002a23-0000-1000-8000-00805f9b34fb value: U{�����
22:18:10.923 INFO: uuid: 00002a28-0000-1000-8000-00805f9b34fb value: 1.1.5
22:18:10.973 INFO: uuid: 00002a26-0000-1000-8000-00805f9b34fb value: 0.1.5
22:18:11.019 INFO: uuid: 00002a27-0000-1000-8000-00805f9b34fb value: 00784292.02
22:18:11.090 INFO: uuid: 00002a25-0000-1000-8000-00805f9b34fb value: A17B5529 
22:18:11.137 INFO: uuid: 00002a24-0000-1000-8000-00805f9b34fb value: INW4J
22:18:11.184 INFO: uuid: 00002a29-0000-1000-8000-00805f9b34fb value: Polar Electro Oy 
22:18:11.186 DEBUG: complete
22:18:11.231 DEBUG: BDDeviceSessionImpl/onDescriptorWrite uuid: 00002a37-0000-1000-8000-00805f9b34fb status: 0
22:18:11.401 ERROR: ACC stream failed. Reason com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command error:  failed with error: ERROR_ALREADY_IN_STATE
22:18:11.777 DEBUG: BleHrClient/Processing service data. Status: 0.  Data length: 2
22:18:11.873 DEBUG: BlePMDClient/pmd data:  HEX: 02 96 5B 06 5D EE AB 7F 09 80 8E F1 B9 FB 34 FB 08 1D 2A 1A 1D 2B EE E5 ED DB F1 99 02 FB A7 F0 D7 EB 0B D7 04 19 D7 06 38 E7 34 12 10 5B 0C 36 24 07 2F 22 E7 57 DE F3 1B F2 1F 07 E4 05 F4 F4 F4 E7 FC 11 FA 0F FC F3 1A 12 04 2F FD 14 25 E8 1B 15 17 3C 01 F4 32 BA FF 13 06 FA F6 DB FA E4 AC F8 C0 AE FE BC A0 E1 E0 0A 0A EB E3 DF 3E F4 ED CB 20 80 FD FA 33 00 81 07 32 E4 FF BE 12 DA 07 91 48 FC EE 47 62 C0 0A 3F C4 AF 42 FE FE A3 00 08 0A EC 1D 06 E7 2A FE E7 20 C3 E3 22 FF D0 0F E9 2B F4 01 0D 17 01 D0 08 34 E7 FA 53 CA BA 01 0A 0D BD 27 00 41 EB EA BF 6F 77 F7 B4 5F AC FB E7 E1 33 DF 7C 09 ED 5F 3F 0B FC 1B E4 62 BF 02 47 24 00 02 13 36 D8 4F 01 10 D7 13 F1 02 EE 3F 88 E0 3E 
22:18:11.875 DEBUG: Error: BlePMDClient/No factor found for type: ACC
22:18:12.795 DEBUG: BleHrClient/Processing service data. Status: 0.  Data length: 2
22:18:13.070 DEBUG: BlePMDClient/pmd data:  HEX: 02 B0 90 73 9F EE AB 7F 09 80 6A F2 DB FB BC FA 0A 38 22 D4 10 3F 07 07 20 60 7B FE 15 F4 70 FF FC 95 47 6F 01 DE F6 EF 70 BB 06 79 98 D0 41 0A 56 DC 1F 84 0C F8 EF F0 C0 FF 29 EC AF 3F 04 D0 47 80 7B E2 2B B0 1E B7 FF 70 0F BF C1 C7 E0 CF 5F F5 12 E4 7B 2E 0E E8 F9 3B 23 FD 12 2C A8 FF 46 D8 E4 A3 32 BA 00 7A 24 3F 7F 15 9C 1B A0 FD E9 1F 20 F0 B8 0B E1 3B 9F 7F F5 48 80 0F 7C 18 13 98 4F 04 08 2A EC D0 42 0D 6E 40 81 05 23 1F BC A0 45 07 31 0C B1 42 10 E0 B7 F0 C6 D7 0B D8 40 30 02 48 E8 1A 01 04 5F C7 7F 76 11 5A 24 AA 3A 0C 50 F2 65 3A B9 57 49 CE FE 35 82 AB D3 81 EC 43 29 1F C3 35 79 67 61 C8 D6 50 84 90 FB 11 11 38 A1 80 F9 E9 B7 30 3C 06 1E 70 CE BE F9 D6 F7 AF FE F1 
22:18:13.071 DEBUG: Error: BlePMDClient/No factor found for type: ACC

sach16795, 2021-09-09:
@JOikarinen Please let me know if you can help me with this. I would highly appreciate any help on this issue.

JOikarinen, 2021-09-10:
Hi @sach16795, good report and good logs. 👍  Let's start the investigations.

From the logs I can see the following things happening:
- the BLE connection to the Polar Verity Sense with id:A17B5529 is established
> 22:18:11.401 ERROR: ACC stream failed. Reason com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command error: failed with error: ERROR_ALREADY_IN_STATE
- this means that when your application sends ACC start to Polar Verity Sense, the device responds that it is already in ACC data streaming state. This is not the state the Polar Verity Sense shall be after reconnection, as the device stops the streaming on disconnect.
> 22:18:13.071 DEBUG: Error: BlePMDClient/No factor found for type: ACC
- this happens because the above stream start failure. 

First thought:
- Is there any change your app instantiates the PolarBleApi twice? There was some similarities on issue discussed in https://github.com/polarofficial/polar-ble-sdk/issues/175#issuecomment-898287604

I will continue thinking of how the described scenario may happen.

sach16795, 2021-09-11:
Hello @JOikarinen, Thanks for the insight. 

From my code flow, the "PolarBleApiDefaultImpl.defaultImplementation(applicationContext, PolarBleApi.ALL_FEATURES)" call is only made once. But I am calling api.shutdown in the onDestroy method of the service.

sach16795, 2021-09-11:
Additional question, Now that I'm reevaluating the flow, I can see that I call PolarBleApi.shutdown() in the onDestroy method, and PolarBleApi.disconnectFromDevice(deviceId) is not called. I might have naively assumed that it would also internally trigger the disconnect call. Can this be the issue?

If so, can you help me understand the functionality of the shutdown call? Thanks in advance

JOikarinen, 2021-09-13:
> If this is not the ideal implementation, can you recommend how I should handle the flow?
> 
> Right now the flow is
> 
> 1. Service starts.
> 2. Service gets PolarBleApi instance.
> 3. Service onDestroy is called where I am calling the PolarBleApi.shutdown()
> 4. The flow repeats again if the user restarts the activity recording.
- the implementation flow you describe is very good

JOikarinen, 2021-09-13:
> I might have naively assumed that it would also internally trigger the disconnect call. Can this be the issue?
- your assumption is correct. ```PolarBleApi.shutdown()``` shall close the connection.

JOikarinen, 2021-09-13:
@sach16795 yet I haven't spotted anything incorrect from your explanation how you use the SDK.  Is it possible to share the logs from the time service is shutdown until the time service is up again and hitting the  ```failed with error: ERROR_ALREADY_IN_STATE``` problem? If possible could you also leave the process id's to the log. 

Have you tried with different Android phones or Android versions, still the same result?

JOikarinen, 2021-09-14:
@sach16795

> In the logs I can see at times if the service process is killed, it tries to restart itself. In the onCreate method, I try to initialize the Polar device and establish the connection. However, once I start to read the accelerometer stream, the following error is shown.

Summing up what we know at the moment:
Once application process dies and the foreground service is killed,  the application process is relaunched together with the foreground service again. The service `onCreate()` does reconnection to Polar Verity Sense, according to logs its created successfully. Only weird part in the logs is the MTU change status 4, the status 4 cannot be found Android GATT specification. However, connection is recreated. Once the ACC stream start is requested for the Polar Verity Sense, it responds with ERROR_ALREADY_IN_STATE. For the Polar BLE SDK library this means unrecoverable situation as the assumption by the Polar BLE SDK library is that after connection is established the Polar Verity Sense is not in the streaming state. 

More things what you could double check:
- once the service is relaunched, is the connection recreated to the device you expect? Do you have other Polar Verity Sense devices around?
- does the disconnection to Polar Verity sense really happen when service dies?

sach16795, 2021-09-17:
> Is it possible to share the logs from the time service is shutdown until the time service is up again and hitting the `failed with error: ERROR_ALREADY_IN_STATE` problem? If possible could you also leave the process id's to the log.

2021-09-17 10:00:26.160 31795 TRACE: STOP, 1631844026160, 4.65975, -6.91605, -1.72656
2021-09-17 10:00:26.160 31795 TRACE: STOP, 1631844026160, 5.04234, -6.72966, -1.82466
2021-09-17 10:00:26.160 31795 TRACE: STOP, 1631844026160, 4.08096, -7.46541, -2.22687
2021-09-17 10:00:26.160 31795 TRACE: STOP, 1631844026160, 2.72718, -7.99515, -2.4525
2021-09-17 10:00:26.160 31795 TRACE: STOP, 1631844026160, 2.18763, -7.69104, -2.10915
2021-09-17 10:00:26.168 31795 INFO: Saved accelDataSample item: 25958631-8ca7-4583-929f-d454031f3684, beaconCount: 19
2021-09-17 10:00:29.657 11971 INFO: Logs can be found in "/storage/emulated/0/Android/data/edu.nus.polar/files"
2021-09-17 10:00:29.880 11971 INFO: PolarApp on create exited
2021-09-17 10:00:29.918 11971 INFO: AccelerometerDataLogger onCreate entered, process id 11971
2021-09-17 10:00:29.934 11971 INFO: Polar device setting initialization begin
2021-09-17 10:00:30.036 11971 INFO: BLE power: true
2021-09-17 10:00:30.060 11971 INFO: Polar device setting initialization exited
2021-09-17 10:00:30.080 11971 INFO: Amplify initialization started
2021-09-17 10:00:30.169 11971 INFO: SIGNED_IN
2021-09-17 10:00:30.256 11971 INFO: Initialized Amplify, process id 11971
2021-09-17 10:00:30.331 11971 INFO: AccelerometerDataLogger registered for EventBus 
2021-09-17 10:00:30.344 11971 INFO: onCreate exited
2021-09-17 10:00:30.375 11971 INFO: Amplify DataStore start
2021-09-17 10:00:30.377 11971 INFO: Amplify datastore stopped.
2021-09-17 10:00:30.377 11971 INFO: AccelerometerDataLogger unregistered from EventBus 
2021-09-17 10:00:30.378 11971 INFO: onDestroy called 139246381
2021-09-17 10:00:30.383 11971 INFO: Registering push notifications token: c6qUaRkPQu63ZcYc6T3yQU:APA91bGbOrmhkehZURNAK-8vh_gkg8mSJ3fS8Ns_Gq2ShegAQiyE8NAv9I5-VQkxGeOe302Zz8gI4qmYJwEUTh9b7Q4YWhOAW5zlsEnmervaxv6qNixIveIk4oQWWBzmG1eQK123BPHF
2021-09-17 10:00:34.109 11971 INFO: Sign in succeeded
2021-09-17 10:01:46.669 11971 INFO: MainActivity onCreate entered
2021-09-17 10:01:46.669 11971 INFO: android.permission.ACCESS_COARSE_LOCATION permission already granted
2021-09-17 10:01:46.669 11971 INFO: android.permission.ACCESS_FINE_LOCATION permission already granted
2021-09-17 10:01:46.670 11971 INFO: android.permission.READ_EXTERNAL_STORAGE permission already granted
2021-09-17 10:01:46.670 11971 INFO: android.permission.WRITE_EXTERNAL_STORAGE permission already granted
2021-09-17 10:01:46.797 11971 INFO: MainActivity registered for EventBus 
2021-09-17 10:01:46.798 11971 INFO: onCreate exit
2021-09-17 10:01:48.427 11971 INFO: Walking training button clicked
2021-09-17 10:01:51.362 11971 INFO: accelerometer button clicked
2021-09-17 10:01:51.362 11971 INFO: AccelerometerDataLogger service start initiated
2021-09-17 10:01:51.375 11971 INFO: AccelerometerDataLogger onCreate entered, process id 11971
2021-09-17 10:01:51.378 11971 INFO: Polar device setting initialization begin
2021-09-17 10:01:51.383 11971 INFO: BLE power: true
2021-09-17 10:01:51.392 11971 INFO: Polar device setting initialization exited
2021-09-17 10:01:51.393 11971 INFO: Amplify initialization started
2021-09-17 10:01:51.405 11971 ERROR: Amplify already configured
2021-09-17 10:01:51.413 11971 INFO: Amplify DataStore start
2021-09-17 10:01:51.435 11971 INFO: AccelerometerDataLogger registered for EventBus 
2021-09-17 10:01:51.443 11971 INFO: onCreate exited
2021-09-17 10:01:51.444 11971 INFO: onStartCommand called
2021-09-17 10:01:51.444 11971 INFO: Trying to connect to device A17B5529
2021-09-17 10:01:51.447 11971 DEBUG: BDScanCallback/commandState state:IDLE action: CLIENT_START_SCAN
2021-09-17 10:01:51.447 11971 DEBUG: BDScanCallback/commandState state:IDLE action: EXIT
2021-09-17 10:01:51.447 11971 DEBUG: BDScanCallback/commandState state:SCANNING action: ENTRY
2021-09-17 10:01:51.447 11971 DEBUG: BDScanCallback/timestamps left: 0
2021-09-17 10:01:51.448 11971 DEBUG: BDScanCallback/Scan started -->
2021-09-17 10:01:51.457 11971 DEBUG: BDScanCallback/Scan started <--
2021-09-17 10:01:51.457 11971 INFO: onStartCommand exited
2021-09-17 10:01:51.481 11971 DEBUG: BleGattBase/Added notification characteristic for 00002a19-0000-1000-8000-00805f9b34fb
2021-09-17 10:01:51.484 11971 DEBUG: BleGattBase/Added notification characteristic for fb005c81-02e7-f387-1cad-8acd2d8df0c8
2021-09-17 10:01:51.484 11971 DEBUG: BleGattBase/Added notification characteristic for fb005c82-02e7-f387-1cad-8acd2d8df0c8
2021-09-17 10:01:51.486 11971 DEBUG: BleGattBase/Added notification characteristic for fb005c51-02e7-f387-1cad-8acd2d8df0c8
2021-09-17 10:01:51.486 11971 DEBUG: BleGattBase/Added notification characteristic for fb005c52-02e7-f387-1cad-8acd2d8df0c8
2021-09-17 10:01:51.486 11971 DEBUG: BDDeviceListenerImpl/new device allocated name: V*Q(&CD&K#*Q\&>9
2021-09-17 10:01:51.486 11971 DEBUG: BDDeviceList/new session added: V*Q(&CD&K#*Q\&>9
2021-09-17 10:01:51.491 11971 DEBUG: BleGattBase/Added notification characteristic for 00002a19-0000-1000-8000-00805f9b34fb
2021-09-17 10:01:51.492 11971 DEBUG: BleGattBase/Added notification characteristic for fb005c81-02e7-f387-1cad-8acd2d8df0c8
2021-09-17 10:01:51.492 11971 DEBUG: BleGattBase/Added notification characteristic for fb005c82-02e7-f387-1cad-8acd2d8df0c8
[... cut ...] FE 09 B0 38 09 20 0C C5 63 05 25 91 25 AB F0 0F F6 FF FF 8E E0 ED 2C A0 FA 99 1E 08 E7 BE E3 1C 00 F1 BB FD FD 59 FF E2 EE 8F F8 ED 2E 01 5E 10 F0 36 10 0A E6 FE FC 81 A0 F7 8F 5F 0C 7D BF F1 

At line 6, the service seems to be abruptly killed by the OS. Later service (AccelerometerDataLogger) is respawned. Another issue I am trying to locate is why the service is killed so often. Please let me know if this was helpful and if any other logs are needed. Absolutely sorry about the delay in reply.

sach16795, 2021-09-17:
> More things what you could double check:
> 
> * once the service is relaunched, is the connection recreated to the device you expect? Do you have other Polar Verity Sense devices around?

No other devices are around. And anyways, I specify the device ID to be connected to. 
> * does the disconnection to Polar Verity sense really happen when service dies?
Not necessarily, from the logs I can see the service die abruptly without onDestroy being called. @JOikarinen I'm not sure how to handle this. Please let me know if you have any insights.

zmukash, 2023-01-25:
@JOikarinen Same error just occurred in my case which is very similar if not identical to the one from @sach16795. Are there any new insights or bugfixes rolled out concerning this issue?

JOikarinen, 2023-01-26:
@zmukash thanks for the update. Let's see again, what do we understand of this bug. 

ERROR_ALREADY_IN_STATE is reply to start stream request in case VeritySense is already streaming. That's easy to reproduce by calling the start stream request twice, e.g.  subscribing to startAccStreaming() and then subscribing it again.  

Yet, I cannot understand the possible root cause of this issue. 

Are you using the foreground service? And if yes, do you know why the system kills it in first place?
If the service is destroyed by the system, do you dispose all the RX subscriptions you have made? For example 
```
     var disposable: Disposable? = null

     fun someFunction() {
           ...
          disposable = polarBleApi.startAccStreaming(polarId, settings)
          ...
     }

  override fun onDestroy() {
          disposable?.dispose()
    }
```

jimmyzumthurm, 2023-09-08:
Hi,
Looking at the logs provided, it seems that you are receiving accelerometer data over the BLE Pmd client Data characteristic , example here : 

22:18:11.873 DEBUG: BlePMDClient/pmd data: HEX: 02 96 5B 06 5D EE AB 7F 09 80 8E F1 B9 FB 34 FB 08 1D 2A 1A 1D 2B EE E5 ED DB F1 99 02 FB A7 F0 D7 EB 0B D7 04 19 D7 06 38 E7 34 12 10 5B 0C 36 24 07 2F 22 E7 57 DE F3 1B F2 1F 07 E4 05 F4 F4 F4 E7 FC 11 FA 0F FC F3 1A 12 04 2F FD 14 25 E8 1B 15 17 3C 01 F4 32 BA FF 13 06 FA F6 DB FA E4 AC F8 C0 AE FE BC A0 E1 E0 0A 0A EB E3 DF 3E F4 ED CB 20 80 FD FA 33 00 81 07 32 E4 FF BE 12 DA 07 91 48 FC EE 47 62 C0 0A 3F C4 AF 42 FE FE A3 00 08 0A EC 1D 06 E7 2A FE E7 20 C3 E3 22 FF D0 0F E9 2B F4 01 0D 17 01 D0 08 34 E7 FA 53 CA BA 01 0A 0D BD 27 00 41 EB EA BF 6F 77 F7 B4 5F AC FB E7 E1 33 DF 7C 09 ED 5F 3F 0B FC 1B E4 62 BF 02 47 24 00 02 13 36 D8 4F 01 10 D7 13 F1 02 EE 3F 88 E0 3E

First byte is 0x02 which corresponds to accelerometer type. I can't give advice about Polar SDK implementation as it is not my domain of expertise, but it does seem that Verity Sense is already streaming ACC data when you try to start the stream, which would explain "ERROR_ALREADY_IN_STATE".

@samulimaa Do you have any idea why might happen ?

closed 2023-10-23 by jimmyzumthurm