polarofficial/polar-ble-sdk #187: How to parse Polar Verity Sense delta flames?

jins-tkomoda, 2021-09-10:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [*] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [*] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

I am tring to implement HR device integration with WebBluetooth. Followings are stream data(Acc), but I found a strange behavior around a samples count. Data1 says samples count is 70 and data length is consistent. But data2 says samples count is 27, but total data length is not so much different with data1. I would be grad if you could tell me the specifications.

** Data1 (correct) **

delta size in bit(index16): 8
samples count(index17): 70 

{"0":2,"1":228,"2":87,"3":52,"4":194,"5":160,"6":38,"7":128,"8":9,"9":128,"10":154,"11":242,"12":201,"13":250,"14":141,"15":248,"16":8,"17":70,"18":232,"19":247,"20":2,"21":249,"22":250,"23":254,"24":26,"25":254,"26":16,"27":13,"28":14,"29":241,"30":206,"31":254,"32":253,"33":3,"34":247,"35":233,"36":33,"37":239,"38":11,"39":52,"40":26,"41":2,"42":233,"43":13,"44":28,"45":218,"46":241,"47":239,"48":234,"49":204,"50":10,"51":15,"52":15,"53":13,"54":36,"55":10,"56":250,"57":232,"58":0,"59":236,"60":232,"61":19,"62":249,"63":16,"64":9,"65":252,"66":249,"67":244,"68":24,"69":252,"70":15,"71":251,"72":253,"73":17,"74":254,"75":15,"76":232,"77":251,"78":6,"79":1,"80":249,"81":14,"82":18,"83":9,"84":251,"85":240,"86":255,"87":255,"88":224,"89":5,"90":240,"91":250,"92":244,"93":13,"94":22,"95":14,"96":253,"97":10,"98":247,"99":250,"100":8,"101":2,"102":251,"103":1,"104":27,"105":31,"106":247,"107":235,"108":219,"109":254,"110":244,"111":11,"112":251,"113":17,"114":250,"115":4,"116":12,"117":1,"118":255,"119":238,"120":17,"121":235,"122":245,"123":253,"124":18,"125":14,"126":246,"127":243,"128":249,"129":6,"130":17,"131":8,"132":1,"133":252,"134":230,"135":9,"136":249,"137":13,"138":247,"139":10,"140":11,"141":17,"142":10,"143":13,"144":243,"145":237,"146":228,"147":224,"148":2,"149":17,"150":25,"151":5,"152":236,"153":243,"154":12,"155":35,"156":249,"157":4,"158":233,"159":10,"160":231,"161":251,"162":13,"163":6,"164":254,"165":55,"166":10,"167":1,"168":239,"169":6,"170":28,"171":208,"172":232,"173":252,"174":251,"175":241,"176":235,"177":255,"178":239,"179":14,"180":25,"181":26,"182":242,"183":230,"184":5,"185":18,"186":12,"187":0,"188":226,"189":11,"190":2,"191":242,"192":19,"193":31,"194":33,"195":222,"196":1,"197":16,"198":227,"199":4,"200":245,"201":21,"202":0,"203":10,"204":28,"205":255,"206":242,"207":251,"208":227,"209":250,"210":10,"211":253,"212":244,"213":250,"214":253,"215":0,"216":253,"217":244,"218":9,"219":0,"220":23,"221":12,"222":251,"223":15,"224":241,"225":245,"226":4,"227":27}

** Data2 (incorrect??) **

delta size in bit(index16): 8
samples count(index17): 27 <- incorrect??
{"0":2,"1":96,"2":122,"3":40,"4":125,"5":159,"6":38,"7":128,"8":9,"9":128,"10":135,"11":242,"12":180,"13":250,"14":146,"15":248,"16":8,"17":27,"18":10,"19":5,"20":255,"21":234,"22":24,"23":255,"24":233,"25":255,"26":248,"27":41,"28":233,"29":3,"30":40,"31":253,"32":2,"33":19,"34":10,"35":3,"36":223,"37":253,"38":25,"39":225,"40":229,"41":229,"42":230,"43":0,"44":7,"45":28,"46":2,"47":14,"48":14,"49":16,"50":244,"51":252,"52":235,"53":14,"54":254,"55":37,"56":231,"57":250,"58":243,"59":244,"60":245,"61":5,"62":10,"63":1,"64":18,"65":22,"66":244,"67":20,"68":242,"69":30,"70":204,"71":247,"72":6,"73":13,"74":0,"75":6,"76":9,"77":254,"78":12,"79":248,"80":0,"81":233,"82":238,"83":248,"84":240,"85":255,"86":1,"87":7,"88":27,"89":22,"90":253,"91":241,"92":5,"93":252,"94":26,"95":251,"96":18,"97":239,"98":6,"99":6,"100":10,"101":127,"102":107,"103":167,"104":127,"105":49,"106":41,"107":24,"108":61,"109":196,"110":9,"111":240,"112":34,"113":111,"114":34,"115":32,"116":62,"117":129,"118":60,"119":59,"120":192,"121":22,"122":6,"123":0,"124":8,"125":30,"126":227,"127":246,"128":245,"129":255,"130":15,"131":8,"132":255,"133":3,"134":255,"135":23,"136":243,"137":20,"138":244,"139":9,"140":4,"141":5,"142":24,"143":221,"144":219,"145":237,"146":9,"147":21,"148":235,"149":248,"150":63,"151":1,"152":251,"153":4,"154":12,"155":36,"156":221,"157":28,"158":255,"159":206,"160":195,"161":253,"162":14,"163":11,"164":0,"165":18,"166":249,"167":26,"168":8,"169":1,"170":232,"171":6,"172":9,"173":224,"174":255,"175":6,"176":252,"177":2,"178":2,"179":19,"180":243,"181":28,"182":251,"183":243,"184":246,"185":10,"186":0,"187":3,"188":11,"189":26,"190":236,"191":245,"192":8,"193":26,"194":239,"195":246,"196":233,"197":31,"198":249,"199":6,"200":238,"201":250,"202":251,"203":4,"204":248,"205":14,"206":5,"207":23,"208":0,"209":247,"210":250,"211":253,"212":251,"213":243,"214":6,"215":239,"216":6,"217":4,"218":250,"219":77,"220":21,"221":53,"222":17,"223":61,"224":51,"225":209,"226":35}

Polar Verity Sense firmware version is 1.1.5

JOikarinen, 2021-09-10:
Hi @jins-tkomoda,

so one received frame can actually contain multiple delta packages. Here is the data decoded

```Type ACC: ”0”:2,
Time stamp "1":96,"2":122,"3":40,"4":125,"5":159,"6":38,"7":128,"8":9,
Delta package "9":128,

Delta decoded
Reference sample "10":135,"11":242,"12":180,"13":250,"14":146,"15":248,
Delta package 1, bits: "16":8,
Delta package 1 size: “17":27,
Delta package 1 data: 
"18":10,"19":5,"20":255, "21":234,"22":24,"23":255,"24":233,"25":255,"26":248,"27":41,"28":233,"29":3,"30":40,"31":253,"32":2,"33":19,"34":10,"35":3,"36":223,"37":253,"38":25,"39":225,"40":229,"41":229,"42":230,"43":0,"44":7,"45":28,"46":2,"47":14,"48":14,"49":16,"50":244,"51":252,"52":235,"53":14,"54":254,"55":37,"56":231,"57":250,"58":243,"59":244,"60":245,"61":5,"62":10,"63":1,"64":18,"65":22,"66":244,"67":20,"68":242,"69":30,"70":204,"71":247,"72":6,"73":13,"74":0,"75":6,"76":9,"77":254,"78":12,"79":248,"80":0,"81":233,"82":238,"83":248,"84":240,"85":255,"86":1,"87":7,"88":27,"89":22,"90":253,"91":241,"92":5,"93":252,"94":26,"95":251,"96":18,"97":239,"98":6,

Delta package 2, bits: "99":6,
Delta package 2 size: "100":10,
Delta package 2 data: 
"101":127,"102":107,"103":167,"104":127,"105":49,"106":41,"107":24,"108":61,"109":196,"110":9,"111":240,"112":34,"113":111,"114":34,"115":32,"116":62,"117":129,"118":60,"119":59,"120":192,"121":22,"122":6,"123":0,

Delta package 3, bits: "124":8
Delta package 3 size: ,"125":30,
Delta package 3 data: 

"126":227,"127":246,"128":245,"129":255,"130":15,"131":8,"132":255,"133":3,"134":255,"135":23,"136":243,"137":20,"138":244,"139":9,"140":4,"141":5,"142":24,"143":221,"144":219,"145":237,"146":9,"147":21,"148":235,"149":248,"150":63,"151":1,"152":251,"153":4,"154":12,"155":36,"156":221,"157":28,"158":255,"159":206,"160":195,"161":253,"162":14,"163":11,"164":0,"165":18,"166":249,"167":26,"168":8,"169":1,"170":232,"171":6,"172":9,"173":224,"174":255,"175":6,"176":252,"177":2,"178":2,"179":19,"180":243,"181":28,"182":251,"183":243,"184":246,"185":10,"186":0,"187":3,"188":11,"189":26,"190":236,"191":245,"192":8,"193":26,"194":239,"195":246,"196":233,"197":31,"198":249,"199":6,"200":238,"201":250,"202":251,"203":4,"204":248,"205":14,"206":5,"207":23,"208":0,"209":247,"210":250,"211":253,"212":251,"213":243,"214":6,"215":239,

Delta package 4, bits: "216":6,
Delta package 4 size: ,"217":4,
Delta package 4 data: 
"218":250,"219":77,"220":21,"221":53,"222":17,"223":61,"224":51,"225":209,"226":35
```

jins-tkomoda, 2021-09-11:
@JOikarinen 

Thank you for the commentary! I will try that.

jins-tkomoda, 2021-09-13:
@JOikarinen
I was able to successfully retrieve and parse the data with a web browser. Thank you very much !

closed 2021-09-13 by jins-tkomoda

pablomainar, 2021-11-03:
Hi @JOikarinen I am also trying to make sense of the delta encoding for Verity Sense and PPG data. These is the first data packet that I get:

"0":1; "1":100; "2":44; "3":183; "4":149; "5":58; "6":210; "7":144; "8":9; "9":128; "10":75; "11":102; "12":253; "13":200; "14":230; "15":253; "16":35; "17":160; "18":253; "19":244; "20":182; "21":248; "22":10; "23":19; "24":106; "25":56; "26":130; "27":197; "28":15; "29":114; "30":56; "31":195; "32":199; "33":230; "34":198; "35":120; "36":49; "37":73; "38":8; "39":183; "40":244; "41":227; "42":205; "43":3; "44":173; "45":172; "46":178; "47":135; "48":9; "49":140; "50":220; "51":113; "52":12; "53":249; "54":68; "55":96; "56":179; "57":201; "58":2; "59":141; "60":0; "61":98; "62":200; "63":255; "64":129; "65":52; "66":97; "67":132; "68":1; "69":58; "70":164; "71":162; "72":132; "73":247; "74":81; "75":208; "76":112; "77":6; "78":2; "79":122; "80":176; "81":49; "82":70; "83":3; "84":70; "85":104; "86":98; "87":3; "88":7; "89":50; "90":156; "91":128; "92":130; "93":240; "94":45; "95":12; "96":33; "97":2; "98":3; "99":98; "100":120; "101":193; "102":6; "103":246; "104":135; "105":140; "106":35; "107":136; "108":14; "109":198; "110":212; "111":177; "112":10; "113":5; "114":117; "115":248; "116":177; "117":74; "118":5; "119":12; "120":18; "121":49; "122":160; "123":3; "124":58; "125":224; "126":253; "127":195; "128":255; "129":252; "130":147; "131":239; "132":253; "133":54; "134":191; "135":238; "136":86; "137":191; "138":1; "139":208; "140":222; "141":235; "142":247; "143":30; "144":5; "145":57; "146":222; "147":218; "148":230; "149":253; "150":252; "151":156; "152":190; "153":227; "154":178; "155":174; "156":3; "157":69; "158":142; "159":223; "160":78; "161":238; "162":253; "163":137; "164":14; "165":233; "166":253; "167":62; "168":255; "169":29; "170":255; "171":234; "172":155; "173":110; "174":254; "175":105; "176":239; "177":239; "178":35; "179":63; "180":0; "181":88; "182":127; "183":250; "184":201; "185":239; "186":1; "187":186; "188":223; "189":247; "190":180; "191":207; "192":253; "193":243; "194":79; "195":0; "196":212; "197":15; "198":1; "199":255; "200":191; "201":1; "202":32; "203":240; "204":0; "205":10; "206":0; "207":3; "208":69; "209":240; "210":255; "211":90; "212":96; "213":2; "214":67; "215":48; "216":255; "217":59; "218":48; "219":3; "220":57; "221":144; "222":255; "223":217; "224":31; "225":1; "226":233; "227":79; "228":254;

Following the documentation and your example, here is how I decode it:
Measurement type (1B): "0": 1 (PPG)
Timestamp (8B): "1":100; "2":44; "3":183; "4":149; "5":58; "6":210; "7":144; "8":9
Frame type (1B): "9":128

Reference sample (resolution=22 and nb_channels=4, so 22*4/8=11B): "10":75; "11":102; "12":253; "13":200; "14":230; "15":253; "16":35; "17":160; "18":253; "19":244; "20":182

Delta package 1 size in bits (1B): "21":248 (**size of 248 bits??**)
Delta package 1 samples count (1B): "22":10
Delta package 1 data (I would say 248 * 10 * 4 channels = 9920 bits = 1240 B, which sounds ridiculous): ...

What am I missing? What does byte 21 mean?

JOikarinen, 2021-11-05:
> Measurement type (1B): "0": 1 (PPG)
> Timestamp (8B): "1":100; "2":44; "3":183; "4":149; "5":58; "6":210; "7":144; "8":9
> Frame type (1B): "9":128
- this is correct

> Reference sample (resolution=22 and nb_channels=4, so 22*4/8=11B): "10":75; "11":102; "12":253; "13":200; "14":230; "15":253; "16":35; "17":160; "18":253; "19":244; "20":182
- reference sample is always full bytes
- the correct reference sample is:  "10":75; "11":102; "12":253; "13":200; "14":230; "15":253; "16":35; "17":160; "18":253; "19":244; "20":182; "21":248; 
    - sample 1: "10":75; "11":102; "12":253;  => 0x4B, 0x66, 0xFD => 0xFD664B = -170421
    - sample 2: "13":200; "14":230; "15":253; => 0xC8, 0xE6, 0xFD => 0xFDE6C8 = -137528
    - sample 3: "16":35; "17":160; "18":253; => 0x23, 0xA0, 0xFD => 0xFDA023 = -155613
    - sample 4: "19":244; "20":182; "21":248; => 0xF4, 0xB6, 0xF8 => 0xF8B6F4 = -477452
 
Delta package 1 size in bits (1B): "22":10; (size of 10 bits)
Delta package 1 samples count (1B): "23":19;

MariahSabioni, 2022-08-03:
@JOikarinen if I may add a beginner's question regarding the same issue:
 I understand now that multiple delta packages can come on the same frame. But for example for the Acc stream, how does one parse the data when the delta size is not 8 bit?  Should I get byte[0]'s 6 bits, then combine byte[0]'s bits with byte[1]'s bits and so on?
Thank you in advance!

JOikarinen, 2022-08-04:
Hi @MariahSabioni,

This is not nice, but unfortunately I must say this statement for non Android/iOS projects:

We are only able to support Android and iOS. The SDK includes source code and you can feel free to utilize / port the code in other platforms too but that is something we are not able to provide support for.

However, let's have a look since your question shall be easy to answer. 

> Should I get byte[0]'s 6 bits, then combine byte[0]'s bits with byte[1]'s bits and so on?

Exactly. Let's take an example. 

```
// Received ACC frame in HEX:  02 38 6C 31 72 A4 D3 23 0D 80 71 07 F0 6A 9E 8D 0A 38 BE 5C BE BA 2F 96 B3 EE 4B E5 AD FB 42 B9 EB BE 4C FE BA 2F 92 BF EE 4B E4 B1 FB 12 B9 EC BD 3C 3E BB 2F 8F D3 DE 4B E3 B5 F7 D2 B8 ED BD 30 7E 7B 2F 8B E3 CE 8B E2 BA F7 A2 B8 EE BC 20 BE 7B 2F 88 F3 CE CB E1 BD EF 52 F8 EF BC 18 FE 3B 2F 84 03 BF CB E0 C2 EF 32 B8 F0 BB 04 4E BC 2E 81 13 AF 0B E0 C6 EF F2 F7 F1 B9 FC 7D BC 2E 7D 27 9F 4B DF CA EB C2 F7 F2 B8 EC CD 7C 2E 7B 37 8F 4B DE CE E3 92 F7 F3 B8 E0 0D FD 2D 77 4B 7F CB DD D2 DF 62 37 F5 B7 D4 4D BD 2D 74 5B 6F CB DC D7 D7 32 37 F6 B5 C8 8D 7D 2D 71 6B 4F 4B DC DC D3 F2 36 F7 B4 BC DD FD 2C 6F 7B 3F 4B DB E0 CF D2 36 F8 B2 B0 2D BE 2C 6C 8F 1F CB DA E3 C7 A2 76 F9

// Header: 02 38 6C 31 72 A4 D3 23 0D 80 
// index      type                                          data
// 0:            Measurement type                 0x02 (Acc data)
// 1..8:         64-bit Timestamp                  0x38 0x6C 0x31 0x72 0xA4 0xD3 0x23 0x0D (0x0D23D3A472316C38 = 946833049921875000)
// 9:            Frame type                              0x80 (compressed)

// Data:
// 10-15:    Reference sample                   0x71 0x07 0xF0 0x6A 0x9E 0x8D
//      Sample 0 (aka. reference sample):
//      Sample 0 - channel 0: 71 07 => 0x0771 => 1905
//      Sample 0 - channel 1: F0 6A => 0x6AF0 => 27376
//      Sample 0 - channel 2: 9E 8D => 0x8D9E => -29282

// Delta dump: 0A 38 BE 5C BE BA 2F 96 B3 EE 4B E5 AD ...
// 16:      Delta size                          size 1:    0x0A (10 bits)
// 17:      Sample amount                 size 1:    0x38 (Delta block contains 56 samples)
// 18:                                                      0xBE (binary: 1011 1110)
// 19:                                                      0x5C (binary: 0101 11 | 00)
// 20:                                                     0xBE (binary: 1011 | 1110)
//      Sample 1 - channel 0, size 10 bits: 00 1011 1110b = 190 
//      Sample 1 - channel 1, size 10 bits: 11 1001 0111b = -105
// 21:                                                     0xBA (binary: 10 | 11 1010)
//      Sample 1 - channel 2, size 10 bits: 11 1010 1011b = -85
...
```
 Sample 0 - channel 0:  1905 
 Sample 0 - channel 1:  27376 
 Sample 0 - channel 2: -29282 
 
 Sample 1 - channel 0:  1905 + 190 = 2095
 Sample 1 - channel 1:  27376 + (-105) = 27271
 Sample 1 - channel 2: -29282 + (-85) =  -29367
 
 Sample 2 - channel 0:  2095 + (delta) = ...
 Sample 2 - channel 1:  27271 + (delta) = ...
 Sample 2 - channel 2: -29367 + (delta) =  ...
 
...

Finally to get the data in milliG you must multiple each value with the factor 0.24399999. 

Sample 0 - channel 0:  1905 * 0.24399999 = 464mg
 Sample 0 - channel 1:  27376  * 0.24399999 = 6679mg
 Sample 0 - channel 2: -29282  * 0.24399999 = -7144mg
 
 Sample 1 - channel 0: 2095  * 0.24399999 = 511mg
 Sample 1 - channel 1: 27271 * 0.24399999 = 6654mg
 Sample 1 - channel 2: -29367 * 0.24399999 = -7165mg

MariahSabioni, 2022-08-05:
@JOikarinen  I totally understand your point regarding support, my apologies.
Thank you for taking the time to answer, it was very clarifying.

marianyamukuru, 2023-01-23:
> @JOikarinen I was able to successfully retrieve and parse the data with a web browser. Thank you very much !

@jins-tkomoda do you mind explaining or sharing how you did it for 10 bit and/ or 6 bit delta frames? I'm having trouble with this conversion.. I'm not sure I understand how to do this correctly