polarofficial/polar-ble-sdk #189: How to calculate the timestamps right?

merschel, 2021-09-10:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [x] Device is not relevant for this question

**Description:**

Hi,

as far as I understand the devices deliver the data batchwise whereby each batch has only one timestamp (which is the timestamp for the last data point in the batch). For the other datapoints one must calculate the timestamps. Here in the github issue forum one can find 2 solutions. 

1. [assume that the sample rate is constant and use this to calculate the remaining timestamps](https://github.com/polarofficial/polar-ble-sdk/issues/154)
2. [use the time difference between two batches, the number of samples and evenly distribute the timestamps across the data points](https://github.com/polarofficial/polar-ble-sdk/issues/104)

If I use the second method, I will always get slightly different sample rates. So only one method can be correct. 

Can you provide us the exact algorithm how to do calculate the timestamps?

[You also communicated that you would make the timestamp calculation available in the SDK.](https://github.com/polarofficial/polar-ble-sdk/issues/154) In which release will this be done?

JOikarinen, 2021-09-11:
Hi @merschel, 

relevant questions👍  

> as far as I understand the devices deliver the data batchwise whereby each batch has only one timestamp (which is the timestamp for the last data point in the batch).
- correct
> For the other datapoints one must calculate the timestamps.
- also correct

Given that we know only one absolute timestamp per each received package (i.e. batch) and that timestamp is for last received sample, we must calculate the estimation for rest of the samples. I have two approaches in my mind:

Approach 1 - using the known frequency:
- the data stream is started with requested frequency, using that frequency the timestamp can be calculated for each sample
- pros: easy and quick calculation
- the algorithm:
```
timeStamp = packageAbsoluteTimeStamp
timeBetweenEachSample = 1/frequency
amountOfSamples  = packageAmountOfSamples

timeStampArray[] = arraySizeOf(amountOfSamples)
timeOfSample = timeStamp

for(i = (amountOfSamples - 1); i >= 0; i--) {
    timeStampArray[i] = timeOfSample
    timeOfSample -= timeBetweenEachSample
}
```

Approach 2 - using the known timestamps:
- as one absolute timestamp is known per received package, we must buffer two consecutive packages
- note, for the first received package the timestamp must be calculated using Approach 1 
- pros: more accurate compared to approach 1 as the calculation is based on absolute time stamps. Frequency for received data can be compared for requested frequency to double check data validity. Data validity check maybe useful in applications where many streams are started with high frequency, as there might occur the data package loses if the BLE communication channel is not capable of delivering all the data
- cons: the received data is buffered, as a consequence the data with timestamps may be delayed for some unpredictable period of time to be used within the application. Especially in Polar Verity Sense where the data is packaged efficiently the time between received packages may be seconds (depending on measured data, its frequency and the movements of the device) 
- the algorithm:
```
timeStamp1 = package_N_absoluteTimeStamp
timeStamp2 = package_N+1_absoluteTimeStamp
amountOfSamples  = package_N+1_amountOfSamples
timeBetweenEachSample = (timeStamp2 - timeStamp1)/amountOfSamples

timeStampArray[] = arraySizeOf(amountOfSamples)
timeOfSample = timeStamp1 + timeBetweenEachSample // sample0 

for(i = 0; i < (amountOfSamples - 1); i++) {
    timeStampArray[i] = timeOfSample 
    timeOfSample += timeBetweenEachSample
}
timeStampArray[amountOfSamples - 1] = timeStamp2
```

> [You also communicated that you would make the timestamp calculation available in the SDK.](https://github.com/polarofficial/polar-ble-sdk/issues/154) In which release will this be done?
- not estimation available but the item is on our backlog

merschel, 2021-09-16:
Thank you for the answers. I have tried both ways. The first one has big time jumps in the resulting time data because the summed sample rate does not fill up the time evenly until the next timestamp of the next packet. You can see this in the following graph. There I have plotted the differences of the resulting timestamp data (Matlab: diff(timeStampArray)). You can see that there is always a larger time jump. 

[image]

Can you confirm this?

The second approach, by its very nature, does not have this problem. On the other hand, this solution would only work if the timestamps sent are trustworthy. For example, there are no gaps between 2 packets. 

If you plot the difference between two calculated timestamps (Matlab diff(timeStampArray)), you also get slight jumps in the time data. In the test data set, these have a standard deviation of a few microseconds. It seems that these are technical fluctuations of the device. 

Can you confirm that the measurement accuracy is around 1 microsecond?

[image]

JOikarinen, 2022-12-13:
Release [4.0.0-beta0](https://github.com/polarofficial/polar-ble-sdk/releases/tag/4.0.0-beta0) fixes the problem with timestamps, as each sample is provided with timestamp.

closed 2022-12-13 by JOikarinen