polarofficial/polar-ble-sdk #190: problem with disconnect Polar H10 and OH1 on iOS

avik-efcom, 2021-09-12:
**Platform on which you observed the bug:** 
- [ ] Android 
- [X] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [X] Polar OH1
- [ ] Polar Verity Sense
- [X] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Hi, 
when I initiate disconnect from the devices it works, but if the device is disconnect (long-distance or turns off(OH1)), the function "devcieDisconected" is not called.
I try to put a print command on disconnect but it never shows
I don't use the auto-connect function so I turn it off
[image]

**How to Reproduce**
only connect and disconnect by long-distance or turn the device off and the function should be called

**Expected behavior**
the function "devcieDisconected" should call after the device disconnected

**Screenshots and logs**
[image]
on the log, you can see the HR notification stop but the function deviceDisconnected not called
[image]

thanks very much

JOikarinen, 2021-09-13:
Hi @avik-efcom,

I tried to reproduce the bug you explained. I took the [example app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-ios/iosBleSdkTestApp) and tried these steps with OH1 and H10:

STEPS:
1. Connect to OH1 
2. Pressed OH1 button to shut it down
3. Followed the console logs to see whether `deviceDisconnected` callback is called

OBSERVATION:
- 2021-09-13 12:04:52.792808+0300 iosBleSdkTestApp[1625:1323635] DISCONNECTED: (deviceId: "58A0EB20", address: 47D0734A-A5CB-58D9-39DE-B5527DCDC576, rssi: -55, name: "Polar OH1 58A0EB20", connectable: true)

STEPS:
1. Connect to H10 
2. H10 goes out of range
3. Followed the console logs to see whether `deviceDisconnected` callback is called

OBSERVATION:
-  2021-09-13 12:02:35.400766+0300 iosBleSdkTestApp[1621:1322550] DISCONNECTED: (deviceId: "3D474722", address: 1E1B9E05-0755-7911-A1CC-C7A24E5E2B76, rssi: -70, name: "Polar H10 3D474722", connectable: true)

Could you double check you have registered the PolarBleApiObserver for the PolarBleSDK?  Done like this in the [example ](https://github.com/polarofficial/polar-ble-sdk/blob/1585e1de6572d59fe6378830b0764375c912b676/examples/example-ios/iosBleSdkTestApp/iosBleSdkTestApp/PolarBleSdkManager.swift#L62) code: ``` api.observer = self```

avik-efcom, 2021-09-13:
[image]

I add now all the api-self functions from the example 
[image]

the connected function works ok
I attach the full log
deviceConnecting (deviceId: "80919F24", address: EC247EE7-23B4-4C7C-4C22-3A58F4FAC340, rssi: -20, name: "Polar OH1 80919F24", connectable: true)
deviceConnected (deviceId: "80919F24", address: EC247EE7-23B4-4C7C-4C22-3A58F4FAC340, rssi: -20, name: "Polar OH1 80919F24", connectable: true)
HR READY
battery level updated: 100
HR notification: 90 rrs: []
Feature acc is ready.
Feature ppg is ready.
Feature ppi is ready.
HR notification: 90 rrs: []
HR notification: 90 rrs: []
OP: list exercises completed
HR notification: 89 rrs: []
HR notification: 90 rrs: []
HR notification: 90 rrs: []
HR notification: 90 rrs: []
HR notification: 90 rrs: []
HR notification: 90 rrs: []
HR notification: 89 rrs: []
HR notification: 89 rrs: []
HR notification: 88 rrs: []
HR notification: 88 rrs: []
**//turn off the OH1 here**
HR notification: 0 rrs: []
**//suppost to get the disconnected print**
**//after a while ask for the exercise list (I did not know the device is disconnected from the code)**
failed to list exercises: PolarBleSdk.DeviceNotConnected

[image]

avik-efcom, 2021-09-13:
I add the logger and got this
[image]

it seems is the device disconnected but I don't know why I did not get it from the deviceDisconnected function

NiwreRelbag, 2021-12-03:
Hello,
I have a similar problem with the H10 (on iOS).
"deviceConnected" is called, but if I remove the H10 from the strap and wait a few minutes, "deviceDisconnected" is never called (and it continues to send ECG data).

When I use the H10 with the Std. CBCentralManager class, "didDisconnectPeripheral" is called about 1 minute after removing the H10 from the strap.

Thanks a lot!

JOikarinen, 2021-12-07:
@NiwreRelbag and @avik-efcom I think I found the root cause for the problem. As you described the ```deviceDisconnected``` was never called if the ```automaticReconnection``` was set to false.  The problem is now fixed in release 3.2.5

closed 2021-12-07 by JOikarinen