polarofficial/polar-ble-sdk #191: Is there a way to access the stored exercise data on the OH1?

KennethEvans, 2021-09-15:
***Platform your question concerns:*** 
- [x ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [x] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
It is my understanding the OH1 stores up to 200 hours of data. I have an exercise on Aug 18 that was recorded by Beat.  Beat has the HR and speed values but they were not uploaded to Flow as they usually are.  I have tried Polar FlowSync, but it gives no information as to what it did, just said it was synced (and the data are still not there). I would like to know if there is a way to access that data.

I apologize if this question is answered somewhere on the site.  I did not find it.

Thanks,

-Ken

JOikarinen, 2021-09-21:
Hi @KennethEvans,

Lets split the answer to cover your question from different aspects

Polar Beat app:
- if you have recorded the training with Beat app and used OH1 as heart rate sensor, to get the exercise synced onto your [Polar Flow](https://www.polar.com/en/flow) account you need to register to [Polar Flow](https://www.polar.com/en/flow) with your Beat app. 
- https://support.polar.com/en/polar-oh1-with-polar-beat-get-started-and-how-to-train
- https://support.polar.com/en/support/force_training_uploads_to_web

OH1:
- you may register the OH1 into your [Polar Flow](https://www.polar.com/en/flow) account too. Once you have paired the OH1 with the Polar Beat app the registration of OH1 is offered to you (if you are signed-in in Beat app) . Same registration of OH1 may be done using [Flow Sync](https://support.polar.com/en/support/FlowSync) or [Flow App](https://support.polar.com/en/support/Flow_app). Once you have registered the OH1 to [Polar Flow](https://www.polar.com/en/flow), it becomes possible to record you exercises on OH1 memory and transfer them later to your Flow account by syncing your OH1 using [Flow Sync](https://support.polar.com/en/support/FlowSync), [Beat app](https://www.polar.com/en/beat) or [Flow App](https://support.polar.com/en/support/Flow_app).
- https://support.polar.com/en/setting-up-your-oh1
- https://support.polar.com/en/how-to-sync-data-from-oh1-to-the-flow-web-service-and-app
- https://support.polar.com/e_manuals/OH1/Polar_OH1_user_manual_English/Content/Start-a-Training-Session.htm

Polar BLE SDK:
- If I understand your question correctly, the question is not directly related to Polar BLE SDK? Or do you have the attempt to read the files saved on OH1 memory using Polar BLE SDK?

@KennethEvans, please let me know did I managed to help you with my answer.

KennethEvans, 2021-09-21:
>> Or do you have the attempt to read the files saved on OH1 memory using Polar BLE SDK?

Yes. Sorry if I was unclear. I would like to know if there is a way to read the file[s] saved in the OH1.

KennethEvans, 2021-09-21:
> > > Or do you have the attempt to read the files saved on OH1 memory using Polar BLE SDK?
> 
> Yes. Sorry if I was unclear. I would like to know if there is a way to read the file[s] saved in the OH1 using the SDK.

KennethEvans, 2021-09-21:
> * If I understand your question correctly, the question is not directly related to Polar BLE SDK? Or do you have the attempt to read the files saved on OH1 memory using Polar BLE SDK?

Yes, I am sorry if I was unclear. I would like to know if there is a way to read the files in the OH1 memory (or even get a list of them) using the SDK.

JOikarinen, 2021-09-25:
Sorry, I also got the original question wrong. 

> I would like to know if there is a way to read the files in the OH1 memory (or even get a list of them) using the SDK.
- yes, that is possible 

1. The OH1 need to be [registered to Polar Flow ecosystem](https://github.com/polarofficial/polar-ble-sdk/issues/191#:~:text=en/support/force_training_uploads_to_web-,OH1%3A,-you%20may%20register) as explained above.
2. Do [exercise recording(s)](https://support.polar.com/e_manuals/OH1/Polar_OH1_user_manual_English/Content/Start-a-Training-Session.htm#:~:text=Start%20a%20training%20session%20as%20an%20independent%20training%20device) with OH1.  
3. List exercises with API function [listExercises](https://github.com/polarofficial/polar-ble-sdk/blob/78609b756fe500154b8b129b16ace64bd35c33b9/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L309)
4. Read exercise with API function [fetchExercise](https://github.com/polarofficial/polar-ble-sdk/blob/78609b756fe500154b8b129b16ace64bd35c33b9/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L322)

KennethEvans, 2021-09-25:
Thanks. I missed that.

I just ran the demo and did LIST EXERCISES.  All it said was list exercises complete. There was no error message.

So I assume there were none, which is what I would expect, since I use Beat and do not do manual. I am only interested because the support person in the incident mentioned in the first post said I needed to do that using FlowSync to get the missing data. It didn't work and looks to have been a waste of time in the first place.

The device **should** be registered with the Polar Flow ecosystem. It is not clear if there would have been an error message if it were not.

I'm also not sure why this is necessary. If it is using the Flow ecosystem, I already know the data are missing on the Flow website even though they are there in Beat.

In any event, I am not interested in pursuing this. I now know how to get the data. At the time I posted the incident I had not been working with the SDK for some time. I am more back up to speed now.

I do however regularly use the app that gets the ECG reading from the H10. That is invaluable to me.

Thank you for your help. I think you have done a good job with the SDK, and I appreciate your responses.

closed 2021-09-26 by JOikarinen