polarofficial/polar-ble-sdk #192: How to check currently-set epoch for timestamp

choplin, 2021-09-16:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [*] Platform is not relevant for this question

***Device:***
- [*] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
We use OH1 to collect raw PPG data via BLE access following [this specification](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf. ). If my understanding is correct, each packet we receive contains several samples and one timestamp for the last sample. This timestamp indicates elapsed time from the currently-set epoch in the device, and the default epoch varies depending on the model, e.g. [1.1.2000 for ECG model](https://github.com/polarofficial/polar-ble-sdk/blob/8e97d4eba5b5aca5c9511ff3c9b85a262c5f292a/README.md#h10-heart-rate-sensor-available-data-types) or [1.1.2017 10:00 for Verity Sense](https://github.com/polarofficial/polar-ble-sdk/issues/154#issuecomment-828940573). Epoch is configurable via [setLocalTime](https://github.com/polarofficial/polar-ble-sdk/blob/085bf6be0f8b19fa9d5222b9632768edfd9fa923/sources/Android/android-communications/src/main/java/polar/com/sdk/api/PolarBleApi.java#L185) API.

My questions are the following:

1. What is the default epoch for OH1? (Is there a documentation about default epoch?)
2. Is there any API to check the currently-set epoch?

Thank you!

JOikarinen, 2021-09-21:
> * What is the default epoch for OH1? (Is there a documentation about default epoch?)
- it is January 2, 1987 10:00:00.000 AM
> * Is there any API to check the currently-set epoch?
- unfortunately there is not. This is slight weakness in current API. However, as a workaround, you may consider to set the current time by setLocalTime API when needed. Once said, the workaround I am proposing, also have its own known issue, please the section the [known issues for Polar OH1 and Polar Verity Sense](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/KnownIssues.md#polar-verity-sense-and-polar-oh1).

choplin, 2021-09-29:
@JOikarinen Thank you for your reply! And sorry for the late response. I've checked the raw binary data retrieved from three OH1 devices to check if the timestamps within the data are based on the default epoch you mentioned by subtracting the timestamp from the data retrieval time. Unfortunately, the epochs estimated from our data are far from the expected default epoch, and also different among the devices.
* 2008-06-08 22:03:28
* 2004-01-15 03:54:45
* 2004-01-18 18:45:44

1. Dates of the estimated epoch seems equals the dates when devices were started to use. Is there any implicit step that resets the default epoch during device initialization?
2. I'm still wondering why years of the estimated epoch vary among the devices. Do you have any idea about this?

JOikarinen, 2021-09-29:
@choplin I am very sorry, I gave you false information in previous comment. 

The epoch time is 1.1.2000 00:00:00 and the default time in OH1 is 1.1.2017 10:00:00, so the stream timestamps starts running from 536580000000000ns

Once you set the OH1 time with setLocalTime API then this known issue applies: [known issues for Polar OH1 and Polar Verity Sense](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/KnownIssues.md#polar-verity-sense-and-polar-oh1). 

Once you set the OH1 time with setLocalTime API then reset back to default time 1.1.2017 10:00:00 requires either the [restoring defaults](https://support.polar.com/en/how-do-i-reset-my-polar-oh1) of OH1 or battery gets totally empty.

choplin, 2021-09-29:
@JOikarinen Thank you letting me know. But, we're sure that we have never called setLocalTime directly because our app does not use SDK but only BLE. 

In our use-case, the OH1 devices were paired and initialized using Polar Beat, then we collected data with our in-house app via BLE access. Maybe Polar Beat call setLocalTime during initialization?

choplin, 2021-10-18:
@JOikarinen Thank you for your response. Does it mean that, since there is no method to tell what epoch is set currently in the device as you explained above, once we connect the device to Beat App, we cannot parse the timestamps in BLE packets directly?

closed 2022-12-13 by JOikarinen