polarofficial/polar-ble-sdk #193: Where are UUIDs defined?

KennethEvans, 2021-09-19:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Where are the UUIDs such as 00002a28-0000-1000-8000-00805f9b34fb for firmware in disInformationReceived() defined?

Is there some sort of spec that defines what data like the 4 PPG values, for example, represent? In the literature channels may means red, green  and blue, for example. I gathered from other questions that they are related to LED pairs, presumably opposing LEDs. I don't know what ambient means.

JOikarinen, 2021-09-21:
> Where are the UUIDs such as 00002a28-0000-1000-8000-00805f9b34fb for firmware in disInformationReceived() defined?
- the specification you are looking is found from the Bluetooth organisation web page -> assigned numbers -> 16bit UUIDS:  https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf

> Is there some sort of spec that defines what data like the 4 PPG values, for example, represent? In the literature channels may means red, green and blue, for example. I gathered from other questions that they are related to LED pairs, presumably opposing LEDs.

- PPG signal doesn't have agreed unit. It is raw value, each manufacturer having the own scale for the value. Because of autogain feature in PolarOH1 there can be jumps in PPG signal during measurements. That is discussed in previous issue, see the https://github.com/polarofficial/polar-ble-sdk/issues/152#issuecomment-827700658

> I don't know what ambient means.
- Ambient is the intensity of the light received by the detectors when none of leds in OH1 is emitting light. So, the ambient tells you what is the outer noise caused by external sources of light (e.g. sun, room light etc.) received by the detectors. You  may remove the ambient from each PPG channel to remove external light error from PPG channels. 
- https://github.com/polarofficial/polar-ble-sdk/issues/10#issuecomment-486531057

KennethEvans, 2021-09-21:
Thanks for the information.

As to the UUIDs. That answered my question and gives me a better understanding of what it is doing. The standard Gatt UUIDs are of the form xxxxxxxx-00002a10-0000-1000-8000-00805f9b34fb. Interestingly 00002a28 is named the Software Revision String, whereas the Firmware Revision String is 00002a26.

Your answer about the PPG vales and ambient was good.  It looks like I asked that question before. Sorry. This is a better answer. I also note in that thread that it was mentioned the HR vales in the PPI data were always 0.  That appears to have changed. See Issue #197.

Thanks for your reply.

closed 2021-09-25 by JOikarinen