polarofficial/polar-ble-sdk #194: Is there a way to determine what features a device supports?

KennethEvans, 2021-09-19:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [x] Polar OH1
- [x] Polar Verity Sense
- [x] Polar H10
- [x] Polar H9
- [x] Other
- [ ] Device is not relevant for this question

**Description:**
I am interested in arrhythmias and hence RR (which to me loosely means the interval between heart rate peaks). 

The H10 supports RR intervals and not PPI.  The OH1 and Sense do not support RR but do support PPI. Other PPG devices, such as the Corsense, report PPI as RR, which is convenient since RR is part of the BLE Heart Specification, so the Corsense intervals can be read without the SDK.

How do I determine whether to use RR or PPI. Currently I am branching on whether the name contains OH1 or Sense. I would like something that will work for the next device with a new name.

Thanks.

JOikarinen, 2021-09-21:
@KennethEvans this is good question and causes some confusion from time to time. In general, both RR and PPI represent the time interval between successive heartbeats. However, the difference come from the method the time interval is measured.  

RR:
- time elapsed between two successive R-waves of the QRS signal on the electrocardiogram
- can be measured by Polar sensors with the strap attached on user chest. The R-waves are detected by changes in ecg voltage
- RR cannot be measured by optical heart rate sensors  

PPI:
- pulse to pulse interval is detected from the changes in blood pressure

In Polar we would like to call data as RR when the heart rate interval is really measured from the R-wave. PPI is used when the measurement method is something different, like optical measurement.

KennethEvans, 2021-09-21:
Yes, I know the difference. Technically you are right that it is PPI not RR (also called RRI) in a PPG device. Pragmatically, however, both loosely mean the interval between heart rate peaks. The BLE Heart Spec provides a place for these numbers.

In addition, the HR reported by these devices is some sort of average of the inverse of the RR or PPI intervals, whichever is there. You don't "measure" HR you calculate it from these intervals. They are related to the HR, so including them in the HR spec output could be justified and would be useful. Other devices do this.

There are many programs that can read RR from the BLE Heart Spec. You wouldn't need a Polar-specific SDK, which those programs are not likely to implement.
[image]

KennethEvans, 2021-09-21:
I still have the question is there a way to tell a priori whether I need to handle RR or PPI, other than by branching on whether the name contains OH1 or Sense, which is what I am doing now.

JOikarinen, 2021-09-25:
Yes, there is an option: 

When you initialise the PolarBle API with the [FEATURE_POLAR_SENSOR_STREAMING](https://github.com/polarofficial/polar-ble-sdk/blob/78609b756fe500154b8b129b16ace64bd35c33b9/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L104) (or with [ALL_FEATURES](https://github.com/polarofficial/polar-ble-sdk/blob/78609b756fe500154b8b129b16ace64bd35c33b9/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L112)) you may listen the callback [streamingFeaturesReady](https://github.com/polarofficial/polar-ble-sdk/blob/78609b756fe500154b8b129b16ace64bd35c33b9/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApiCallback.java#L61). On that callback you may decide whether connected device support either [ECG](https://github.com/polarofficial/polar-ble-sdk/blob/78609b756fe500154b8b129b16ace64bd35c33b9/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L55) or [PPI](https://github.com/polarofficial/polar-ble-sdk/blob/78609b756fe500154b8b129b16ace64bd35c33b9/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L58)

I understand your statement, the PPI measured with OH1 or Verity Sense could be added to the standard BLE Heart Rate service, into the RR-Interval Field. Even it is technically possible, it is not that straight forward from data quality perspective. The [answer](https://github.com/polarofficial/polar-ble-sdk/issues/197#issuecomment-927098699) in #197  opens the dilemma a bit. To get reliable PPI there need to be different algorithm running on device to detect PPI reliably.

closed 2021-09-25 by JOikarinen

KennethEvans, 2021-09-25:
I think this issue should remain open for now.

The problem with your solution is that the hrNotification may come in before the streamingFeaturesReady. I need to know whether to use RR or PPI in the hrNotification. There is an item that says if there is RR data available. (I am writing this without having the code available.) It is unclear if this means "not supplied by the device" or just "not in this particular data". (The number of RR values may be 0, and this happens for low HR.)

And then there is the problem that when using the PPI feature, the data are wrong. #197.

I don't understand your explanation of needing reliable PPI. You cannot calculate the HR unless you know the intervals and it is inaccurate if the intervals are unreliable.

Personally this is a problem for me. I have developed an arryrhmia where the RR intervals may vary significantly. I cannot use the H10 as it apparently throws out some RR values as part of its noise reduction algorithms, giving me an abnormally low HR (that is wrong as I can see by counting RR peaks in the H10 ECG). I have to use the OH1 or Sense, but these don't give RR. Apps like Beat don't look at RR anyway, but I have apps that do (if it is reported as RR).

JOikarinen, 2021-09-26:
> I don't understand your explanation of needing reliable PPI. You cannot calculate the HR unless you know the intervals and it is inaccurate if the intervals are unreliable.

You are correct, and I understand your pain here. I will try to explain how the system works currently, however I cannot fully open up the design decisions and algorithm(s) interns as them being Polar intellectual property.

JOikarinen, 2021-09-26:
> The problem with your solution is that the hrNotification may come in before the streamingFeaturesReady. I need to know whether to use RR or PPI in the hrNotification. There is an item that says if [...]

I am not sure does it help in your solution, but the RR-Interval bit on HR Service data is always false for OH1 and Verity Sense. This information is available in [rrAvailable](https://github.com/polarofficial/polar-ble-sdk/blob/78609b756fe500154b8b129b16ace64bd35c33b9/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarHrData.java#L41) field.

KennethEvans, 2021-09-26:
> I am not sure does it help in your solution, but the RR-Interval bit on HR Service data is always false for OH1 and Verity Sense. This information is available in [...]

Ok, now that I know what it is, according to the spec *"The  value  of  the  RR-Interval bit  may  change  while  in  a  connection."* So it just means there are no RR values in this particular packet. It isn't a way to determine the device capability. I'll just stick with checking the name for OH1 or Sense.

It has become moot as the PPI using the SDK are not valid anyway, and I don't see any other reasonable way to get an RR equivalent. The fact they are not valid is a problem, but this is better handled in #197.

Thank you for your help.