polarofficial/polar-ble-sdk #197: Receiving HR=0 that is Almost Certainly Invalid and Other Problems

KennethEvans, 2021-09-19:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [x] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Near the time the first SDK was released I wrote a demo to compare HR and RR for two devices. At one time it was part of the SDK examples. I have not used it for some time, but recently got a Sense and tried comparing it to the OH1 I have been using. There are several problems.

1. There are long periods where the HR returned from the API is zero. (This is based on LOG messages, not any plotting, but the plotting seems to be accurate.) "Long" means several seconds, usually not more than 10. I do not believe this is accurate. A screenshot is attached.  I am recalling sequences of HR = 0 happened with the original app after it was first connected, but not after it had "settled down". I assumed it was valid and an artifact of the OH1. Now it is continuing to have periods of HR = 0 after long periods of time. I now suspect it was not ever an artifact but a bug.

HR = 0 in both the hrNotificationReceived data and the PPI data at about the same time. It seems to be consistent in this respect.

_Note:  It is not obvious, but the line starting with Tot is shows the total of the accumulated RR values compared to the total elapsed time, in case you are curious. They tend to be the same for the OH1 and Sense.  This is not true for some other PPG devices. ;-) It was added for debugging and has not been removed._

When using Beat, as I do regularly, the HR is never 0 (but maybe they are suppressing zeros). 

I have my own BLE app that uses the BLE Heart Spec and works with any device that supports it, which is most of them. It uses BLE directly and does use the SDK. It does not show any HR = 0.  I know HR = 0 is **NOT** suppressed. Attached is a screenshot for 5 min with the Sense. It has no zeros in the 5 minutes. I believe this is the correct output from the device.

2. The status light is blinking magenta (might be considered purple) while it is connected with the SDK. It does not do this when connected with Beat or my app. This is on both the OH1 and the Sense. I do not know what this means.

3. The load for this app is very light. The data come in at about 1 Hz compared to 130 Hz for ECG. I do not believe it is a result of CPU overload.

**How to Reproduce**
This project is on GitHub at https://github.com/KennethEvans/KE.Net-HR-Compare. I am comparing an OH1 and a Sense. It happens every time and with both devices. Since this problem started, I have converted the app to using SDK 3.2.1 instead of 1.1.0. The behavior is the same before and after I converted, but it is not good to submit a bug using some old version. The current version in the repository uses 3.2.1.

**Expected behavior**
HR should seldom, if ever, be zero.

**Screenshots and logs**
If applicable, add screenshots or logs to help explain your problem
[image]
[image]

JOikarinen, 2021-09-25:
Hi @KennethEvans, 

the reason for 0 HR is that when enabling the PPI stream the firmware switches on different heart rate calculation algorithm. The reason why algorithm is changed is that Pulse to Pulse interval (PPI) calculation is much more sensitive for any movements of the device or other possible interference. Algorithm calculates primary the PPI and then from the PPI the heart rate in bpm. That heart rate is then delivered with PPI stream and heart rate service. I believe the consequences of that algorithm switch is why you see problems when using PPI stream via SDK and why you not seeing problems with the Beat app.

KennethEvans, 2021-09-25:
Thank you for the information. I assume the reason for the blinking magenta or purple light is related to the algorithm change. 

The problem is that this algorithm is apparently wrong. It is unlikely my HR is 0 at all and even more so for several seconds. I now consider the output from my compare program to be worthless for PPG devices. Data that lie are worse than no data.

There seem to be problems calculating PPI as mentioned in #194. I don't understand the details at this time. You cannot calculate HR without knowing PPI, and the HR will be no more accurate than the PPI it is based on.

I would be concerned that the HR is even accurate in the OH1 and Sense. However, **when** HR is not zero, the numbers appear to agree, and I have seen at least one study covering multiple devices, presumably using the usual BLE Heart spec, that indicate good agreement with the H10. (I trust the H10.) The problem seems to be with this internal algorithm.

KennethEvans, 2021-10-21:
I will repeat what I have said before: HR for all these devices, ECG or PPG, is not measured directly, it is calculated from measuring intervals.

The Sense appears to give a HR consistent with the H10 and other brands. Thus, in its normal operating mode it is using some PPI that is more accurate than what is coming from the SDK.

It would be nice if that PPI were made available. (I am assuming here the problem is not in the SDK.)

In addition the HR in normal operating mode is not 0. It would be a real problem if that HR were not thought to be reliable.

KennethEvans, 2021-10-30:
I have converted it to using a single instance of PolarBleApiDefaultImpl.defaultImplementation and am using 3.2.2. The logic seems to be working.

I have not seen the purple light on after leaving the HRActivity (the one that uses PolarBleApi). That seems to be fixed.

However after restarting the HRActivity after closing it, it consistently does not connect to the same devices it connected to the first time. (It was doing that before, as well.) I see that mApi is pointing to the same instance as the first time. That is, from the Logcat:
```
HRActivity onDestroy: mAPi=com.polar.sdk.impl.BDBleApiImpl@7b4d788
HRActivity onResume: mAPi=com.polar.sdk.impl.BDBleApiImpl@7b4d788 
```
(It is now doing mApi.disconnectFromDevice in a try/catch in onDestroy.)
mApi is is set this way in onCreate:
```
        mApi = PolarBleApiDefaultImpl.defaultImplementation(this,
                PolarBleApi.FEATURE_BATTERY_INFO |
                        PolarBleApi.FEATURE_DEVICE_INFO |
                        PolarBleApi.FEATURE_HR |
                        PolarBleApi.FEATURE_POLAR_SENSOR_STREAMING
        );
```
On invoking this line after the first time, there is never a deviceConnected callback. 

This was the same thing that was happening with 3.2.1 and having two instances. It seems like something is not being reset after mApi.shutdown(). I'm not seeing anything else I could call to shut it down further.

I have committed and pushed the changes to https://github.com/KennethEvans/KE.Net-HR-Compare.

Unfortunately this issue is really separate from the main topic (Receiving HR=0 that is Almost Certainly Invalid and Other Problems). I'm not sure what to do about that.

JOikarinen, 2021-10-31:
Yep, the [getInstance](https://github.com/polarofficial/polar-ble-sdk/blob/7d64d7d8e690488ca48bdcd75a76359352b4576b/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApiDefaultImpl.kt#L20) is hiding under `defaultImplementation` function.

JOikarinen, 2022-10-24:
Closing as this issue is containing discussion on multiple different aspects of SDK, original problem described is reported internally at Polar

closed 2022-10-24 by JOikarinen