polarofficial/polar-ble-sdk #198: Recording on Polar H10

Mezah, 2021-09-21:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I have a couple of questions related to data recording as this feature is a little bit confusing.

1. I tried to record **HR** and **RR** at the same time using different file names but it only start recording one and the other    
fail with this exception `recording start failed:
com.polar.androidcommunications.api.ble.model.gatt.client.psftp.BlePsFtpUtils$PftpResponseError: Request failed:  Error:
106` Am I doing something wrong? or this operation is not possible.
2. can I start recording without having to delete the existing exercises stored in the device.
I tried to start recording with different exercise ID each time but it fails with the above exception.

The use case I am trying to do is start recording once device is connected and keep recording and when there is a disconnection I will stop recording, load data and start a new record, I was hoping to keep al the recorded files in the device till I can make sure there are saved properly but I couldn't do that.

Thanks in advance

JOikarinen, 2021-09-25:
Hi @Mezah, 

> 1. I tried to record **HR** and **RR** at the same time using different file names but it only start recording one and the other
>    fail with this exception `recording start failed: [...]
- Unfortunately only [one recoding](https://github.com/polarofficial/polar-ble-sdk#:~:text=List%2C%20read%20and%20remove%20for%20stored%20internal%20recording%20(sensor%20supports%20only%20one%20recording%20at%20the%20time)) is possible at the time. 
- Error 106 means OPERATION_NOT_PERMITTED

> 2\. can I start recording without having to delete the existing exercises stored in the device.
>     I tried to start recording with different exercise ID each time but it fails with the above exception.
- similar to above, only one training saved on H10 memory is possible, so the previous training must be removed before starting the new recording

JOikarinen, 2021-09-25:
Did small enhancement on fd2e735, so that error reason is also print instead of just error code.

closed 2021-09-25 by JOikarinen