polarofficial/polar-ble-sdk #200: H10 Heart Rate Sensor does not work properly with the "Polar BLE SDK" app

khangdang4vn, 2021-09-27:
***Platform your question concerns:*** 
- [x] Android (Version 11)
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description: My H10 Heart Rate Sensor does not work properly with the "Polar BLE SDK" app. It can only collect the HR data, not the ECG or ACC data.
I have another device which is Polar Verity Sense Armband. It works well with the mentioned app.
The firmware version is 3.1.1**

JOikarinen, 2021-09-28:
Hi @khangdang4vn,

could you explain a bit more, if you get the data streams working with Polar Verity Sense, I think you shall be very close to get H10 working too with Polar BLE SDK

khangdang4vn, 2021-09-28:
Hi, after clicking the button "Listen Broadcast" on the sample app, I look into the Run log, I can see the HR Broadcast:
....
D/MainActivity: HR BROADCAST 93170422 HR: 69 batt: true
....
But when I try other buttons to read ECG or ACC data. the Run log shows:
....
E/MainActivity: ECG stream failed. Reason java.lang.Throwable: Settings are not available
E/MainActivity: ACC stream failed. Reason java.lang.Throwable: Settings are not available
....

JOikarinen, 2021-09-29:
Hi @khangdang4vn,

thanks for update, now I understood. What you need to do is that you connect first to the device of interest, that can be done either with "Auto connect" or "Connect to ... " button found in example app. 

If you press "Auto connect" then SDK searches for Polar devices and connects to one which is closest (i.e. highest RSSI). If you press "Connect to ..." you may set the correct DeviceId in [MainActivity](https://github.com/polarofficial/polar-ble-sdk/blob/ebd2edbff18314e8ae2ad8a555bb4a35b9bf94e4/examples/example-android/androidBleSdkTestApp/app/src/main/java/com/polar/androidblesdk/MainActivity.kt#L34) the source code.

Example app is the show case of API's in the SDK without much of logic build into example app itself.

closed 2021-09-29 by JOikarinen

khangdang4vn, 2021-09-29:
Hi @JOikarinen,

Thanks for your response. Yes, I did connect my device to the app so I that I can use "Listening to HR Broadcast" (I also set the correct DeviceID too). But other ECG and ACC streaming functions do not work.

Well, I just want to correct one thing that my colleague tried SDK with Polar Verity Sense (it works), but I tried with Polar H10 (it did not). Now I just tried SDK with  Polar Verity Sense, it does not work in the same way with Polar H10. May this be the issue with my phone? I'm using Android version 11.