polarofficial/polar-ble-sdk #201: Polar BLE SDK performs differently between Android version 10 and Android version 11 connecting to Polar Verity Sense

khangdang4vn, 2021-09-29:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Hi, I got a Polar Verity Sense and tried it with the SDK. I reset the sensor back to factory settings. I tried it with 2 phones, Android version 11 and Android version 10. With Android 11, it can connect with the device, do HR broadcast, and read PPI data (but still cannot read other data types such as ACC or ECG). With Android 10, it can't even connect with the device, and also cannot read the HR.

JOikarinen, 2021-09-30:
Hi @khangdang4vn,

let's think about the possible solutions for your observations

> With Android 11, it can connect with the device, do HR broadcast, and read PPI data (but still cannot read other data types such as ACC or ECG)
- ECG type is not supported by Polar Verity Sense, with [example app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp) you shall see something like this in the app logs:
```
2021-09-30 07:24:20.999 11063-11063/com.polar.androidblesdk W/MainActivity: Settings are not available for feature ECG. REASON: com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command error:  failed with error: ERROR_INVALID_MEASUREMENT_TYPE
2021-09-30 07:24:21.994 11063-11198/com.polar.androidblesdk W/MainActivity: Full stream settings are not available for feature ECG. REASON: com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command error:  failed with error: ERROR_INVALID_MEASUREMENT_TYPE
2021-09-30 07:24:22.002 11063-11063/com.polar.androidblesdk E/MainActivity: ECG stream failed. Reason java.lang.Throwable: Settings are not available
```

> With Android 11, it can connect with the device, do HR broadcast, and read PPI data (but still cannot read other data types such as ACC or ECG)
- what happens when you press Start ACC stream? There shall be a screen with settings selection pop up?

> With Android 10, it can't even connect with the device, and also cannot read the HR.
- I tested with Android 10 (Nokia 9), but couldn't observe the problem you mention. The first thing to double check (which happens me daily) is that your Polar Verity Sense is not connected to other phone (in your case it can be the Android 11 phone you mentioned). What is your phone model? I've seen that many phone manufacturers want to optimise the bluetooth stack, which may cause some unexpected behaviours.

khangdang4vn, 2021-11-24:
Hi,

I want to give an update about this issue. After updating Android Studio to the latest version, I can test well with all devices.

Please close this issue. 

Thank you & best regards,

closed 2021-11-25 by JOikarinen