polarofficial/polar-ble-sdk #203: Updated SDK doesnt seem to ever find local devices

astralmedia, 2021-10-11:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [x] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

First off, thank you for the great sdk :) 

We have been using it in an old react-native project for a long time.

Recently we upgraded to the latest everything, including the PolarSDK, and somehow when we hit the `searchForDevice()` method on the latest aar package, it never finds local devices.

We have the permissions setup just right, and we are getting this output from the sdk:

`BDScanCallback/commandState state:SCANNING action: CLIENT_START_SCAN` 

But no data comes back.

We are hitting the method in a try/catch with this:

`Consumer<PolarDeviceInfo> foundDevice = polarDeviceInfo -> {
                Log.d("ReactNative", "DeviceFound wrapper method");
                searchDeviceDataReturned(polarDeviceInfo);
            };
            Consumer<Throwable> errorReturned = errorFound -> {searchDeviceErrorReturned(errorFound);};
            Action complete = () -> {searchDeviceComplete();};

            Disposable scanDisposable = api.searchForDevice().observeOn(AndroidSchedulers.mainThread()).subscribe(
                foundDevice,
                errorReturned,
                complete
            );`

Thank you for any input.

astralmedia, 2021-10-12:
Thanks for getting back to me...

- Local device I mean one within BLE range that should be found by searchForDevice()
- Earlier version was like 2 years ago I think...not sure which # but I did update everything in the codebase and am seeing the SDK itself say its searching.
- How would I get that? I have it spit out the API logs, but only  seeing :

BDScanCallback/commandState state: SCANNING action: ENTRY
BDScanCallback/Scan started --->
BDScanCallback/Scan started <--

I def have FINE & COARSE LOCATION permissions as well

JOikarinen, 2021-11-25:
@astralmedia do your still have this issue holding your development? I am still still interested whether you can get example application working as that would tell us whether the environment (i.e. phone model etc.) is okay and we could narrow down your original problem to something related to your application itself. 

However, suddenly came to my mind that your original problem actually looks a bit like a missing permission. I am interested, what are the permissions requested by your application and what is the target SDK?

closed 2022-12-13 by JOikarinen