polarofficial/polar-ble-sdk #207: Stream PPI : heart rate and rr_interval on Linux with Polar H10

AlexandreDuval, 2021-10-19:
Hi,
I am having issues to request the stream of the PPI to be able to get the heart rate in bpm and the peak to peak interval (rr_interval).

I am coding in Python.

I am able with the ECG using the request bytes :
bytearray([0x02, 0x00, 0x00, 0x01, 0x82, 0x00, 0x01, 0x01, 0x0E, 0x00])

Same with the ACC : bytearray(
[0x02, 0x02, 0x00, 0x01, 0xC8, 0x00, 0x01, 0x01, 0x10, 0x00, 0x02, 0x01, 0x08, 0x00])

On the document, the ECG is 0x00, the ACC is 0x02. The PPI is 0x03.
I have tried it, but no result when reading the data received.

JOikarinen, 2021-10-21:
> Hi, Thank you for the quick answer. I still have an issue.
> 
> The first bytes of the data I receive are 0x00 or 0x02. Witch means the Polar only return me the ECG and ACC.
> 
> Is the stop stream request necessary at the end of the script to prevent this error?

Which Polar device you are making the requests? Are you using Polar H10 device? The PPI is not available in Polar H10, it is only available Polar OH1 and Polar Verity Sense which has the optical measurement. If you are interested of RR interval from Polar H10, then you can listen the standard BLE HR sensor service.

kt-eyescanner, 2022-03-10:
Hi @JOikarinen can you please put up a small code example for how to retrieve real time data on heart rate for the Polar Verity Sense, using Python? I want the data in my computer (not creating app), but I couldn't find any example code regarding this online. Would be really helpful! Thanks :)

JOikarinen, 2022-03-11:
@kt-eyescanner Unfortunately, we are only able to support Android and iOS. The SDK includes source code and you can feel free to utilize / port the code in other platforms too but that is something we are not able to provide support for. 

@kt-eyescanner have you check this https://github.com/pareeknikhil/biofeedback/blob/master/Polar%20Device%20Data%20Stream/Accelerometer/main.py

kt-eyescanner, 2022-03-11:
@JOikarinen, thank you for fast response! I've looked at the Github repository provided and but since I want data from PPG heart rate measurement and not from accelerometer as in the code example I tried to modify the ACC_WRITE. Do you know which array I should use instead of :

ACC_WRITE = bytearray(
    [0x02,
    0x02,
    0x00,
    0x01,
    0x34,
    0x00,
    0x01,
    0x01,
    0x10,
    0x00,
    0x02,
    0x01,
    0x08,
    0x00,
    0x04,
    0x01,
    0x03
    ]
)

And are there other parameters that I should change in order for it to measure PPG instead of ACC?