polarofficial/polar-ble-sdk #210: Xcode 12.5.1 fails to import Polar SDK 3.2.2

kononov-serg, 2021-11-09:
**Platform on which you observed the bug:** 
iOS

**Device on which you observed the bug:**
Device is not relevant for this bug

**Describe the bug**
I'm using XCode version 12.5.1, as it is recommended for the polar sdk 3.2.2, and I was trying to add polar library to my react native project. I added dependency to the pod file:

`require File.join(File.dirname(`node --print "require.resolve('expo/package.json')"`), "scripts/autolinking")
require File.join(File.dirname(`node --print "require.resolve('react-native/package.json')"`), "scripts/react_native_pods")
require File.join(File.dirname(`node --print "require.resolve('@react-native-community/cli-platform-ios/package.json')"`), "native_modules")

platform :ios, '12.0'

require 'json'
podfile_properties = JSON.parse(File.read('./Podfile.properties.json')) rescue {}

target 'mhmobile' do
  pod 'PolarBleSdk', '3.2.2'
  
  use_expo_modules!
  config = use_native_modules!
...`

and just added a file with polar sdk import:
`import Foundation
import PolarBleSdk
import RxSwift
import CoreBluetooth

/// PolarBleSdkManager demonstrates how to user PolarBleSDK API
class PolarBleSdkManager : ObservableObject { 
...`

During  project building, error occures with Swift version mismatch. I tried different versions of Xcode: 13.1, 13.0 (this on has required Swift version), and every time a has errors in build time.

**How to Reproduce**
If you know how the bug may be reproduced please add the steps here

**Screenshots and logs**
Failed to build module 'PolarBleSdk' from its module interface; the compiler that produced it, 'Apple Swift version 5.5 (swiftlang-1300.0.31.1 clang-1300.0.29.1)', may have used features that aren't supported by this compiler, 'Apple Swift version 5.4.2 (swiftlang-1205.0.28.2 clang-1205.

JOikarinen, 2021-11-09:
@kononov-serg PolarBleSdk is compiled using the [Swift 5.5.1](https://www.swift.org/download/) version. The warning says that your Swift version is 5.4.2. I recommend you update the Xcode to version 13.

kononov-serg, 2021-11-10:
@JOikarinen I have updated Xcode to version 13.1, now I have following error during project build:

Failed to build module 'PolarBleSdk'; this SDK is not supported by the compiler (the SDK is built with 'Apple Swift version 5.5 (swiftlang-1300.0.31.1 clang-1300.0.29.1)', while this compiler is 'Apple Swift version 5.5.1 (swiftlang-1300.0.31.4 clang-1300.0.29.6)'). Please select a toolchain which matches the SDK.

JOikarinen, 2021-11-10:
@kononov-serg sorry for the wrong information. The previous release of Polar BLE SDK was obviously build with Swift 5.5, as you observed. I created new [release](https://github.com/polarofficial/polar-ble-sdk/releases/tag/3.2.4) which is now build with Swift 5.5.1.  The new release shall will fix the problem.

closed 2021-11-10 by JOikarinen

kononov-serg, 2021-11-10:
@JOikarinen thanks for the quick response! When it will be available in the pod repository?

JOikarinen, 2021-11-11:
@kononov-serg should be available in pod repository: https://github.com/CocoaPods/Specs/commit/96a406a1bea9e7bc1cac9befca476dc7b1b2a1da

Are you able to get it?

kononov-serg, 2021-11-11:
@JOikarinen I tried to update dependency in the pod file to the new verion:

```
require File.join(File.dirname(`node --print "require.resolve('expo/package.json')"`), "scripts/autolinking")
require File.join(File.dirname(`node --print "require.resolve('react-native/package.json')"`), "scripts/react_native_pods")
require File.join(File.dirname(`node --print "require.resolve('@react-native-community/cli-platform-ios/package.json')"`), "native_modules")

platform :ios, '12.0'

require 'json'
podfile_properties = JSON.parse(File.read('./Podfile.properties.json')) rescue {}

target 'mhmobile' do
  
  pod 'PolarBleSdk', '3.2.4'
  
  use_expo_modules!
  config = use_native_modules!
```

but `pod install` failed with the following error:

```
[!] CocoaPods could not find compatible versions for pod "PolarBleSdk":
  In Podfile:
    PolarBleSdk (= 3.2.4)

None of your spec sources contain a spec satisfying the dependency: `PolarBleSdk (= 3.2.4)`.
```

Then I tried to point dependency to the git, as followig:

`pod 'PolarBleSdk', :git => 'https://github.com/polarofficial/polar-ble-sdk.git'`

It worked, problem with library import has gone, but the new error appeared during application start:
[image]

Here are some logs:

```
dyld[40944]: Library not loaded: @rpath/RxSwift.framework/RxSwift
  Referenced from: /Users/-/Library/Developer/CoreSimulator/Devices/8A148C46-CD3C-4FA5-A3F1-1349CA0716F9/data/Containers/Bundle/Application/5E89B01C-78F5-4062-A08A-1B44B2EBA6F1/test.app/Frameworks/PolarBleSdk.framework/PolarBleSdk
  Reason: tried: '/Users/-/Library/Developer/Xcode/DerivedData/test-etpublyqwguvxleghwdaemaduiyq/Build/Products/Debug-iphonesimulator/RxSwift.framework/RxSwift' (no such file), '/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot/usr/lib/swift/RxSwift.framework/RxSwift' (no such file), '/usr/lib/swift/RxSwift.framework/RxSwift' (no such file), '/Users/-/Library/Developer/CoreSimulator/Devices/8A148C46-CD3C-4FA5-A3F1-1349CA0716F9/data/Containers/Bundle/Application/5E89B01C-78F5-4062-A08A-1B44B2EBA6F1/test.app/Frameworks/RxSwift.framework/RxSwift' (no such file), '/Users/-/Library/Developer/CoreSimulator/Devices/8A148C46-CD3C-4FA5-A3F1-1349CA0716F9/data/Containers/Bundle/Application/5E89B01C-78F5-4062-A08A-1B44B2EBA6F1/test.app/Frameworks/PolarBleSdk.framework/Frameworks/RxSwift.framework/RxSwift' (no such file), '/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot/usr/lib/swift/RxSwift.framework/RxSwift' (no such file), '/usr/lib/swift/RxSwift.framework/RxSwift' (no such file), '/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot/usr/lib/swift/RxSwift.framework/RxSwift' (no such file), '/usr/lib/swift/RxSwift.framework/RxSwift' (no such file), '/Users/-/Library/Developer/CoreSimulator/Devices/8A148C46-CD3C-4FA5-A3F1-1349CA0716F9/data/Containers/Bundle/Application/5E89B01C-78F5-4062-A08A-1B44B2EBA6F1/test.app/Frameworks/RxSwift.framework/RxSwift' (no such file), '/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot/usr/lib/swift/RxSwift.framework/RxSwift' (no such file), '/usr/lib/swift/RxSwift.framework/RxSwift' (no such file), '/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot/usr/lib/swift/RxSwift.framework/RxSwift' (no such file), '/usr/lib/swift/RxSwift.framework/RxSwift' (no such file), '/Users/-/Library/Developer/CoreSimulator/Devices/8A148C46-CD3C-4FA5-A3F1-1349CA0716F9/data/Containers/Bundle/Application/5E89B01C-78F5-4062-A08A-1B44B2EBA6F1/test.app/Frameworks/RxSwift.framework/RxSwift' (no such file), '/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot/System/Library/Frameworks/RxSwift.framework/RxSwift' (no such file)
Library not loaded: @rpath/RxSwift.framework/RxSwift
  Referenced from: /Users/-/Library/Developer/CoreSimulator/Devices/8A148C46-CD3C-4FA5-A3F1-1349CA0716F9/data/Containers/Bundle/Application/5E89B01C-78F5-4062-A08A-1B44B2EBA6F1/test.app/Frameworks/PolarBleSdk.framework/PolarBleSdk
  Reason: tried: '/Users/-/Library/Developer/Xcode/DerivedData/test-etpublyqwguvxleghwdaemaduiyq/Build/Products/Debug-iphonesimulator/RxSwift.framework/RxSwift' (no such file), '/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Library/Developer/CoreSimulator/Profiles/Runtimes/iOS.simruntime/Contents/Resources/RuntimeRoot/usr/lib/swift/RxSwift.framework/RxSwift' (no such file), '/usr/lib/swift/RxSwift.framework/RxSwift' (no such file), '/Users/-/Library/Developer/CoreSimulator/Devices/8A148C46-CD3C-4FA5-A3F1-1349CA0716F9/data/Containers/Bundle/Application/5E89B01C-78F5-4062-A08A-1B44B2EBA6F1/test.app/Frameworks/RxSwift.framework/RxSwift' (no such file), '/Users/-/Library/Developer/Core
```
 Can this error be related to the PolarSdk?

I tried to add `pod 'RxSwift'` dependency to the pod file, but it didn't help.

JOikarinen, 2021-11-12:
When I to try reproduce your problem, I follow the steps like these:

1. I have defined the Podfile like this:
```
source 'https://github.com/CocoaPods/Specs.git'

platform :ios, '12.0'
use_frameworks!

target 'iosBleSdkTestApp' do
    pod 'PolarBleSdk', '3.2.4'
end

post_install do |installer|
        installer.pods_project.targets.each do |target|
          target.build_configurations.each do |config|
            config.build_settings['BUILD_LIBRARY_FOR_DISTRIBUTION'] = 'YES'
          end
        end
      end
```
    - the post_install is needed. It is explained here https://github.com/polarofficial/polar-ble-sdk#cocoapods

2. I run the pod install:
`pod update PolarBleSdk`

Comment:
- the [ PolarBleSdk 3.2.4](https://github.com/CocoaPods/Specs/commit/96a406a1bea9e7bc1cac9befca476dc7b1b2a1da) podspec file defines the RxSwift dependency. So there shouldn't be a need to add it separately.

kononov-serg, 2021-11-13:
@JOikarinen Thank you very much for the help! I will try to apply your advices

haianh230797, 2021-11-30:
same isssue, i got the error

> When I to try reproduce your problem, I follow the steps like these:
>
> 1. I have defined the Podfile like this:
>
> ```
> source 'https://github.com/CocoaPods/Specs.git'
>
> platform :ios, '12.0' [...]

[image]

I still get this error after applying the above, Pls support

kononov-serg, 2021-11-30:
@haianh230797 in my case the solution was to add xcframeworks to my project, as it is suggested here - https://github.com/polarofficial/polar-ble-sdk#xcframeworks

haianh230797, 2021-12-01:
> @haianh230797 in my case the solution was to add xcframeworks to my project, as it is suggested here - https://github.com/polarofficial/polar-ble-sdk#xcframeworks

thanks a lot, im using Xcode 12.4 and i used SDK v.3.2.1 then it work.
because after v.3.2.2, xcode needs to be raised to v.13

kononov-serg, 2021-12-01:
I'm using XCode v13.1 and Polar SDK v3.2.4