polarofficial/polar-ble-sdk #211: PPG Sample Rate and PPI Time Stamp

CapelaA10, 2021-11-11:
***Platform your question concerns:*** 
- [√ ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ √] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
The documentation says that the PPG sample rate can be fixed by enabling sdk mode, enabling or not the sdk mode the sample rate is always 55Hz.
Without sdk mode, other question is that the PPI stream always returns a 0 in the timeStamp, should that be a bug?

JOikarinen, 2021-11-12:
Hi @CapelaA10,

I used the [Example application](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp) for repeating your steps. This is what I did:
- Connect to Polar Verity Sense
- Start PPG stream, Selected sampling rate 135Hz
- Observed the received samples are sampled with 55Hz frequency. This is the [known issue](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/KnownIssues.md#polar-verity-sense). 
- Stop PPG Stream
- Enable SDK Mode 
- Start PPG stream, Selected sampling rate 135Hz
- Observed the received samples are sampled with 135Hz frequency.

I added this code to calculate the sample rate in subscription of PPG stream:

```
     val previousTimeStamp = 0;
     ...
     { polarOhrPPGData: PolarOhrData ->
                                val delta = polarOhrPPGData.timeStamp - previousTimeStamp
                                if (delta > 0) {
                                    val sampleDelta = delta / polarOhrPPGData.samples.size
                                    val sampleRrte = 1.toDouble() / sampleDelta * 1000 * 1000 * 1000
                                    Log.d(TAG, "Calculated sample rate: $sampleRrte")
                                }
                                previousTimeStamp = polarOhrPPGData.timeStamp
                                

```

With these steps I couldn't reproduce the problem you described. Am I probably missing some steps?

> Without sdk mode, other question is that the PPI stream always returns a 0 in the timeStamp, should that be a bug?
- this is because PPI data doesn't have the sampling frequency, it is the Peak-to-Peak interval. So the timestamp  [timeStamp](https://github.com/polarofficial/polar-ble-sdk/blob/40ae613f121f03ad9c1ae04ba8c8167a9bbe11cc/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarOhrPPIData.java#L62) is not making sense in PPI data. The timeStamp shall be deprecated from PPI data stream, I will take a look.

CapelaA10, 2021-11-12:
Hey @JOikarinen, thank you for the fast feedback.
Makes sense for PPI TimeStamp.
For the PPG I will make the same steps you made, then I will get back to you.

CapelaA10, 2021-11-15:
Hey @JOikarinen.
After testing I see the sample rate at 135hz with SDK mode enable.
Is possible to have the sdk mode enable and receive PPI at the same time? I saw the docs and it says "Not supported" any plans to be supported?

JOikarinen, 2021-11-16:
> After testing I see the sample rate at 135hz with SDK mode enable.
- 👍  Okay good. 

> Is possible to have the sdk mode enable and receive PPI at the same time? I saw the docs and it says "Not supported" any plans to be supported?
- unfortunately that is not possible and currently there is no plans to change the functionality on this.

closed 2021-11-16 by JOikarinen