polarofficial/polar-ble-sdk #212: Stack Overflow polar.com.sdk.impl.BDBleApiImpl$1.d(BDBleApiImpl.java:170)

klandell, 2021-11-15:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [X] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
App using Polar SDK crashes on Android 12.  I have been using this SDK on an Android 11 device for awhile now with no issues, but running the same app after upgrading to Android 12 causes a stack overflow error when attempting to auto connect to a device.

**How to Reproduce**
From the stack trace, the exception appears to be thrown after calling this block of code.

```JavaScript
public void connect() {
    if (autoConnectDisposable == null) {
        autoConnectDisposable = polarApi.autoConnectToDevice(-70, "180D", null)
                .subscribe(
                        () -> {
                            autoConnectDisposable = null;
                        },
                        throwable -> {
                            autoConnectDisposable = null;
                        }

                );
    }
}
```

**Screenshots and logs**
Caused by java.lang.StackOverflowError: stack size 1039KB
       at java.lang.AbstractStringBuilder.append(AbstractStringBuilder.java:449)
       at java.lang.StringBuilder.append(StringBuilder.java:137)
       at polar.com.sdk.impl.BDBleApiImpl$1.d(BDBleApiImpl.java:170)
       at com.androidcommunications.polar.api.ble.BleLogger.d(BleLogger.java:44)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.startLScan(BDScanCallback.java:329)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.startScanning(BDScanCallback.java:316)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.scannerScanningState(BDScanCallback.java:225)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.commandState(BDScanCallback.java:132)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.changeState(BDScanCallback.java:141)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.scannerIdleState(BDScanCallback.java:149)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.commandState(BDScanCallback.java:124)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.changeState(BDScanCallback.java:141)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.callStartScanL(BDScanCallback.java:364)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.startLScan(BDScanCallback.java:337)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.startScanning(BDScanCallback.java:316)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.scannerScanningState(BDScanCallback.java:225)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.commandState(BDScanCallback.java:132)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.changeState(BDScanCallback.java:141)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.scannerIdleState(BDScanCallback.java:149)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.commandState(BDScanCallback.java:124)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.changeState(BDScanCallback.java:141)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.callStartScanL(BDScanCallback.java:364)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.startLScan(BDScanCallback.java:337)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.startScanning(BDScanCallback.java:316)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.scannerScanningState(BDScanCallback.java:225)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.commandState(BDScanCallback.java:132)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.changeState(BDScanCallback.java:141)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.scannerIdleState(BDScanCallback.java:149)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.commandState(BDScanCallback.java:124)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.changeState(BDScanCallback.java:141)
       at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.callStartScanL(BDScanCallback.java:364)
[... 79 lines cut ...]
       at io.reactivex.rxjava3.internal.operators.flowable.FlowableFilter.subscribeActual(FlowableFilter.java:37)
       at io.reactivex.rxjava3.core.Flowable.subscribe(Flowable.java:15747)
       at io.reactivex.rxjava3.internal.operators.flowable.FlowableMap.subscribeActual(FlowableMap.java:38)
       at io.reactivex.rxjava3.core.Flowable.subscribe(Flowable.java:15747)
       at io.reactivex.rxjava3.internal.operators.flowable.FlowableTakeUntilPredicate.subscribeActual(FlowableTakeUntilPredicate.java:33)
       at io.reactivex.rxjava3.core.Flowable.subscribe(Flowable.java:15747)
       at io.reactivex.rxjava3.core.Flowable.subscribe(Flowable.java:15693)
       at io.reactivex.rxjava3.internal.operators.flowable.FlowableReduceSeedSingle.subscribeActual(FlowableReduceSeedSingle.java:50)
       at io.reactivex.rxjava3.core.Single.subscribe(Single.java:4813)
       at io.reactivex.rxjava3.internal.operators.single.SingleDoOnSuccess.subscribeActual(SingleDoOnSuccess.java:35)
       at io.reactivex.rxjava3.core.Single.subscribe(Single.java:4813)
       at io.reactivex.rxjava3.internal.operators.single.SingleToObservable.subscribeActual(SingleToObservable.java:35)
       at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13095)
       at io.reactivex.rxjava3.internal.operators.observable.ObservableIgnoreElementsCompletable.subscribeActual(ObservableIgnoreElementsCompletable.java:31)
       at io.reactivex.rxjava3.core.Completable.subscribe(Completable.java:2851)
       at io.reactivex.rxjava3.internal.operators.completable.CompletableAndThenCompletable$SourceObserver.onComplete(CompletableAndThenCompletable.java:67)
       at io.reactivex.rxjava3.internal.operators.completable.CompletableCreate$Emitter.onComplete(CompletableCreate.java:65)
       at polar.com.sdk.impl.BDBleApiImpl.lambda$autoConnectToDevice$4(BDBleApiImpl.java:363)
       at polar.com.sdk.impl.-$$Lambda$BDBleApiImpl$FNAJbH9-3kL_wPIEdVw9YLVVv_c.subscribe(:2)
       at io.reactivex.rxjava3.internal.operators.completable.CompletableCreate.subscribeActual(CompletableCreate.java:40)
       at io.reactivex.rxjava3.core.Completable.subscribe(Completable.java:2851)
       at io.reactivex.rxjava3.internal.operators.completable.CompletableAndThenCompletable.subscribeActual(CompletableAndThenCompletable.java:35)
       at io.reactivex.rxjava3.core.Completable.subscribe(Completable.java:2851)
       at io.reactivex.rxjava3.core.Completable.subscribe(Completable.java:2926)
       at edu.mssm.hpims.ehive.Polar.connect(Polar.java:293)

JOikarinen, 2021-11-16:
@klandell thanks for the report. I am able to reproduce the problem. Let's investigate.

JOikarinen, 2021-11-16:
@klandell there is [new permissions for Bluetooth](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions) introduced for Android 12 devices. Now when you change your app target to SDK 31 those new permissions need to be granted by user when your app is running on Android 12 device.

The needed changes are now made to [example application](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp). You may check the commit https://github.com/polarofficial/polar-ble-sdk/commit/6b0904534e5b38abbf277004c8c96d533ab4e8ae

I will update the ReadMe also.

JOikarinen, 2021-11-16:
Did little update to example application here https://github.com/polarofficial/polar-ble-sdk/commit/bec621b91cdc1e9f8d6d601a523777f026dc4be5

Also updated the ReadMe. 

@klandell I will close this issue, please feel free to reopen if the above comments doesn't solve your problem.

closed 2021-11-16 by JOikarinen

klandell, 2021-11-16:
Thank you for the extremely quick turnaround as always. The updated permissions solve the issue for me.