polarofficial/polar-ble-sdk #213: Read PPG data from Polar OH1

lonelyzerg, 2021-11-19:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x] Other
- [ ] Platform is not relevant for this question

***Device:***
- [x] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

I' trying to read PPG data from my Polar OH1 sensor using Python bleak libarary and get no response. Basically I am writing `bytearray([0x02, 0x01, 0x00, 0x01, 0x82, 0x00, 0x01, 0x01, 0x0E, 0x00])` to PMD Control `"FB005C81-02E7-F387-1CAD-8ACD2D8DF0C8"`, and listen to PMD Data `"FB005C82-02E7-F387-1CAD-8ACD2D8DF0C8"` using `await client.start_notify(PMD_DATA, data_conv)` where data_conv is the callback function. I get no response from the sensor.

After reading the [documentation](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf) I found that  the result of reading PMD Control is `bytearray(b'\x0f\x0e\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')`.

 According to the documentation, 2nd byte represent supported measurement type, and instead of 0x6e in the example of Ploar Verity Sense, my Ploar OH1 gives me 0x0e. If they represent the same thing, this means my sensor does not support ecg/ppg/ppi/acc.

I wonder if there is any problem with firmware so I tried to read the firmware field ("00002A26-0000-1000-8000-00805f9b34fb"). The result is 1.3.9 but I do not know what is the latest firmware version.

It would be really nice if someone can explain to me how reading from the sensor works.

JOikarinen, 2021-11-22:
Hi @lonelyzerg, 
I have some ideas for you questions. Let's get started

> I am writing `bytearray([0x02, 0x01, 0x00, 0x01, 0x82, 0x00, 0x01, 0x01, 0x0E, 0x00])`
- start PPG stream request looks correct, except the resolution. The resolution 14 (0x0E) is not supported by Polar OH1. As a consequence the response returned by the Polar OH1 indicates the error. Response to your request should look like this `HEX: F0 02 01 05 00`
0xF0 = Response code, 0x02 = Start stream op code, 0x01 = Measurement type PPG, 0x05 = Error code (5 = ERROR_INVALID_PARAMETER). To enhance your code, you may listen result for the request to validate the request you made was correct. 
- the supported resolution for PPG stream in Polar OH1 is 22 (0x16). You may try this `bytearray([0x02, 0x01, 0x00, 0x01, 0x82, 0x00, 0x01, 0x01, 0x16, 0x00])` Future more, to enhance your code, you could read the settings before requesting the stream, that way you can be sure you are using the settings supported by the device. For example if you see the [example app](https://github.com/polarofficial/polar-ble-sdk/blob/dcad1148853a12e23ad52ed045313cbf45753cf2/examples/example-android/androidBleSdkTestApp/app/src/main/java/com/polar/androidblesdk/MainActivity.kt#L357), you see that each of the stream start request is preceded by settings request. 

> After reading the [documentation](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf) I tried to read PMD Control. The result is `bytearray(b'\x0f\x0e\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')`.
- the result looks correct. 0x0E = 1110b => bit0 = ECG is not supported, bit1 = PPG supported, bit2 = ACC supported, bit3 = PPI supported

> The result is 1.3.9 but I do not know what is the latest firmware version.
- the latest firmware is [2.1.20](https://support.polar.com/en/updates/polar-oh1-firrmware-update-2120?product=OH1). Unfortunately the SDK doesn't have support for firmware update, so you shall do it following these instructions https://support.polar.com/en/how-can-i-update-polar-oh1

lonelyzerg, 2021-11-22:
Thank you! After changing the resolution I am able to get the data I want from the sensor. However I'm not really clear on how to request stream and most of my knowledge are from reading code writen by others. Is there any documentation on the definition of each byte when writing to PMD control point?

lonelyzerg, 2021-11-22:
@JOikarinen One more question. Is there any way to read saved data recorded by polar oh1 as an independent trainings device? I read from Polar website that there are apps that sync the recorded data, however I need to gather samples from multiple participants at the same time so I will need an efficient way to read saved data from the sensor. Thank you!

JOikarinen, 2021-11-23:
> Is there any documentation on the definition of each byte when writing to PMD control point and their valid values?

The [document](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf) you referenced earlier is the one available at the moment.

JOikarinen, 2021-11-23:
> I read from Polar website that there are apps that sync the recorded data
- the [Polar Flow app](https://support.polar.com/en/support/Flow_app) does sync the training devices with Polar ecosystem. It can register the Polar device with Polar Flow ecosystem, then read and write data with training device.
>  Is there any way to read saved data recorded by polar oh1 as an independent trainings device?
- unfortunately no such documentation or support available at the moment

lonelyzerg, 2021-11-28:
Thank you for your reply! I wonder if PPG and/or PPI data are also saved in the sensor? If so is there anyway I can download the data after syncing the sensor?

JOikarinen, 2021-11-29:
> I wonder if PPG and/or PPI data are also saved in the sensor? If so is there anyway I can download the data after syncing the sensor?
- Unfortunately there is no feature in Polar OH1 or Polar Verity Sense you could save PPG or PPI data in there sensor memory

jimmyzumthurm, 2023-09-08:
> Thank you for your reply! I wonder if PPG and/or PPI data are also saved in the sensor? If so is there anyway I can download the data after syncing the sensor?

Hi @lonelyzerg . FYI, we have now support of recording raw data (PPG, ACC, GYRO, MAG, PPI, HR) inside device memory to be retrieved later for Verity Sense with version > 2.1.0 and Polar SDK version > 5, see doc : https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/OfflineRecordingExplained.md Unfortunately this can't be supported for OH1, as it has really small memory compared to Verity Sense, so not much data could be recorded inside it.

closed 2023-09-08 by jimmyzumthurm