polarofficial/polar-ble-sdk #215: How to activate SDK mode on Verity Sense

cjs30, 2021-12-07:
I am writing a Javascript app to display PPG from the Polar Verity Sense simultaneously with ACC and ECG data from the H10 - I am not using the Polar API/SDK but querying the UUIDs using the Web Bluetooth API. All of the H10 channels work as expected, but I am having issues with the Verity Sense. I have got as far as parsing the Delta frames but the sampling rate is all-over-the place and sensitive to movement across the sensor (i.e. more Packets seem to be returned when there is movement across the device). I could be making a mistake, or this could be the 'known issue'. The solution, presumably, is to use the SDK mode. But, how do I activate that? Is there anyone that can offer advice?
The project is to calculate blood pressure using the timing difference between the ECG or ballistocardiogram using the accelerometer on the H10 and the PPG from the Verity Sense in an appropriate location. The pulse wave velocity is dependent on blood pressure (amongst other things).

JOikarinen, 2021-12-08:
Hi @cjs30, your project sounds very interesting. Let me try to answer your questions. 

> I have got as far as parsing the Delta frames but the sampling rate is all-over-the place and sensitive to movement across the sensor (i.e. more Packets seem to be returned when there is movement across the device). I could be making a mistake, or this could be the 'known issue'.

If your Verity Sense has firmware 1.1.5 it has the [known issue](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/KnownIssues.md#polar-verity-sense) related to PPG sampling rate. The PPG stream settings returns incorrect sampling rate when settings are read in normal operation mode. The returned value for sampling rate is 135Hz, the returned sampling rate should be 55Hz. Even the requested sampling rate is 135Hz while starting the PPG stream the actual sampling rate is 55Hz. 

> In essence, what does enableSDKMode actually do in terms of writing to the Verity?
- once SDK mode is enabled the Verity Sense closes all measurements, it has currently running in normal operation mode (e.g. optical heart rate measurement). This makes it possible to use wider range of sensor capabilities (e.g. sampling PPG with wider range of sampling options). More explanation here https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/SdkModeExplained.md
- if your question is what the [enableSDKMode](https://github.com/polarofficial/polar-ble-sdk/blob/6a9896f4676e648c8e1441d67d830ce471bef821/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L478) function does under the hoods:
   - to enable the SDK mode you sent command [0x02, 0x09] where the 0x02 = Request Measurement start and 0x09 = SDK Mode
   - to disable the SDK mode you sent command [0x03, 0x09] where the 0x03 = Request Measurement stop and 0x09 = SDK Mode
   - Verity Sense shall respond [0xF0, 0x02, 0x09, 0x00, 0x00] 0xF0 = control point response, 0x02 = op_code(Start Measurement) or 0x03 = op_code(Stope Measurement) ,0x09 = measurement_type(SDK Mode), 0x00 = error_code(success), 0x00 = more_frames(false)
- if you requested the PPG settings in SDK mode then returned PPG sampling rate values are correct (i.e. no known issues in SDK mode) and you may select the sampling rate which best matches your use case.

cjs30, 2021-12-08:
Excellent! The 'under the hood' bit is exactly what I wanted. I will give it a go. Many thanks.

JOikarinen, 2021-12-08:
@cjs30 please feel free to reopen the issue for more question. I will close it for now.

closed 2021-12-08 by JOikarinen

cjs30, 2021-12-09:
@JOikarinen I just wanted to thank you again for your rapid reply. I now have the SDK mode working correctly on the Verity with PPG at maximum speed and the trace is fantastic. In the end, the best signal seemed to be the sum of the three PPG channels (not ambient) and then throw it through a high pass filter (timeconstant 1s with no low pass filtering required). The delta frame parsing routine is a bit convoluted with the logic being a conversion into an unsigned half-nibble array which can be flexibly parsed with even-value bit lengths. I had written a similar bit of software with a MAX30100 on a pi but, this is much better. Still a lot of work to do, but many many thanks. I will share the HTML/javascript file once it is reading both the H10 and Verity simultaneously with the ACCs used to get precise timing alignment.
[image]

ghelmfelt, 2022-12-09:
Hello, 
Have any of you been working with Polar Verity in Python?? I worked a lot with the Polar OH1 but I understand that having more sensors will have some changes.
Thank you very much in advance for any information you have.