polarofficial/polar-ble-sdk #216: How to return Polar data values (shown on run log) from a Disposable/Observable to use in other functions

khangdang4vn, 2021-12-08:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

Hello, do you mind me asking how I can actually get the Polar data values as a variable and use it in other functions (like to actually save the data to the phone's storage).

Specifically, I'm writing a function to store Polar HR data in an ArrayList<Int> and return it as an Observable.

`private fun createpolarhrListObservable(): Observable<ArrayList<Int>> {
        return create { emitter ->
            if (broadcastButtonclicked) {
                broadcastDisposable = api.startListenForPolarHrBroadcasts(null)
                    .subscribe(
                        { polarBroadcastData: PolarHrBroadcastData ->
                            polarhrList += polarBroadcastData.hr
                            emitter.onNext(polarhrList)
                            Log.d(
                                TAG,
                                "HR BROADCAST ${polarBroadcastData.polarDeviceInfo.deviceId} " +
                                        "HR: ${polarBroadcastData.hr} " +
                                        "batt: ${polarBroadcastData.batteryStatus} " +
                                        "polarhrListCollected: $polarhrList " +
                            )
                        },
                        { error: Throwable ->
                            Log.e(TAG, "Broadcast listener failed. Reason $error")
                        },
                        {
                            Log.d(
                                TAG, "complete" +
                                        "publish: $polarhrList "
                            )
                        }
                    )
            } else {
                Log.e(TAG, "Need to turn on Broadcast Button")
            }
        }
    }
`

Then I create a publish subject (AsyncSubject) then subscribe it to the Observable to get the values as ArrayList<Int>.

`private var _polarhrList = AsyncSubject.create<ArrayList<Int>>()`
`fun createpolarhrList(): ArrayList<Int> {
            createpolarhrListObservable().subscribe(_polarhrList)
            _polarhrList.onComplete()
            val myList: ArrayList<Int> = _polarhrList.getValue()
            Log.d(
                TAG,
                "publish: $myList "
            )
            return myList
    }`

But at this step, "_polarhrList.getValue()" returns null values.

Could you please guide me on how to get values from Polar streaming/broadcast data and use them in other functions to save them to the phone's internal storage?

I would appreciate it very much!

Thank you!

JOikarinen, 2021-12-08:
Hi @khangdang4vn, 

in general Rx is very flexible for different kinds of options. Let's discuss a bit. 

> Could you please guide me on how to get values from Polar streaming/broadcast data and use them in other functions to save them to the phone's internal storage?
- the most simple approach, pseudo code: 
```
 // open the file stream
 val file = openFile("somefile.txt")
 val broadcastDisposable = api.startListenForPolarHrBroadcasts(null)
       .doFinally {
           // close the file once stream is either completed, error has stop the stream or stream is disposed
          file.close()
      }
     .subscribe(
        { broadcastData: PolarHrBroadcastData ->
           // write to the file
           file.write(broadcastData)
        }
      )
```

- if I got your code snippet correctly, the idea is that your collect the items into the array as long as stream emits. Once the stream is completed then all the items collected to the array is handled (in your case items are saved to the file). You can make it work, but conceptually I wouldn't recommend the approach. In theory the broadcast steam is infinite stream, SDK won't ever complete it. As a consequence you may run out of memory as the stream never completes (of course only in theory). In practice the stream is completed by error when BLE disconnects, by other error or you dispose the stream. However, I recommend that you process the data once it received (see the above pseudo code).

khangdang4vn, 2021-12-08:
Hi @JOikarinen,

Thank you very much for your response. Yes, you got the idea from my code correctly. I tried to process the data in a similar way to the provided pseudocode.

First, I created a private var _polarhrMeasurements to collect the items into HashMap. (like open the file stream in pseudocode)
```
private var _polarhrMeasurements = HashMap<String, ArrayList<Int>>()
```
Then, inside the function onCreate, when users click the broadcast button, the items start to be stored in an ArrayList polarhrList in subscribe() (like write into the file in the pseudocode). And when the stream is complete in doFinally(), the HashMap will get the Arraylist (like close the file in pseudocode)

```
override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        this.window.setSoftInputMode(WindowManager.LayoutParams.SOFT_INPUT_STATE_ALWAYS_HIDDEN)
        setContentView(R.layout.activity_polar)
        Log.d(TAG, "version: " + PolarBleApiDefaultImpl.versionInfo())
        broadcastButton = findViewById(R.id.broadcast_button)
        broadcastButton.setOnClickListener {
            if (!this::broadcastDisposable.isInitialized || broadcastDisposable.isDisposed) {
                toggleButtonDown(broadcastButton, R.string.listening_broadcast)
                 broadcastDisposable = api.startListenForPolarHrBroadcasts(null)
                     .doFinally{
                         // collect the items once stream is either completed, error has stop the stream or stream is disposed
                         _polarhrMeasurements[POLAR_HR] = polarhrList
                         Log.d(
                             TAG,
                             "doFinally: $_polarhrMeasurements "
                         )
                     }
                    .subscribe(
                        { polarBroadcastData: PolarHrBroadcastData ->
                            polarhrList += polarBroadcastData.hr
                        },
                    )
            }
        }
}
```
However, when I create a function to return the HashMap, it doesn't have any values.
```
fun getPolarhrMeasurements(): HashMap<String, ArrayList<Int>> {
        Log.d(
            TAG,
            "getPolarhrMeasurements: $_polarhrMeasurements "
        )
        return _polarhrMeasurements
    }
```

JOikarinen, 2021-12-09:
@khangdang4vn some ideas what may have happened. 

- Do you call `getPolarhrMeasurements()` before the `broadcastDisposable` is disposed? As a consequence the  `.doFinally{}` is executed after the `getPolarhrMeasurements()` is called and `_polarhrMeasurements` has no data yet. For example this will return empty result:
```
...
getPolarhrMeasurements()
broadcastDisposable.dispose()
...
```
 
- Do you clear the `polarhrList` before the `getPolarhrMeasurements()` gets called? `_polarhrMeasurements` hashMap has the reference to `polarhrList` data. For example this will return empty result:
```
    fun getPolarhrMeasurements(): HashMap<String, ArrayList<Int>> {
        polarhrList.clear()
        Log.d(TAG, "getPolarhrMeasurements: $_polarhrMeasurements ")
        return _polarhrMeasurements
    }
```

khangdang4vn, 2021-12-10:
Hi @JOikarinen 

I checked again everything. I call `getPolarhrMeasurements()` after `broadcastDisposable`, and do not clear `polarhrList` before `getPolarhrMeasurements()`

I have a feeling that I can not return any values from parameters I pass inside the subscription. Just a note that the function `getPolarhrMeasurements()` is outside of the function `onCreate()` and `_polarhrMeasurements` is the global parameter of the class. I don't know if changing anything in the subscription to Polar's API will affect the global parameter.

JOikarinen, 2021-12-11:
@khangdang4vn your problem is a bit of mystery. I did little test using the [example](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp) application. I tried to simulate your situation, by modifying example code like this:

 ```
    private val POLAR_HR = "POLAR_HR"
    private var polarhrList = ArrayList<Int>()
    private var _polarhrMeasurements = HashMap<String, ArrayList<Int>>()

    // called from the Activity onCreate() function
    private fun setupBroadcastListenerButton() {
        broadcastButton.setOnClickListener {
            if (!this::broadcastDisposable.isInitialized || broadcastDisposable.isDisposed) {
                toggleButtonDown(broadcastButton, R.string.listening_broadcast)
                broadcastDisposable = api.startListenForPolarHrBroadcasts(null)
                    .doFinally {
                        _polarhrMeasurements[POLAR_HR] = polarhrList
                        Log.d(TAG, "doFinally: $_polarhrMeasurements ")
                    }
                    .subscribe(
                        { polarBroadcastData: PolarHrBroadcastData ->
                            Log.d(TAG, "HR BROADCAST ${polarBroadcastData.polarDeviceInfo.deviceId} HR: ${polarBroadcastData.hr} batt: ${polarBroadcastData.batteryStatus}")
                            polarhrList += polarBroadcastData.hr
                        },
                        { error: Throwable ->
                            toggleButtonUp(broadcastButton, R.string.listen_broadcast)
                            Log.e(TAG, "Broadcast listener failed. Reason $error")
                        },
                        { Log.d(TAG, "complete") }
                    )
            } else {
                toggleButtonUp(broadcastButton, R.string.listen_broadcast)
                broadcastDisposable.dispose()
                getPolarhrMeasurements()
            }
        }
    }

   // called when broadcast listening is ended
    private fun getPolarhrMeasurements(): HashMap<String, ArrayList<Int>> {
        Log.d(TAG, "getPolarhrMeasurements: $_polarhrMeasurements ")
        return _polarhrMeasurements
    }
 ```
 
 I cannot reproduce you problem.

khangdang4vn, 2021-12-21:
Hi @JOikarinen,

Thank you very much for your help to simulate my situation. You're right, there is no problem with this. Well, there was some issue with my application (written in Java), but I figured. It works well now. 

And because I call _polarhrMeasurements from another class (not in PolarActivity), so I need to put the variable inside the companion object. (I'm not familiar with Kotlin)
```
class PolarActivity : AppCompatActivity() {
    companion object {
        var polarhrMeasurements = HashMap<String, ArrayList<Int>>()
}}
```
Thank you again and I hope you're having a happy holiday!

closed 2021-12-21 by khangdang4vn