polarofficial/polar-ble-sdk #218: Synchronizing two devices

cjs30, 2021-12-15:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I have a Javascript application that reads PPG from a Sense (using SDK mode) and ECG from a H10 - both at maximum rates. The question is how best to synchronize the timings. I have a function generator (asymmetric square waves) flashing an LED at the Sense and the electrical signal going to the H10 so that I can see the timing errors between the two streams. The Sense reports a device EPOC of around 2000-01-01T00:00:00.3 calculated using the computer time of the returned data packet and the time of the last Sense sample. Clearly there is a 300ms error (which I suspect is because the Sense BLE signal has some delay?) also the EPOC is somewhat variable which suggests my methodology is wrong. The H10 seems to report a EPOC of 19 years before the time it was switched on (i.e. it is variable and there is no clock running on the device once powered down). How can I synchronize the streams? The Sense isn't really the issue (if the clock remains sync'd with the computer), the problem is more the H10. Is there a way of setting the H10 EPOC? And, how does one deal with the BLE delays during setting of the EPOC? I have attached a screen shot with the Device EPOCs each time I stop and restart the streams (and also the PPG and ECG signals overlaid).

[image]

JOikarinen, 2021-12-16:
@cjs30 cool! I have just quick comments for now. I will come back on this as soon as possible.  

> The Sense reports a device EPOC of around 2000-01-01T00:00:00.3 calculated using the computer time of the returned data packet and the time of the last Sense sample. Clearly there is a 300ms error [...]
- I got interested on this. Comparing the time of the LED flash and timestamp recorded by Sense from the same event you  see 300ms time difference. How do you set the time for Sense? You suspect, the time error is because time setup over BLE causes delays before time is actual set in Sense?   

> Is there a way of setting the H10 EPOC?
- EPOCH time of H10 is 1.1.2000 and default wall clock time is 2019-01-01T00:00:00.0 (just checked by H10 firmware 3.1.1). EPOCH time cannot be set, but the wall clock time yes.

cjs30, 2021-12-16:
Many thanks! I suspect I am doing something wrong.

>How do you set the time for Sense? You suspect, the time error is because time setup over BLE causes delays before time is actual set in Sense?

The way I deal with 'time' is to get the computer time (Date.now()) at the point I receive the data first data packet from the Sense (with its time encoded at the start of the packet). Then I calculate the time offset between the computer and the Sense and also between the H10 and the computer when I get the first data packet from it. I think this is the only way I can synchronize the devices. I don't know enough about the delays within BLE to know what the lag is between the Sense creating its timestamp within the data package and the BLE data packet arriving at the triggered function. That delay is (presumably) critical (or, I am misunderstanding some element of the system). I don't attempt to modify the clock running on the Sense or H10. If I could then that 'might' be useful if the delays were more predictable.  Ultimately, without a predictable or known latency between requesting and receiving a response I can't see how to synchronize two devices without an external trigger sensed by both devices. There must be a way. If I could get a 'round-trip' message (which includes the device time) with a total journey time from the computer to the device and back to the computer with a latency of less than 1ms, that would be great.

> > Is there a way of setting the H10 EPOC?
> 
> * EPOCH time of H10 is 1.1.2000 and default wall clock time is 2019-01-01T00:00:00.0 (just checked by H10 firmware 3.1.1). EPOCH time cannot be set, but the wall clock time yes.

How do I set the wall clock time on the H10 (without using the Library)? And, even better, is there a way of requesting/querying the device time without starting a stream? Or, where in the Library can I find the code that sets/reads it? I think I have a general problem finding the 'under-the-hood' code with the byte-data to 'write' and the responses expected. I am just working off the [PMD Specification](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf). I have browsed various parts of the SDK but my knowledge of the file structures of Android and IOS programming is insufficient to find the code sitting behind the various functions...I am not a programmer, I just dabble! Any help or advice you could provide would be extremely welcome.

JOikarinen, 2021-12-16:
Okay, I understand your setup and mostly the problems too. I am afraid I don't have breakthrough, but let me share the information I have.  

> The way I deal with 'time' is to get the computer time (Date.now()) at the point I receive the data first data packet from the Sense (with its time encoded at the start of the packet).

- Each BLE data packet you receive from H10 and Sense contains multiple samples, applies for both data types you are interested, i.e. ECG and PPG. The is received timestamp represent the timestamp of the last sample within received packet. So if you want to calculate (i.e. estimate) the timestamps of other samples in received packet then you need to calculate backwards from last sample. 

> I don't know enough about the delays within BLE to know what the lag is between the Sense creating its timestamp within the data package and the BLE data packet arriving at the triggered function.
- this is difficult one. I am afraid the lag is very difficult to define. The data will travel through many layers of software, each causing some delay. The BLE communication causes unpredictable delays, due the nature of the communication. Maybe the delay could predicted to be within some range. Another problem is the Sense and the way how it compress the data. The compression can really cause variable delays, i.e. in some cases Sense can compress data efficiently and some other moments it cannot. 

> Ultimately, without a predictable or known latency between requesting and receiving a response I can't see how to synchronize two devices without an external trigger sensed by both devices.
- I agree with the problem

> How do I set the wall clock time on the H10 (without using the Library)?
- unfortunately, that is not supported feature outside the SDK library. 

> And, even better, is there a way of requesting/querying the device time without starting a stream?
- unfortunately this is not supported even by the SDK library

cjs30, 2021-12-17:
Many thanks! In the end I have activated the Sense ACC stream and the H10 ACC stream (with times back calculated from the delta frames assuming constant sampling rate) to get the following:
[image]
I can now create an alignment routine. By the way, to avoid device orientation problems I have calculated the sqrt of the sum of the squares of the different axes. That gives an orientation independent scalar reporting the full acceleration on the device. Now I can simply strap the two devices together and run an alignment routine to minimized the sum of the squares difference between the traces. Of course, the next thing is to feed in a higher frequency vibration and see if the timings in the samples is steady. I notice that the delta frames vary in bit depth according to the largest delta within the frame, so as long as the signals aren't changing too rapidly I guess the data packets should be small. I can easily set a flag for when the bit depth of the delta frames gets large and mark those sections as 'dubious'.
Many thanks for your help - I think I am now good to go!

closed 2021-12-17 by cjs30

JOikarinen, 2021-12-18:
Excellent! It would be interesting to hear how is your project progressing, please keep us up to date.