polarofficial/polar-ble-sdk #219: Failure to import PolarBleSdk with M1 chip

hwixley, 2021-12-17:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [x ] Device is not relevant for this bug

**Describe the bug**
Failure to import PolarBleSdk after pod install on M1 chip, which produced the following error:
`import PolarBleSdk`
Gave this error:
`Failed to build module 'PolarBleSdk' from its module interface; the compiler that produced it, 'Apple Swift version 5.5.1 (swiftlang-1300.0.31.4 clang-1300.0.29.6)', may have used features that aren't supported by this compiler, 'Apple Swift version 5.4.2 (swiftlang-1205.0.28.2 clang-1205.0.19.57)'`

**How to Reproduce**
Pod install the polar dependencies on an M1 mac as follows:
```
target 'project-name' do
    use_frameworks!
    pod 'PolarBleSdk', '~> 3.2'
end

pod_install do |installer|
  installer.pods_project.targets.each do |target|
   target.build_configurations.each do |config|
     config.build_settings['BUILD_LIBRARY_FOR_DISTRIBUTION'] = 'YES'
   end
 end
end
```

**Expected behavior**
Description of what you expected to happen. 

**Screenshots and logs**
If applicable, add screenshots or logs to help explain your problem.

JOikarinen, 2021-12-18:
Hi @hwixley, let's investigate.

> ```
> Failed to build module 'PolarBleSdk' from its module interface; the compiler that produced it, 'Apple Swift version 5.5.1 (swiftlang-1300.0.31.4 clang-1300.0.29.6)', may have used features that [...]
- which version of XCode you have installed? From the error log it looks like you have some older version than XCode 13.2? The SDK is build with XCode 13.2, thus swift version 5.5.1.

hwixley, 2021-12-18:
Yes, you are completely right, I was using Xcode version 12.5.1. Sorry for flagging this as a bug I misread the error log and initially thought my Xcode version was newer than the compiled PolarBleSdk library and so was rather confused.

Thanks for the help, it's working now after updating Xcode to 13.2.1!

closed 2021-12-18 by hwixley