polarofficial/polar-ble-sdk #221: What is the zero date for the H10 timestamp?

KennethEvans, 2022-01-03:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I am trying the get the timestamps for ECG data from the H10 in Java time. I have seen a lot about the epoch time and default time for the OH1 and Sense in these issues, but I can't find that information for the H10. I'm sorry if I missed it.

If I take the timestamp (converted to ms) from the polarEcgData and subtract it from the current time (e.g. new Date()), I get around 1041551348000 ms, which corresponds to Thu Jan 02 18:49:08 EST 2003. (I am in the US EST time zone.)

It is different from what I understand for the OH1 and Sense:

epoch = Jan 01 2000 00:00:00
default = Jan 01 2017 10:00:00

I assume whether the Polar times are UTC or not, doesn't matter. It has no way of knowing what time zone it is in when the ECG values were gathered.

Is that the default time? Or is it just meaningless? I guess the real question is how do I get the current time in Java from the timestamp?

Thanks.

JOikarinen, 2022-01-03:
Hi @KennethEvans, 

H10 is working similar as OH1 and Sense. Here is some explanations

Epoch time:
- In Polar sensors (H10, H9, VS and OH1) the epoch time is chosen to be Jan 01 2000 00:00:00. The traditional Unix epoch Jan 01 1970 00:00:00 wasn't chosen because there were some memory limitations in earlier Polar sensors, nowadays the memory constraint does not exist but the decision was to stay in epoch Jan 01 2000 00:00:00

Sensor time:
- the time sensor keeps in its memory. The time is nanoseconds since epoch Jan 01 2000 00:00:00
- the time of the sensor is set to default value when sensor resets or battery runs out. For the Polar H10 (v.3.1.1) the default time is Jan 01 2019 00:00:00 (GMT), which in nanosecond format is 599616000000000000
- sensor time can be set by the API function [setLocalTime()](https://github.com/polarofficial/polar-ble-sdk/blob/f9a3912d6e6440cca13fcfbb55d6324e480d4e47/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L202) 

Time stamp: 
- time stamp is the sampling time of sample in the the data stream represent in nanoseconds. 

Example:
- reset the H10 by removing battery
    -  the time is set to 599616000000000000
- start ECG stream and observe received timestamps
    -  the timestamp received can be for example 599616023171427290. 
- stop ECG stream
- set the time to the sensor with API function [setLocalTime()](https://github.com/polarofficial/polar-ble-sdk/blob/f9a3912d6e6440cca13fcfbb55d6324e480d4e47/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L202). For example I set Mon Jan 03 08:10:38 2022 GMT+02:00 2022
      -  the time is set to 694505438000000000 
- start ECG stream and observe received timestamps
    -   the timestamp received can be for example 694512749912654440.
    
Some considerations:
- you may use the epoch time to human readable time converters in your code, which are usually based on the unix epoch of Jan 01 1970 00:00:00. Then you shall add the offset between Jan 01 1970 00:00:00 and Jan 01 2000 00:00:00 to timestamp before passing the timestamp to converter. Offset in nanoseconds is 946684800000000000

JOikarinen, 2022-01-03:
Now there is a new documentation explaining time related matters:
https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/TimeSystemExplained.md

KennethEvans, 2022-01-03:
The new documentation is very useful. It makes things clearer all in one place. Thanks.

However, it does not explain what is happening for me.

The battery was replaced on my H10 about May 2, 2020.
This was after FW 3.1.1 was installed.
I have never done setLocalTime() or any other reset of the H10.

Thus I would expect the time stamps to be over a year too early. What I get is timestamps that are very nearly equal to 599616000000, meaning they are close to the date set at battery replacement, not over a year earlier as expected.

Specifically this is the code (everything is ms not ns):
```
    long offset0 = 946684800000L; // epoch to Polar
    long offset1 = 599616000000L; // Polar to default
    long ts = polarEcgData.timeStamp / 1000000;
    long ts1 = ts + offset0;
    long ts2 = ts + offset0 + offset1;
    long now = new Date().getTime();
    Log.d(TAG, String.format("%15d %-8s %.2f years" , offset1, "offset1", msToYears(offset1)));
    Log.d(TAG, String.format("%15d %-8s %.2f years" , ts, "ts", msToYears(ts)));
    Log.d(TAG, String.format("%15d %-8s %.2f years" , ts1, "ts1", msToYears(ts1)));
    Log.d(TAG, String.format("%15d %-8s %.2f years" , ts2, "ts2", msToYears(ts2)));
    Log.d(TAG, String.format("%15d %-8s %.2f years" , now, "now", msToYears(now)));
```
and this is the result (ms from Java epoch, Jan 1, 1970 00:00:00 UTC):
```
    599616000000 offset1  19.01 years
    599632833276 ts       19.01 years
   1546317633276 ts1      49.03 years
   2145933633276 ts2      68.05 years
   1641244784077 now      52.04 years
```
Years is just converting ms to years assuming a 365-day year. It's more intuitive than interpreting ms.
```
    public double msToYears(long longVal) {
        return longVal / (1000. * 60. * 60. * 24. * 365.);
    }
```
It's certainly possible I don't understand or am doing something stupid, but I'm not seeing it.

Thanks.

JOikarinen, 2022-01-04:
> Thus I would expect the time stamps to be over a year too early. What I get is timestamps that are very nearly equal to 599616000000, meaning they are close to the date set at battery replacement, not over a year earlier as expected.

Thanks @KennethEvans for pointing out the inconsistency in the documentation. The H10 has indeed different behaviour compared to OH1 and VeritySense what comes to time keeping. The H10 is actually loosing the time information once it shuts down. The moment of H10 shutdown may be a bit vague depending on the conditions. If you remove the H10 from the strap and H10 has no BLE connection the H10 will shutdown within a minute.

So with the H10, if the time stamps of the stream needs to be comparable to wall time in your application, it would be a good idea to use [setLocalTime()](https://github.com/polarofficial/polar-ble-sdk/blob/f9a3912d6e6440cca13fcfbb55d6324e480d4e47/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L202) function before starting the stream.

KennethEvans, 2022-01-04:
Ok. However, I still don't know how to get the timestamp. None of those combinations I tried,  based on the documentation, is even close to now.

I haven't tried setLocalTime(), but the time it is using is very close to that. It is typically reattached to the strap shortly before using it.

KennethEvans, 2022-01-04:
I did setLocalTime() and nothing changed:
```
    599616000000 offset1  19.01 years
    599617031784 ts       19.01 years
   1546301831784 ts1      49.03 years
   2145917831784 ts2      68.05 years
   1641326619017 now      52.05 years
```
Specifically this is what I did:
```
            Calendar calNow = Calendar.getInstance();
            mApi.setLocalTime(mDeviceId, calNow);
            Log.d(TAG, "Doing setLocalTime to " + calNow.getTime());
            mEcgDisposable =
                    mApi.requestStreamSettings(mDeviceId,
                            PolarBleApi.DeviceStreamingFeature.ECG).
                            toFlowable().
                            flatMap((Function<PolarSensorSetting,
                                    Publisher<PolarEcgData>>) sensorSetting -> {
           ...
```

JOikarinen, 2022-01-05:
Let's go through print output and see do we agree.

>     599616000000 offset1  19.01 years
looks correct, years since 2000

>     599632833276 ts       19.01 years
looks correct, years since 2000

>    1546317633276 ts1      49.03 years

looks correct, years since 1970. The H10 time was Tuesday, January 1, 2019 4:40:33.276 GMT0 at the moment the sample was taken.

>    2145933633276 ts2      68.05 years

Here might be something. The output  is correct: 19 years + 30 years + 19 years = 68 years. However, I suppose this is the point where our thinking collide. 'ts' and 'ts1' are already representing the time in Polar H10, so I don't get the reason for 'ts2'

>    1641244784077 now      52.04 years
looks correct, years since 1970

> Specifically this is what I did:
>
> ```
>             Calendar calNow = Calendar.getInstance();
>             mApi.setLocalTime(mDeviceId, calNow);
>             Log.d(TAG, "Doing setLocalTime to " + [...]

You may try to chain the `setLocalTime()`, to make sure it is completed before starting the stream. I tried it and in my test it worked. Something like this:

 ```
             Calendar calNow = Calendar.getInstance();
             //mApi.setLocalTime(mDeviceId, calNow);
             Log.d(TAG, "Doing setLocalTime to " + calNow.getTime());
             mEcgDisposable =
                     mApi.setLocalTime(mDeviceId, calNow)
                             .andThen(requestStreamSettings(mDeviceId,PolarBleApi.DeviceStreamingFeature.ECG))
                             .toFlowable()
                             .flatMap((Function<PolarSensorSetting,
                                     Publisher<PolarEcgData>>) sensorSetting -> {
            ...
 ```

KennethEvans, 2022-01-05:
> looks correct, years since 2000
> 
> 1546317633276 ts1 49.03 years
> 
> looks correct, years since 1970. The H10 time was Tuesday, January 1, 2019 4:40:33.276 GMT0 at the moment the sample was taken.

No. The H10 time should just have been reset to now. It shouldn't be near 2019.

I will try your suggestion. I'm not sufficiently competent in ReactiveX to have figured that out myself.

As to ts3, it seems to me the Java time in milliseconds is the timestamp plus offset1 + offset2. My understanding is ts/1000000 is the sample time in milliseconds since whatever the internal time was reset to, using the Polar epoch of 2000. It doesn't look like my understanding is correct. 😉

If it were working, I'd just use whatever combination gives me the timestamp that is the current time and avoid all that thinking. 😎

I appreciate your help.

KennethEvans, 2022-01-05:
I tried what you said, and it gives: Exception: com.polar.sdk.api.errors.PolarServiceNotAvailable
This exception is generated from  the throwable part of the RxJava block and happened at the call to mApi.setLocalTime(). This is my code (not quite the same as yours as I added mApi.):
```
            Calendar calNow = Calendar.getInstance();
//            mApi.setLocalTime(mDeviceId, calNow);
            Log.d(TAG, "Doing setLocalTime to " + calNow.getTime());
            mEcgDisposable = mApi.setLocalTime(mDeviceId, calNow)
                    .andThen(mApi.requestStreamSettings(mDeviceId,
                            PolarBleApi.DeviceStreamingFeature.ECG))
                    .toFlowable()
           ...
```
As another request for help with RxJava, I would like the Calendar calNow = and the log message to be in the block that calls mpi.setLocalTime() (i.e. the _andThen_ block). I can never seem to figure out how to do that.

The full stack trace is:
```
2022-01-05 13:19:43.104 5641-5641/net.kenevans.polar.polarecg E/KE.NetECG: ECG Error: null
    com.polar.sdk.api.errors.PolarServiceNotAvailable
        at com.polar.sdk.impl.BDBleApiImpl.sessionServiceReady(BDBleApiImpl.java:878)
        at com.polar.sdk.impl.BDBleApiImpl.sessionPsFtpClientReady(BDBleApiImpl.java:905)
        at com.polar.sdk.impl.BDBleApiImpl.setLocalTime(BDBleApiImpl.java:280)
        at net.kenevans.polar.polarecg.ECGActivity.streamECG(ECGActivity.java:882)
        at net.kenevans.polar.polarecg.ECGActivity$5.streamingFeaturesReady(ECGActivity.java:1128)
        at com.polar.sdk.impl.BDBleApiImpl.lambda$setupDevice$54$com-polar-sdk-impl-BDBleApiImpl(BDBleApiImpl.java:1041)
        at com.polar.sdk.impl.BDBleApiImpl$$ExternalSyntheticLambda13.accept(Unknown Source:6)
        at io.reactivex.rxjava3.internal.operators.single.SingleDoOnSuccess$DoOnSuccess.onSuccess(SingleDoOnSuccess.java:54)
        at io.reactivex.rxjava3.internal.operators.single.SingleSubscribeOn$SubscribeOnObserver.onSuccess(SingleSubscribeOn.java:68)
        at io.reactivex.rxjava3.internal.operators.single.SingleCreate$Emitter.onSuccess(SingleCreate.java:68)
        at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient.readFeature$lambda-18(BlePMDClient.kt:294)
        at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient.$r8$lambda$Fv5o0Zoem-5g7LfJiXjmj4ukYlA(Unknown Source:0)
        at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient$$ExternalSyntheticLambda2.subscribe(Unknown Source:4)
        at io.reactivex.rxjava3.internal.operators.single.SingleCreate.subscribeActual(SingleCreate.java:40)
        at io.reactivex.rxjava3.core.Single.subscribe(Single.java:4855)
        at io.reactivex.rxjava3.internal.operators.single.SingleSubscribeOn$SubscribeOnObserver.run(SingleSubscribeOn.java:89)
        at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run(Scheduler.java:644)
        at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:65)
        at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:56)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:919)
```

JOikarinen, 2022-01-06:
Okay, lets start with this. 

> > looks correct, years since 2000
> > 1546317633276 ts1 49.03 years
> > looks correct, years since 1970. The H10 time was Tuesday, January 1, 2019 4:40:33.276 GMT0 at the moment the sample was taken.
> 
> No. The H10 time should just have been reset to now. It shouldn't be near 2019.
- even you set the time `setLocalTime()' or not, the H10 always reset back to Jan 01 2019 00:00:00 when ever it has been detach from the strap and there is no BLE connection for around 30s-1min. So it looks like the H10 has been on for ~ 4h40min when the timestamp is taken.

JOikarinen, 2022-01-06:
> I tried what you said, and it gives: Exception: com.polar.sdk.api.errors.PolarServiceNotAvailable 

My apologies, I tested my suggestion with the [example application](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp) and it worked. As a consequence I totally forget important aspects what that change may cause.

Here is the example code again, which shall work:

```
...
 Calendar calNow = Calendar.getInstance();
 calNow.setTime(new Date()); //this line was missing from previous sample
            
 ecgDisposable = api.setLocalTime(deviceId, calNow)
     .andThen( api.requestStreamSettings(deviceId, PolarBleApi.DeviceStreamingFeature.ECG))
     .toFlowable()
     .flatMap((Function<PolarSensorSetting, Publisher<PolarEcgData>>) sensorSetting -> api.startEcgStreaming(deviceId, sensorSetting.maxSettings()))
     .observeOn(AndroidSchedulers.mainThread())
     .subscribe(
     ...                    
```
                    
Then comes the important part. As can be seen from the [setLocalTime](https://github.com/polarofficial/polar-ble-sdk/blob/580fee6cb38cbdd9eaec95276e4f58623c72c219/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L202) it requires the feature [FEATURE_POLAR_FILE_TRANSFER](https://github.com/polarofficial/polar-ble-sdk/blob/580fee6cb38cbdd9eaec95276e4f58623c72c219/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L108). So your API initialisation shall look something like this:

```
 api = PolarBleApiDefaultImpl.defaultImplementation(getApplicationContext(),
                PolarBleApi.FEATURE_POLAR_SENSOR_STREAMING |
                        PolarBleApi.FEATURE_BATTERY_INFO |
                        PolarBleApi.FEATURE_POLAR_FILE_TRANSFER |
                        PolarBleApi.FEATURE_DEVICE_INFO |
                        PolarBleApi.FEATURE_HR);
```

Because of the [FEATURE_POLAR_FILE_TRANSFER](https://github.com/polarofficial/polar-ble-sdk/blob/580fee6cb38cbdd9eaec95276e4f58623c72c219/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L108) some additional dependencies are required too:

```
    ...
    implementation 'com.google.protobuf:protobuf-javalite:3.17.3'
    implementation 'commons-io:commons-io:2.10.0'
    ...
```

KennethEvans, 2022-01-06:
This is what I found:

1. I was already using FEATURE_POLAR_FILE_TRANSFER. (I believe that is in the new documentation.)
2. calNow.setTime(new Date()) is not necessary. Calendar.getInstance() has the current time. However you have to set the instance in UTC time (since that's all the device knows about):
```
            TimeZone timeZone = TimeZone.getTimeZone("UTC");
            Calendar calNow = Calendar.getInstance(timeZone);
```
3. I was not using the extra implementations, but they were not the cause of the problem. If they were necessary, I would have expected to get a build error.
4. It does not work with setLocalTime() before the start of the RxJava code as I originally did it. If this is true, it should be documented, and I probably should have got an exception trying to do it. It seems to me that from the documentation, I could just as well have called this in deviceConnected() or anywhere else after it is connected. Apparently not.
5. It does work in the _andThen_ construct. It is beyond my expertise to know why.
6. The correct timestamp in Java time requires adding just the offset to the Polar epoch (946684800000L). This is ts1 in my code. My ts2 is not the correct thing to do.

This is what I get now (with some more debugging added). The device knows nothing of which time zone it is in, but the offsets are from a UTC time. This is confusing, but as long as you do setLocalTime() with the current UTC time, it is ok. Note that ts1 and now are the same (within seconds) in either UTC or the current time zone (EST for me.)

```
 599616000000 offset1  19.01 years  Jan 01 1989 00:00:00 UTC Dec 31 1988 19:00:00 EST
 694806050785 ts       22.03 years  Jan 07 1992 17:40:50 UTC Jan 07 1992 12:40:50 EST
1641490850785 ts1      52.05 years  Jan 06 2022 17:40:50 UTC Jan 06 2022 12:40:50 EST
2241106850785 ts2      71.07 years  Jan 06 2041 17:40:50 UTC Jan 06 2041 12:40:50 EST
1641490852231 now      52.05 years  Jan 06 2022 17:40:52 UTC Jan 06 2022 12:40:52 EST
```
I hope this helps clarify the issue.

JOikarinen, 2022-01-07:
> This is what I found:
> 
> 1. I was already using FEATURE_POLAR_FILE_TRANSFER. (I believe that is in the new documentation.)
👍

> 2. calNow.setTime(new Date()) is not necessary. Calendar.getInstance() has the current time. However you have to set the instance in UTC time (since that's all the device knows about): [...]
True

> 3. I was not using the extra implementations, but they were not the cause of the problem. If they were necessary, I would have expected to get a build error.
> 4. It does not work with setLocalTime() [...]

About the statement 4 and 5.

`api.setLocalTime()` is returning RX Completable observable. To trigger the action you shall subscribe to it. To simple solution would be
```
api.setLocalTime(deviceId, calendar).subscribe()
```

However, ```andThen()``` operator came to my mind, because the set of local time is probably good idea to do first, before starting the ECG stream. ```AndThen()``` guarantees the time is set first to device and only after time set the execution proceeds next actions on the down stream chain.   

> 6. The correct timestamp in Java time requires adding just the offset to the Polar epoch (946684800000L). This is ts1 in my code. My ts2 is not the correct thing to do.
> 
Yes, I agree. 

> This is what I get now (with some more debugging added). The device knows nothing of which time zone it is in, but the offsets are from a UTC time. This is confusing, but as long as you do [...]
- Exactly correct. I understand the confusion here. The device doesn't really know about the time zones. Either you set `Mon Jan 03 08:10:38 2022 GMT+02:00` or `Mon Jan 03 08:10:38 2022 GMT+00:00`, device only things the time is set to `Mon Jan 03 08:10:38 2022`. That smells like a bug, let's think what shall be done for it.

KennethEvans, 2022-01-07:
Looks like we're on agreement about everything.

`
Exactly correct. I understand the confusion here. The device doesn't really know about the time zones. Either you set Mon Jan 03 08:10:38 2022 GMT+02:00 or Mon Jan 03 08:10:38 2022 GMT+00:00, device only things the time is set to Mon Jan 03 08:10:38 2022. That smells like a bug, let's think what shall be done for it.`

Yes, I'm more used to devices that know what time it is. Without a connection to the Internet, or the outside world in general, the H10 **cannot** know what time it is. I don’t think it's a bug. It's all it can do.

`
api.setLocalTime() is returning RX Completable observable. `

I didn't notice that. I guess that is a clue it needs to be called in an RxJava chain. I do think the documentation should indicate that. Again, RxJava is not my forte.

I'd still like to put the Calendar code onside the

KennethEvans, 2022-01-07:
I didn't intend to close it. My phone often does things on its own accord. 😉

JOikarinen, 2022-01-08:
Thanks for good topic. Are we good with all the questions at the moment? Shall we close the issue?

The issues that sensors are not taking care of timezone is now explained in
https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/TimeSystemExplained.md

The timezone issue cannot be easily solved on SDK library so I have triggered internal bug that issues shall be fixed in sensor firmware.

KennethEvans, 2022-01-08:
Are you sure about

`timestamp is represented in nanoseconds since 2000-01-01T00:00:00Z. The sensor time is set to your local time, then timestamp is nanoseconds to the local time. For example if sensor time is set to 2022-01-03T08:00:00 UTC+02:00 or 2022-01-03T08:00:00 UTC+00:00 the timestamp still is 694512000000000000`

The sensor has no way to know the local time,  and so it cannot set it to local time.  When the H10 is disconnected, it sets to the default time, and that is UTC. When I set the time using setLocalTime, it makes a difference if I use a Calendar in local time or UTC time. If I use local time, the timestamps are off by my UTC offset.

Your use of "set" is ambiguous. What is important is what Calendar is used, local or UTC. That is specific and what the programmer needs to know.

I think everything in the sensor is UTC. It can't do better than that.

I also think the important point is that _setLocalTime needs to be done with a Calendar in the UTC time zone if you want the timestamps to be in local time._

Alternatively you could do setLocalTime with a local Calendar, then adjust the timestamps by the offset. I don't see any point in doing it that way, though.

There's also a typo ("do" vs "does") above this section.

KennethEvans, 2022-01-08:
The other programmer pitfall is you can't call setLocalTime just anywhere. I don't why though.

JOikarinen, 2022-01-09:
> The sensor has no way to know the local time, and so it cannot set it to local time.
I think, we have mutual understanding on this matter. But still I would express the situation a bit differently. If time is set using a function `setLocalTime()`, sensor shall know the local time (and UTC time too). However, sensor sets its internal time by ignoring the timezone part. Once the time set for the sensor the sensor thinks the 2022-01-03T08:00:00 UTC+02:00 are 2022-01-03T08:00:00 UTC+00:00 the same time 2022-01-03T08:00:00Z. The problem then comes with the timestamp, the correct timestamp in my opinion would be always in UTC. With the example times given:

If sensor time set is `2022-01-03T08:00:00 UTC+00:00`:
- Time stamp shall be from 2000-01-01T00:00:00Z to 2022-01-03T08:00:00 UTC+00:00. In this case the sensor works as intended. 

If sensor time set is `2022-01-03T08:00:00 UTC+02:00`:
- Time stamp shall be from 2000-01-01T00:00:00Z to 2022-01-03T**06:00:00** UTC+00:00. In this case the sensor actually gives the timestamp from 2000-01-01T00:00:00Z to 2022-01-03T08:00:00Z
 
> The other programmer pitfall is you can't call setLocalTime just anywhere.
- `setLocalTime` can be used when ever the BLE connection is made to sensor, no problem.

KennethEvans, 2022-01-09:
Yes, the same time has equivalent representations in UTC and local. They are different, but it is the same time. The issue is how you use setLocalTime and what you set. That is what I think is most important in the documentation.

If you want your timestamps in UTC you do something different than if you want them in local. (I want them in local because the HR and RR data are in local because I have to make the timestamp for those myself. It's not in the hrNotification data. In general, anything in the UI should be local. In other cases UTC as you like may be better.)

**Making clear how to do either is what I would like to see.**

The other issue about which I'm completely confused (and which is very much related to how to do either):

```
setLocalTime can be used when ever the BLE connection is made to sensor, no problem.
```

That is not my experience. The only place it has worked is with the andThen usage in the RxJava chain as you proposed. It did not work when called just before the chain (near the beginning of this topic). There is a Bluetooth connection to the sensor then.

I also replaced the block before the andThen with a new Completable.fromCallable() that sets the calendar, does logging, and returns setLocalTime. It seems to me that is RxJava effectively equivalent to just specifying setLocalTime as the Completable instead.

I thought I posted  that but I don't see it. It didn't work. The time was not set.

It will take me a while to redo it and post the code as I have some other obligations. This is an issue where we aren't on the same page.

KennethEvans, 2022-01-09:
This is what I did that seems equivalent to just using setLocalTime(). It did not work. No exceptions, but the time was not reset.

```
        if (mEcgDisposable == null) {
            mEcgDisposable = Completable.fromCallable(() -> {
                // Set the local time to get correct timestamps. H10
                // apparently
                // resets its time to 01:01:2019 00:00:00 when
                // connected to strap
                TimeZone timeZone = TimeZone.getTimeZone("UTC");
                Calendar calNow = Calendar.getInstance(timeZone);
                Log.d(TAG, "setLocalTime to " + calNow.getTime());
                return mApi.setLocalTime(mDeviceId, calNow);
            })
                    .andThen(
                            mApi.requestStreamSettings(mDeviceId,
                                    PolarBleApi.DeviceStreamingFeature.ECG))
                    .toFlowable()
                     ...
```

JOikarinen, 2022-01-10:
Notice 1:
```
TimeZone timeZone = TimeZone.getTimeZone("UTC");
Calendar calNow = Calendar.getInstance(timeZone);
Log.d(TAG, "setLocalTime to " + calNow.getTime());
```
Intend in you code snippet is that you set the UTC time to the sensor. Your code works correctly. However, the call `calNow.getTime()` in logging statement gives the time in your local time, which may confuse. The time set to device is, as expected the UTC time, you may print out the time set to device like this: 
```
Log.d(TAG, "hours set to device " + calNow.get(Calendar.HOUR_OF_DAY));
Log.d(TAG, "minutes set to device "+ calNow.get(Calendar.MINUTE));
Log.d(TAG, "seconds set to device "+ calNow.get(Calendar.SECOND));
Log.d(TAG, "milli seconds set to device "+ calNow.get(Calendar.MILLISECOND));
Log.d(TAG, "time zone offset in millis set to device " +  calNow.get(Calendar.ZONE_OFFSET));
```

Notice 2:
I think, I see the problem in your approach with `Completable.fromCallable()`, the code inside that block is executed as expected. However, I would put this in to the category of ReactiveX surprises, explanation:

```
 mEcgDisposable = Completable.fromCallable(() -> {
             // The code inside this block is instance of Callable()
              ...
              // setLocalTime returns completable
              Completable completable =  api.setLocalTime(deviceId, calNow);
              // Here the Callable() returns variable completable
              return completable;
            })
```
Okay, then let's see what `.fromCallable()` does for any Callable return values, this is from the `.fromCallable()` documentation `executes the {@link Callable} function, ignores its normal result and emits {@code onError} or {@code onComplete} only.`.  Your code will work if doing it like this:
```
 mEcgDisposable = Completable.fromCallable(() -> {
             // The code inside this block is instance of Callable()
              ...
                api.setLocalTime(deviceId, calNow).subscribe();
                return "What ever is returned here is ignored";
            })
```
which in my opinion looks a bit of dangerous approach, so I would do it maybe like this:

```
 ...         
mEcgDisposable = api.setLocalTime(deviceId, calNow)
                  .andThen(mApi.requestStreamSettings(mDeviceId, PolarBleApi.DeviceStreamingFeature.ECG))
                 ...
```
By doing like suggested it is sure that time setup is done first before anything else in that RX chain. That may be wanted functionality, if down stream of the RX chain you call `api.startEcgStreaming()`.  By doing time setup first, we can be sure the streamed data will have the correct time stamp starting from very first data streamed. Alternatively you may do time setup somewhere else in your code by subscribing the  `setLocalTime`

KennethEvans, 2022-01-10:
Thank you for the detailed explanation.

I will accept the category of ReactiveX surprises, explanation. I'm definitely ReactiveX knowledgeable. However I don't understand your explanation.  It is a Completeable, just uses an anonymous rather than a predefined method. The Calendar call and the logging work as expected. The rest of the chain works as expected.  The difference is that the setLocalTime() does not set the time. 

The other instance where setLocalTime() does not set the time is the one I listed on Jan 4:

```
            Calendar calNow = Calendar.getInstance();
            mApi.setLocalTime(mDeviceId, calNow);
            Log.d(TAG, "Doing setLocalTime to " + calNow.getTime());
            mEcgDisposable =
                    mApi.requestStreamSettings(mDeviceId,
                            PolarBleApi.DeviceStreamingFeature.ECG).
                            toFlowable().
                            flatMap((Function<PolarSensorSetting,
                                    Publisher<PolarEcgData>>) sensorSetting -> {
           ...
```
This is the simpler approach, but setLocalTime() does not set the time.

So here are two cases (or one case if you ignore the anonynmous function result) that indicate you can't do setLocalTime() just anywhere.

What does work is your original andThen solution that you listed at the bottom of your last post:

```
mEcgDisposable = api.setLocalTime(deviceId, calNow)
                  .andThen(mApi.requestStreamSettings(mDeviceId, PolarBleApi.DeviceStreamingFeature.ECG))
```

so, I have a solution to my problem.  It's just that I don't understand why this one works and the others do not.  It seems to me that setLocalTime() is not working right. If so, that is a problem, I think. (But not mine. 😉)

This is just a conjecture: Maybe setLocalTime() needs to be subscribed to? I know there is all that stuff about lazy loading. Maybe it doesn't get loaded? Again, we're out of my expertise range.

JOikarinen, 2022-01-11:
Let me try to encourage you to learn RxJava: 

The communication with BLE device is asynchronous, you sent the command and BLE device responds asynchronously at some point of time. As a consequence it has been practical to make Polar BLE SDK API functions mostly using asynchronous programming patterns. Currently Polar BLE SDK has adopted the RxJava as an approach, but the SDK could   implement the same API's with callback pattern or nowadays with Kotlin suspend functions. It takes some effort to adapt for asynchronous thinking, but it will payback. 

```
Completable completableSetLocalTime = mApi.setLocalTime(mDeviceId, calNow);
```
So for example, once you call function `setLocalTime()` what happens is it returns the observable object. What does the object contains is the "recipe" or the instructions what shall be done once it is observable is subscribed.

```
 completableSetLocalTime
                .subscribe(
                        () -> Log.d(TAG, "time is set to device"),
                        error -> Log.d(TAG, "set time failed: $error")
                );
```
Then at some point you want to execute the instructions contained in the observable object, that is done by subscribing to object. At this point of time the actual command to setup the time is fired over BLE. You may pass the observer object for subscribe() function as a parameter to  listen when the command is completed and whether command did succeed.  

```
completableSetLocalTime
    .andThen( "Do something")
    .subscribeOn(Schedulers.io())
    .subscribe()
```
The real power of RxJava, in my opinion,  is how easy is to chain the asynchronous calls, use the operators to modify the data flows and define on which thread any part of the code shall run.

KennethEvans, 2022-01-13:
`Let me try to encourage you to learn RxJava:`

That isn't going to happen. It isn't applicable to most of what I do, and where it might be I already have working code. 

I am not a computer scientist. I, in fact, have never in my life had a computer course. What I have learned has been to achieve an end, and I am more interested in the end than how to get there. In other words I'm not impressed because it is "cool" or a popular thing to do.

Over the years I have found the way to functional, reliable, and maintainable code is the KISS principle. ReactiveX is the opposite of KISS.

That doesn't mean you shouldn't use it. You should use what works for you. The best software, the best language is the one you know how to use. What is important is if what you produce works, and the Polar SDK works.

JOikarinen, 2022-01-14:
Thanks for good explanation and very good point of view. 

> It is not my intent to be critical in a putdown way nor do I wish to argue (as opposed to discuss).
Exactly, we have good conversations. You have many good arguments, for example `providing conventional Javadoc` is definitely something shall be done, i.e. bundle the documentation to provided `.aar`

KennethEvans, 2022-01-14:
`Exactly, we have good conversations. `

Agree. I think you can close this issue. I have a solution that works and do not have anything more to add. 😉

I now better appreciate that the interactions with the device are asynchronous.

closed 2022-01-15 by JOikarinen