polarofficial/polar-ble-sdk #222: Getting ERROR_INVALID_MTU

KennethEvans, 2022-01-04:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I am occasionally getting ERROR_INVALID_MTU errors. This is with an app that does ECG streaming.
```
com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command error:  failed with error: ERROR_INVALID_MTU
```
I was in the middle of doing something the last time this occurred and didn't copy the full stack trace, unfortunately.
These are random and somewhat infrequent.  (Once a day?) They seem to be fixed by restarting the app.

JOikarinen, 2022-01-05:
> middle of doing something
Does it happen middle of ECG data stream already flowing? Or during the time you setup the ECG stream?

> They seem to be fixed by restarting the app.
Do you mean that once this happens then none of the attempted operations with H10 is working?

It would be helpful, if you could catch the logs when this happens.

KennethEvans, 2022-01-05:
Yes, I will be careful about getting details next time. The only thing I remember for sure is that the app wasn't working and needed to be restated. There was no ECG data being plotted. It worked normally after restarting. Sorry.

KennethEvans, 2022-01-08:
I got this again today. Again, it worked ok the next time (with no changes to the code). It is happening in the RxJava chain for streaming ECG for an H10. Here is the stack trace:

```
2022-01-08 15:06:55.806 6184-6184/net.kenevans.polar.polarecg E/KE.NetECG: ECG Error: pmd cp command error:  failed with error: ERROR_INVALID_MTU
    com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command error:  failed with error: ERROR_INVALID_MTU
        at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient.sendControlPointCommand$lambda-14(BlePMDClient.kt:245)
        at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient.$r8$lambda$_mWIojlvs7ItI9lsdIAR6dRGbxQ(Unknown Source:0)
        at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient$$ExternalSyntheticLambda1.subscribe(Unknown Source:6)
        at io.reactivex.rxjava3.internal.operators.single.SingleCreate.subscribeActual(SingleCreate.java:40)
        at io.reactivex.rxjava3.core.Single.subscribe(Single.java:4855)
        at io.reactivex.rxjava3.internal.operators.single.SingleSubscribeOn$SubscribeOnObserver.run(SingleSubscribeOn.java:89)
        at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run(Scheduler.java:644)
        at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:65)
        at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:56)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:919)
2022-01-08 15:06:55.807 6184-6184/net.kenevans.polar.polarecg E/KE.NetECG: Utils: ECGActivity: ECG Error: 
    Exception: com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command error:  failed with error: ERROR_INVALID_MTU
    pmd cp command error:  failed with error: ERROR_INVALID_MTU

```

KennethEvans, 2022-01-24:
I am continuing to get these though infrequently.  This is the last stack trace. Seems to be the same each time.

```
2022-01-23 18:52:39.935 21382-21382/net.kenevans.polar.polarecg E/KE.NetECG: ECG Error: pmd cp command error:  failed with error: ERROR_INVALID_MTU
    com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command error:  failed with error: ERROR_INVALID_MTU
        at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient.sendControlPointCommand$lambda-14(BlePMDClient.kt:245)
        at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient.$r8$lambda$_mWIojlvs7ItI9lsdIAR6dRGbxQ(Unknown Source:0)
        at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient$$ExternalSyntheticLambda1.subscribe(Unknown Source:6)
        at io.reactivex.rxjava3.internal.operators.single.SingleCreate.subscribeActual(SingleCreate.java:40)
        at io.reactivex.rxjava3.core.Single.subscribe(Single.java:4855)
        at io.reactivex.rxjava3.internal.operators.single.SingleSubscribeOn$SubscribeOnObserver.run(SingleSubscribeOn.java:89)
        at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run(Scheduler.java:644)
        at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:65)
        at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:56)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:919)
2022-01-23 18:52:39.936 21382-21382/net.kenevans.polar.polarecg E/KE.NetECG: Utils: ECGActivity: ECG Error: 
    Exception: com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command error:  failed with error: ERROR_INVALID_MTU
    pmd cp command error:  failed with error: ERROR_INVALID_MTU

```

JOikarinen, 2022-01-24:
@KennethEvans let's see what we have at the moment. From the stack trace I can assume that error happens at the moment your application sends command to H10 sensor, is that correct assumption? `ERROR_INVALID_MTU`comes from the H10 sensor. In means that MTU (maximum transmission unit) negotiation fails, which I haven't seen happening very often. At the moment this bug is a bit of mystery. 

Btw, what is the Android phone and its version you are using?

JOikarinen, 2022-01-25:
> My code is not in the stack trace, so I am not sure exactly when it occurs, but it has not connected.
Do you mean that when this happens the phone and H10 are not connected, and when connection is attempt the error happens. This could make some sense, as MTU negotiation happens at the moment the BLE connection is initialised. 

> Sounds like an issue with the H10.
It can be something in H10 or in SDK library. I also have my doubts on phone manufacturer BLE stack, which in Android world varies a lot between phone manufacturers. 

However, still this issue would need more information of what is exactly happening. What would help a lot if you could provide the logs of the event. Enabling the logs maybe done with [setApiLogger()](https://github.com/polarofficial/polar-ble-sdk/blob/38cb692dc9daf454325086de26425cd6d102a45c/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L184)

KennethEvans, 2022-01-25:
`Do you mean that when this happens the phone and H10 are not connected, and when connection is attempt the error happens. This could make some sense, as MTU negotiation happens at the moment the BLE connection is initialised.`
I would assume so, but I have no way of knowing.

My limited understanding of MTU is that it is negotiated and is related to throughout. (Higher MTU, more throughput.) I know from experience that in the early days of BLE the Android implementation of BLE had serious glitches. In any event fixing it via Android or the carrier is unlikely to be successful. The error message is not informative. Finding out how why the device sent it is more likely to be successful.

I have been working on my app a lot recently. I am implementing my own QRS detection algorithm. With my heart rate, none of the Polar devices reports it correctly at times. I have got 43 from the H10 when visually counting the peaks gives twice that. The Sense is not off that much,   but I don't think it's right either. This turned out to be more challenging than expected, but as of yesterday I feel the app is working correctly. (Apart from the QRS calculation I had to dig deeply into Android plot to get the app to handle orientation changes (without stopping the API) correctly, a different kind of challenge.) Bottom line is I don't expect to be working on it much, and hence won't be seeing these errors. 

As to setApiLogger. I don't know how that works. The link was just to the abstract definition of the call. I need more information. 

Finally, it's not a serious problem. If it's not easy to find the root cause, maybe it's not worth the time.

JOikarinen, 2022-01-26:
No problem, let's keep this open for a while if something new comes out which could give new ideas for investigation. 

 If you get back on debugging, the logging in Java code can be enabled like this:
```
api.setApiLogger(s -> Log.d("API_LOGGER", s));
```

KennethEvans, 2022-01-28:
Ok, I tried it. It's cool that you can then see the API messages in Logcat.

The problem is that normally I don't want all that output cluttering up the Logcat.  I can set filters to get or not get it. However, once the app has stopped if you change the filter (e.g. to one that includes the API messages) then Logcat is cleared.

The messages also go to the Run tab, but only for the ones that are being logged.

Thus I have no good way to get to these log messages after the MTU happens.

Have I missed something?

JOikarinen, 2022-01-29:
If you are currently using Android Studio logcat view, then one solution to see back to history is that you read logcat recordings from command line. That way you may see what has the application done before it crashed or stopped for some other reason. For example, to you may read the logcat to the file like this:
```
adb logcat > logcat.txt
```

If you use logcat from command line just like that, without any filters, then it is good to note that all you phone logs are in the file (from all the apps) and you may not want share such log file to anyone. 

One of my practice is to do the following. 

Clean all the logcat history
```
$adb logcat -c
```
Start recording the logcat of the app I am interested in
```
$adb logcat --pid=$(adb shell pidof -s my.exampleapp.packagename) > log_file.txt
```
Once logcat is recording, I start doing my test or what ever I was intended to do with the app. At the end of my testing stopping the logcat with Ctrl+C. 

So if something weird have happened while a was testing my app I can always go back to logs see what was going on in my app and probably find the root cause.

KennethEvans, 2022-01-29:
Thanks. I've never done it that way.

This error only occurs infrequently, so clearing and looking at a specific pid won't be useful. All I need is to see the logs when it happens.

I do have API logging turned on. If I get the error again, I'll try that.

Also I can't remember for sure, but I may be catching the exception. If so, the app will still be running, and the Android Studio filters should work.

We'll see what happens.

KennethEvans, 2022-01-29:
Later. I tried the command-line adb, and it looks like that will work. However, I can't see any advantage over using the interface in Android Studio.  I'm not sure when it gets cleared. (It apparently does get cleared when you do a Clear in Android Studio.)

However, I noticed that the API log messages are in the Run tab in Android Studio, even when they are filtered out in the Logcat tab. (I have to do this as there are too many.) So this is even better.

Should be good to go when it happens again.

theMagicalTree, 2022-05-24:
I'm experiencing this same error occasionally also. It happens when I reconnect following the device going out of range and then returning into range. HR data is received fine but the ECG stream doesn't initialize. Could it be a problem with the way the ECG streaming was abruptly terminated when the device went out of range?....

> 2022-05-24 20:05:23.332 14108-25683/com.nickDTaylor.cardioCapture D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=6 mScannerId=0
2022-05-24 20:05:24.267 14108-14108/com.nickDTaylor.cardioCapture I/BluetoothAdapter: STATE_ON
2022-05-24 20:05:24.267 14108-14108/com.nickDTaylor.cardioCapture I/BluetoothAdapter: STATE_ON
2022-05-24 20:05:24.267 14108-14108/com.nickDTaylor.cardioCapture D/BluetoothLeScanner: Stop Scan with callback
2022-05-24 20:05:24.273 14108-14108/com.nickDTaylor.cardioCapture I/BluetoothAdapter: STATE_ON
2022-05-24 20:05:24.274 14108-14108/com.nickDTaylor.cardioCapture I/BluetoothAdapter: STATE_ON
2022-05-24 20:05:24.278 14108-14108/com.nickDTaylor.cardioCapture D/BluetoothGatt: connect() - device: C6:10:34:8D:07:E6, auto: false
2022-05-24 20:05:24.278 14108-14108/com.nickDTaylor.cardioCapture I/BluetoothAdapter: isSecureModeEnabled
2022-05-24 20:05:24.279 14108-14108/com.nickDTaylor.cardioCapture D/BluetoothGatt: registerApp()
2022-05-24 20:05:24.280 14108-14108/com.nickDTaylor.cardioCapture D/BluetoothGatt: registerApp() - UUID=b10b5a6c-7a2b-445f-b6be-6f47476da642
2022-05-24 20:05:24.284 14108-14108/com.nickDTaylor.cardioCapture D/Connection: Device connecting 9F6CCE20
2022-05-24 20:05:24.286 14108-24421/com.nickDTaylor.cardioCapture D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
2022-05-24 20:05:24.833 14108-25496/com.nickDTaylor.cardioCapture D/Connection: AFter requesting connection
2022-05-24 20:05:27.487 14108-24421/com.nickDTaylor.cardioCapture D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=C6:10:34:8D:07:E6
2022-05-24 20:05:27.498 14108-14108/com.nickDTaylor.cardioCapture D/Connection: Device connected 9F6CCE20
2022-05-24 20:05:27.837 14108-25496/com.nickDTaylor.cardioCapture D/RECONNECT: Reconnected
2022-05-24 20:05:28.099 14108-14108/com.nickDTaylor.cardioCapture D/BluetoothGatt: discoverServices() - device: C6:10:34:8D:07:E6
2022-05-24 20:05:28.116 14108-24421/com.nickDTaylor.cardioCapture D/BluetoothGatt: onSearchComplete() = Device=C6:10:34:8D:07:E6 Status=0
2022-05-24 20:05:28.119 14108-14108/com.nickDTaylor.cardioCapture D/Connection: HR Feature ready 9F6CCE20
2022-05-24 20:05:28.122 14108-14108/com.nickDTaylor.cardioCapture D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a37-0000-1000-8000-00805f9b34fb enable: true
2022-05-24 20:05:28.129 14108-24421/com.nickDTaylor.cardioCapture D/BluetoothGatt: configureMTU() - device: C6:10:34:8D:07:E6 mtu: 512
2022-05-24 20:05:28.438 14108-24421/com.nickDTaylor.cardioCapture D/BluetoothGatt: setCharacteristicNotification() - uuid: fb005c81-02e7-f387-1cad-8acd2d8df0c8 enable: true
2022-05-24 20:05:28.779 14108-24421/com.nickDTaylor.cardioCapture D/BluetoothGatt: onPhyUpdate() - status=0 address=C6:10:34:8D:07:E6 txPhy=2 rxPhy=2
2022-05-24 20:05:30.827 14108-24421/com.nickDTaylor.cardioCapture D/BluetoothGatt: setCharacteristicNotification() - uuid: fb005c82-02e7-f387-1cad-8acd2d8df0c8 enable: true
2022-05-24 20:05:30.973 14108-14108/com.nickDTaylor.cardioCapture D/HR: HR 77
2022-05-24 20:05:30.982 14108-14108/com.nickDTaylor.cardioCapture D/HR: HR 79
2022-05-24 20:05:32.925 14108-24421/com.nickDTaylor.cardioCapture D/BluetoothGatt: setCharacteristicNotification() - uuid: fb005c51-02e7-f387-1cad-8acd2d8df0c8 enable: true
2022-05-24 20:05:34.436 14108-14108/com.nickDTaylor.cardioCapture D/HR: HR 81
2022-05-24 20:05:34.581 14108-14108/com.nickDTaylor.cardioCapture D/HR: HR 82
2022-05-24 20:05:34.582 14108-14108/com.nickDTaylor.cardioCapture D/HR: HR 85
2022-05-24 20:05:34.584 14108-14108/com.nickDTaylor.cardioCapture D/HR: HR 86
2022-05-24 20:05:34.584 14108-27099/com.nickDTaylor.cardioCapture D/BluetoothGatt: setCharacteristicNotification() - uuid: fb005c52-02e7-f387-1cad-8acd2d8df0c8 enable: true
2022-05-24 20:05:35.216 14108-14108/com.nickDTaylor.cardioCapture D/FileTransfer: Polar FTP ready 9F6CCE20
2022-05-24 20:05:35.217 14108-14108/com.nickDTaylor.cardioCapture D/HR: HR 88
2022-05-24 20:05:35.995 14108-1067/com.nickDTaylor.cardioCapture D/Connection: Streaming feature is ready: ACC
2022-05-24 20:05:35.995 14108-14108/com.nickDTaylor.cardioCapture D/HR: HR 89
2022-05-24 20:05:35.996 14108-1067/com.nickDTaylor.cardioCapture D/Connection: Streaming feature is ready: ECG
2022-05-24 20:05:36.044 14108-14108/com.nickDTaylor.cardioCapture D/Connection: Battery level 9F6CCE20 90%
2022-05-24 20:05:36.044 14108-14108/com.nickDTaylor.cardioCapture D/BATTERY: LEVEL received from strap
2022-05-24 20:05:37.019 14108-14108/com.nickDTaylor.cardioCapture D/HR: HR 90
2022-05-24 20:05:37.117 14108-14108/com.nickDTaylor.cardioCapture D/ECG: Ecg stream failed in ECGStreamHandler with com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command error:  failed with error: ERROR_INVALID_MTU

KennethEvans, 2022-05-24:
As an update, my app is more or less working so I am not currently developing it. Consequently I am not getting ERROR_INVALID_MTU. They were infrequent in any event.

JOikarinen, 2022-05-27:
Thanks for the update @KennethEvans. I close the bug for now, let's reopen the issue if we get new observations.

closed 2022-05-27 by JOikarinen