polarofficial/polar-ble-sdk #223: Accelerometer raw data format question

wilbertmatthew, 2022-01-12:
I am using the Polar 10 SDK and have written software to continuously stream the Polar 10 Accelerometer Z axis data, which comes from my chest movements. The data I am receiving, when sitting in a chair, is in the 650-670 value range. I have no idea what those values mean and can't any accelerometer documentation. I am streaming ECG data and receiving and plotting +/- data values. I would like to plot accelerometer +/- values against ECG data.

JOikarinen, 2022-01-13:
Hi @wilbertmatthew,

[accelerometer data](https://github.com/polarofficial/polar-ble-sdk/blob/11159a41b7ba84f42bec3bbfa006c729cf7924b2/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarAccelerometerData.java#L11) comes on with three axises x,y and z. The unit is millig and it includes gravity. 

[image]

So, when you are sitting in a chair, then your acceleration is 0 to all the directions. But as the each axis includes the gravity component you see some value on each axis, like 650-670 on z axis in your case. 

If you are wearing the strap and H10 is attached and you are sitting on you chair, I can assume that your chair is slightly tilt backwards as you see values around +650 millig on Z axis.

wilbertmatthew, 2022-01-13:
Thank you for your informative response. You are correct about my setup test. Any comment about how I can use the Z axis in the sitting chair position to return the vibrations from my heart chest movement?

JOikarinen, 2022-01-14:
@wilbertmatthew I am not sure did I get your question right. What you can do, is that you may analyse and filter the received data for acceleration changes on the axis you are interested. However, it is beyond my expertise what is the correct filter, if you want to filter the out the gravity component for example.

jimmyzumthurm, 2023-09-08:
As you didn't give an update for over a year, I will close this ticket now. @wilbertmatthew  Please feel free to open a new one if you still have issues related to that.

closed 2023-09-08 by jimmyzumthurm