polarofficial/polar-ble-sdk #225: ecg timestamps return many same values fro different ecg data values

wilbertmatthew, 2022-01-13:
I am streaming ECG data and plotting the values. I can confirm, through debugging and data plotting, that I am receiving a range of different ECG data values. I use Java System.currentTimeMillis() to get a timestamp for each data value as a test that each value has a different timestamp. When I use ecgDisposable =
                    api.requestStreamSettings(deviceId, PolarBleApi.DeviceStreamingFeature.ECG)
                            .toFlowable()
                            .flatMap((Function<PolarSensorSetting, Publisher<PolarEcgData>>) sensorSetting -> 
                             api.startEcgStreaming(deviceId, sensorSetting.maxSettings()))                        
                            .observeOn(Schedulers.newThread())
                            .subscribe(
                                    polarEcgData -> {
                                        for (Integer data : polarEcgData.samples) {
                                            long ecgTimestamp = polarEcgData.timeStamp/1000000; // convert nano seconds to milliseconds
                                           sendData(((float) data / 1000.0, ecgTimestamp);
                                        } catch (Exception ex) {
                                                ex.printStackTrace();
                                            }                                         
                                        } 
to get the ECG api timestamp I get different (about 30) ECG data values returned in succession with the same ECG timestamp.

How can I get a different timestamp for each data value using the ECG timestamp API?

JOikarinen, 2022-01-14:
@wilbertmatthew I understand your question and your dilemma. Here is the explanation:

Each of the `ECG` data package you receive from the H10 sensor may contain [multiple samples](https://github.com/polarofficial/polar-ble-sdk/blob/11159a41b7ba84f42bec3bbfa006c729cf7924b2/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarEcgData.java#L17) but always one [timestamp](https://github.com/polarofficial/polar-ble-sdk/blob/11159a41b7ba84f42bec3bbfa006c729cf7924b2/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarEcgData.java#L22). As [documentation](https://github.com/polarofficial/polar-ble-sdk/blob/11159a41b7ba84f42bec3bbfa006c729cf7924b2/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarEcgData.java#L20) says, the timestamp represents the time of the last sample. In your scenario you may calculate the timestamps for the previous samples found in the data package, that can be achieved for example by using the sampling frequency of ECG data. The sampling frequency is the one which you requested in  `api.startEcgStreaming(deviceId, sensorSetting.maxSettings()))`  by `sensorSetting.maxSettings()`.

The reason why there is no timestamp for each of the sample is the attempt to reduce the data flow over BLE.

wilbertmatthew, 2022-01-14:
Thanks so much for your response. I understand. Is there a Java example of how to view the ECG settings and make changes to them? I have seen the Kotlin version DialogUtility, but I have no understanding of how to mix Kotlin with Java.

wilbertmatthew, 2022-01-20:
I get the sample frequency by 1/131 hz. How do I use it to calculate the timestamps for the previous values?

JOikarinen, 2022-01-20:
Here is thread where this is discussed: https://github.com/polarofficial/polar-ble-sdk/issues/189#issuecomment-917356141 Let me know did it help you.

wilbertmatthew, 2022-01-21:
Thanks it need help.

closed 2022-12-13 by JOikarinen