polarofficial/polar-ble-sdk #226: no PPG data for verity sense with python

Knusbert, 2022-01-15:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Hi,
I try to read a ppg stream from a polar verity sense device with python 3.8.8. I looks like it behaves as expected but I don't get any data from the ppg stream. I checked the input format via 
```
PPG_SETTING = bytearray([0x01, #get settings
                         0x01]) #type PPG
PMD_CHAR = 'FB005C81-02E7-F387-1CAD-8ACD2D8DF0C8'

def notification_handler(sender, data):
        print(', '.join('{:02x}'.format(x) for x in data))

await client.start_notify(PMD_CHAR,  notification_handler)
await client.write_gatt_char(PMD_CHAR,PPG_SETTING, response=True)
await client.stop_notify(PMD_CHAR)

> f0, 01, 01, 00, 00, 00, 01, 87, 00, 01, 01, 16, 00, 04, 01, 04
```
This response looks like expected and I used this settings as PPG_SETTING_ to start the stream which response I also checked. 
```
await client.start_notify(PMD_CHAR, notification_handler)
await client.write_gatt_char(PMD_CHAR,PPG_SETTING, response=True)
await client.stop_notify(PMD_CHAR)

> f0, 02, 01, 00, 00, 05, 01, 00, 00, 80, 3f
```
However, if I try to read the data stream I don't get any data.
```
PMD_DATA = 'FB005C82-02E7-F387-1CAD-8ACD2D8DF0C8'

  await client.start_notify(PMD_DATA, notification_handler)
  n = 135
  while n<500:
      await asyncio.sleep(1)
      n = n + 135
  await client.stop_notify(PMD_DATA)
```
I double checked with a polar watch and it gave me a HR reading. Any idea why this is not giving anything back with the python code?

JOikarinen, 2022-01-17:
Hi @Knusbert,

I've checked what are the responses, if I use the example app to read the PPG settings and then start the PPG stream. I can confirm that your 

> f0, 01, 01, 00, 00, 00, 01, 87, 00, 01, 01, 16, 00, 04, 01, 04
- the response for PPG Settings read looks okay

> f0, 02, 01, 00, 00, 05, 01, 00, 00, 80, 3f
- the response for PPG stream start looks okay

Next what shall happen are the notifications sent by Verity Sense on PMD_DATA characteristics.  I am not familiar with python, so I cannot say much about you Python code snippets.

JOikarinen, 2022-01-17:
Btw, [here](https://github.com/pareeknikhil/biofeedback/blob/master/Polar%20Device%20Data%20Stream/Accelerometer/main.py) is another python solution @nikhilpareek149. I wonder may that give you any help?

Knusbert, 2022-01-19:
Hi, thanks for the hint. I'm aware of this solution and took this as a guide for my code. I already compared my code which is basically the same as mine:
    ```
att_read = await client.read_gatt_char(PMD_CONTROL)
await client.write_gatt_char(PMD_CONTROL, ACC_WRITE)
await client.start_notify(PMD_DATA, data_conv)
```
the code from @nikhilpareek149 was written for a Polar H10 but from what I understood it should be the same for the verity sense (besides that the H10 as ECG)

JOikarinen, 2022-01-20:
> the code from @nikhilpareek149 was written for a Polar H10 but from what I understood it should be the same for the verity sense (besides that the H10 as ECG)
Yes, start of ECG stream compared to start of PPG stream shall be very similar.

dkang55, 2022-03-21:
Hi @EyescannerJE  @Knusbert @JOikarinen was there a resolution to this problem by chance? Its seems that issue comes from await client.read_gatt_char( ) reading of PMD_CHAR = 'FB005C81-02E7-F387-1CAD-8ACD2D8DF0C8', which is not returning the correct byte-array/att_read.

EyescannerJE, 2022-03-21:
Hi, @dkang55 unfortunately not. I figured that the verity sense needed different settings for the PPG stream than the H10. But the documentation is lacking on this.

Knusbert, 2022-03-21:
Hi, I haven't succeeded sofar. I tried both the PPG and ECG commands on the verity (also ACC) but nothing worked. As mentioned above the communication with the device works and I do get a reply when I ask for the input format. Only the stream reading does not seem to work. I also disabled my watch and my mobile devices which could connect to the verity sense and block the stream.

EyescannerJE, 2022-03-21:
I actually got ACC data streaming to work for a short period of time but then it stopped again and I don't know why.

dkang55, 2022-03-21:
Many thanks for your responses @EyescannerJE @Knusbert. Would you guys be willing to share the code you are using to try to debug? I will also try on my end as well and can report back to see if I get a different output.

kt-eyescanner, 2022-03-22:
Hi @dkang55 I work with @EyescannerJE  and this is the code we use for debugging.

```python
import asyncio
import math
import os
import signal

import sys
import time

import pandas as pd
from bleak import BleakClient
from bleak.uuids import uuid16_dict
import matplotlib.pyplot as plt
import matplotlib

""" Predefined UUID (Universal Unique Identifier) mapping are based on Heart Rate GATT service Protocol that most
Fitness/Heart Rate device manufacturer follow (Polar H10 in this case) to obtain a specific response input from 
the device acting as an API """

uuid16_dict = {v: k for k, v in uuid16_dict.items()}

## This is the device MAC ID, please update with your device ID
ADDRESS = 'A0:9E:1A:A2:DC:59'

## UUID for model number ##
MODEL_NBR_UUID = "00002a24-0000-1000-8000-00805f9b34fb"
## UUID for manufacturer name ##
MANUFACTURER_NAME_UUID = "0000{0:x}-0000-1000-8000-00805f9b34fb".format(
    uuid16_dict.get("Manufacturer Name String")
)

## UUID for battery level ##
BATTERY_LEVEL_UUID = "0000{0:x}-0000-1000-8000-00805f9b34fb".format(
    uuid16_dict.get("Battery Level")
)

## UUID for connection establsihment with device ##
PMD_SERVICE = "FB005C80-02E7-F387-1CAD-8ACD2D8DF0C8"

## UUID for Request of stream settings ##
PMD_CONTROL = "FB005C81-02E7-F387-1CAD-8ACD2D8DF0C8"

## UUID for Request of start stream ##
PMD_DATA = "FB005C82-02E7-F387-1CAD-8ACD2D8DF0C8"

## UUID for Request of PMD Stream ##
#PMD_WRITE = bytearray([0x02,    0x02,    0x00, 0x01, 0x34,    0x00,    0x01,    0x01,    0x10,    0x00,    0x02,    0x01,    0x08,    0x00, 0x04, 0x01, 0x03 ])
#PMD_WRITE = bytearray([0x02, 0x01, 0x00, 0x01, 0x34, 0x00, 0x01, 0x01, 0x10, 0x00, 0x02, 0x01, 0x08, 0x00, 0x04, 0x01, 0x03])
#PMD_WRITE = bytearray([0xF0,0x01,0x00,0x00,0x00,0x00,0x01,0x82,0x00,0x01,0x01,0x0E,0x00])

# ## UUID for Request of ACC Stream ##
ACC_WRITE = bytearray(
    [0x02,
    0x02,
    0x00,
    0x01,
    0x34,
    0x00,
    0x01,
    0x01,
    0x10,
    0x00,
    0x02,
    0x01,
    0x08,
    0x00,
    0x04,
    0x01,
[... 142 lines cut ...]
    print("[CLOSED] application closed.")

    sys.exit(0)

async def main():
    try:
        async with BleakClient(ADDRESS) as client:
            signal.signal(signal.SIGINT, keyboardInterrupt_handler)
            tasks = [
                asyncio.ensure_future(run(client, True)),
            ]

            await asyncio.gather(*tasks)
    except Exception as error:
        print(error)
        #pass

if __name__ == "__main__":
    print('statement')
    os.environ["PYTHONASYNCIODEBUG"] = str(1)
    loop = asyncio.new_event_loop()
    asyncio.set_event_loop(loop)
    loop.run_until_complete(main())

```

marianyamukuru, 2023-01-12:
@Knusbert @dkang55 @johan-sightic @kt-eyescanner  I am having the same issue and was wondering if you were able to solve it?

sivaprasadravi, 2023-07-05:
Hi @Knusbert @dkang55  I am using polar verity sense. For me I want to connect the sensor to my laptop using bluetooth and I want to get the live heart rate data in python. Is it possible for you to get a code for this?

closed 2023-10-12 by samulimaa