polarofficial/polar-ble-sdk #228: RR Interval Data Interpretation

wilbertmatthew, 2022-01-23:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

wilbertmatthew, 2022-01-23:
I cannot find any documentation that explains how to interpret heart rate RR interval data values. I believe that that the rrsMS list contains RR Interval values for every one second heart beat? Why are there two values returned in the list and how are they interpreted? I can see one being the RR Interval in ms between two R peaks, but not a need for a list of values.

How are the two values interpreted that are returned in the rrs list? Are they the mV values for each R peak and to be divide by 1/1024? Also I can see a need for one value but not a list of values.

KennethEvans, 2022-01-24:
That's the way the Bluetooth Low Energy specification works.  You get one packet about once a second from the device. If the heart rate is fast, there can be more than one RR interval in that interval. Sometimes there aren't any. The problem is that there are no timestamps, so you have to decide how to arrange these intervals in time.

The RR values from the device are in 1/1024 sec units.

Yes, you would like one value every time there is an R peak, but that's not what you get.  The RR intervals are what is important. The HR is some sort of a running average of the inverse of the RR intervals. It doesn't happen at a particular moment in time.

closed 2022-12-13 by JOikarinen

GameDes1, 2024-07-14:
@zmukash did you find out how the sequence of capturing, respectively the saving is?