polarofficial/polar-ble-sdk #229: Not receiving Polar H10 battery level on iOS

Rexios80, 2022-01-26:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Battery level received callback is never called. It works on Android.

The code that isn't working: https://github.com/Rexios80/polar/blob/master/ios/Classes/SwiftPolarPlugin.swift

Possibly relevant past issues:
 - https://github.com/polarofficial/polar-ble-sdk/issues/130
 - https://github.com/polarofficial/polar-ble-sdk/issues/39

Those are old enough (and closed) that they're probably irrelevant, but it's weird that I'm having the same issue.

**How to Reproduce**
Run on iOS and attempt to get battery level notifications

**Expected behavior**
Battery level is properly reported

JOikarinen, 2022-01-27:
the [SwiftPolarPlugin](https://github.com/Rexios80/polar/blob/master/ios/Classes/SwiftPolarPlugin.swift) is not implementing protocol `PolarBleApiDeviceInfoObserver`

I guess these changes will fix your problem:

Make `SwiftPolarPlugin` to implement `PolarBleApiDeviceInfoObserver` protocol:

```
public class SwiftPolarPlugin: PolarBleApiDeviceInfoObserver,
    ...
{
```

and then in init
```
  init(channel: FlutterMethodChannel) {
        ...
        api.deviceInfoObserver = self
        ...
    }
```

Once those changes done, both `batteryLevelReceived` and `disInformationReceived` callbacks shall work.

Rexios80, 2022-01-27:
Interesting. Is that a new thing or did I just miss it?

JOikarinen, 2022-01-27:
Maybe you just missed it, it has been there for a while. 

The SDK splits the API to multiple Protocols. Also the  `PolarBleApiPowerStateObserver` may give the impression that it is related to battery status, but it is only related on phone Bluetooth power states.

Rexios80, 2022-01-27:
That fixed it. Thanks!

closed 2022-01-27 by Rexios80