polarofficial/polar-ble-sdk #230: How to record acc, gyro and mag on device

datamagnus, 2022-01-27:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I was wondering how I can modify the Android example-app to record the streams in SDKmode to .txt file on my android device?
The data should also include the time stamp and maybe the internal sensor clock.
My plan was to add a extra button to start recording the streams if they are active on my device. If it is easier to code I was thinking of just start recording when the stream is activated. Then later export them to the computer for post processing of the data. Unfortunately I'm not too familiar with the rxjava, rxandroid and the Kotlin language, but I was thinking of using outputstream, openFiloutput and .write form java.io. I still didn't manage to succeed with the task if it is even possible.
Could you help and push me in the right direction in this case? 
Maybe you even have a simple solution that I just can append to the code as I down think this should be to hard for someone experienced.

Thanks.

orestesgaolin, 2022-09-27:
We also would love to be able to record the data on the device for further retrieval. The practical example would be to be able to collect the data even when the user is outside of the Bluetooth range of their phone.

Could you clarify @JOikarinen if this is even possible? I don't see any mention of that in the SDK mode documentation nor in the [1.1.5](https://support.polar.com/en/updates/polar-verity-sense-11-firmware-update?product=Verity+Sense) update announcement.

JOikarinen, 2022-09-28:
Hi @orestesgaolin,

> Could you clarify @JOikarinen if this is even possible?
- unfortunately, that's not possible. The data stream is closed when the Bluetooth connection disconnects. There is no mechanism to save the data during user out of range with VeritySense.
- in Polar H10 however, there is [offline recording](https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L289) for Heart rate and RR data

JOikarinen, 2022-09-28:
I am closing this issue. The original question was about the possibility to add file save to example apps. For now there is no plans to add data logging/data save functionality to example apps to keep example apps as tiny and simple as possible without mixing android/iOS platform specific features.

closed 2022-09-28 by JOikarinen