polarofficial/polar-ble-sdk #231: Polar BLE SDK example app can not connect to H10

pencili, 2022-01-30:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I'm trying to use the Polar SDK Android example app with my Polar H10 HRM but to no success. I have tried the sample with both Android Studio and IntelliJ with multiple Android phones but they all have the same problem which is the app can’t connect to the H10.

I have updated the code to match the H10 Device ID and whilst all my phones can pair with the H10 when I debug the app on the phone “nothing happens” when I press the “connect to” button and the debug output shows:

D/BluetoothAdapter: isLeEnabled(): ON
D/API LOGGER: BDScanCallback/commandState state:IDLE action: ENTRY
D/API LOGGER: BDScanCallback/commandState state:IDLE action: CLIENT_START_SCAN
D/API LOGGER: BDScanCallback/commandState state:IDLE action: EXIT
D/API LOGGER: BDScanCallback/commandState state:SCANNING action: ENTRY
D/API LOGGER: BDScanCallback/timestamps left: 2
D/API LOGGER: BDScanCallback/Scan started -->
D/BluetoothAdapter: isLeEnabled(): ON
D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=6 mScannerId=0
D/API LOGGER: BDScanCallback/Scan started <--

when I press the “Scan Devices” button the debug output just shows:
D/API LOGGER: BDScanCallback/commandState state:SCANNING action: CLIENT_START_SCAN

Does anyone have any suggestions that I could try? 

TIA

JOikarinen, 2022-01-31:
Hi @pencili, 

I haven't been able reproduce the problem you describe, but let me explain detailed steps what I do when creating the connections with the example app:
1. On the phone enable the bluetooth 
2. On the phone enable the location (Required in Android 11 and older, since Android 12 this step is not needed anymore ) 
3. In the example app code, I change [deviceId](https://github.com/polarofficial/polar-ble-sdk/blob/9703eb54cb275da6ff29db8c2ed5dea2a9c67fdf/examples/example-android/androidBleSdkTestApp/app/src/main/java/com/polar/androidblesdk/MainActivity.kt#L36) to one printed on my H10 sensor (in my case 968BEA2E) and then install the app to my Android phone
4. Once application launches, I accept the needed permissions
5. I wear the H10, using the strap, to turn H10 on
6. On example app I press "Connect to 968BEA2E" 

I can produce similar logcat logs like you have, if I miss the step 2 or 5. 

Furthermore, you mention `all my phones can pair with the H10`. You probably mean that you can pair the H10 from the phone bluetooth settings? With BLE devices it is not needed to pair or connect devices from phone bluetooth settings, but shouldn't cause any harm either. However, it brings my mind possibility that your H10 is connected to another phone while you try the Polar SDK example in other phone, could that be possible?

JOikarinen, 2022-12-13:
Closing the issue. Please don't hesitate to reopen if needed.

closed 2022-12-13 by JOikarinen

Pavan-Bellam, 2024-06-03:
I am facing the same issue.

My logs:

2024-06-03 19:17:19.067  4640-4640  MainActivity            com.polar.androidblesdk              D  CONNECTING: 83E7B82B
2024-06-03 19:17:22.634 16339-4991  NearbyMediums           com.google.android.gms.persistent    I  Found Fast Ble Advertisements :
                                                                                                    BleAdvertisement { version=2, socketVersion=2, isFast=true, serviceIdHash=null, data=[ 0x23 0x49 0x36 0x34 0x41 0x11 0x32 0xe3 0xa3 0x9a 0xba 0xcf 0x16 0x88 0xa7 0xbf 0x6a 0x21 0xd4 0xbf 0xa9 0xcf 0xc9 ], deviceToken=[ 0x43 0x5b ], rxAdvertisement=null } : 1 times.
                                                                                                    Total 1 fast advertisements.
2024-06-03 19:17:22.637 16339-3847  NearbyConnections       com.google.android.gms.persistent    I  Processing lost BlePeripheral BlePeripheral{advertisement=BleAdvertisement { version=2, socketVersion=2, isFast=true, serviceIdHash=null, data=[ 0x23 0x46 0x52 0x43 0x42 0x11 0x32 0xa6 0x09 0x88 0x0e 0x95 0x64 0xac 0xac 0x03 0x3a 0x10 0x17 0x50 0xed 0xaf 0x19 ], deviceToken=[ 0x88 0x37 ], rxAdvertisement=null }, address=null, psm=0, deviceToken=<ByteString@39ec218 size=2 contents="\2107">, isSecondProfile=false, rxInstantConnectionAdv=null}.
2024-06-03 19:17:22.637 16339-3847  NearbyConnections       com.google.android.gms.persistent    I  Lost BleEndpoint for BlePeripheral BlePeripheral{advertisement=BleAdvertisement { version=2, socketVersion=2, isFast=true, serviceIdHash=null, data=[ 0x23 0x46 0x52 0x43 0x42 0x11 0x32 0xa6 0x09 0x88 0x0e 0x95 0x64 0xac 0xac 0x03 0x3a 0x10 0x17 0x50 0xed 0xaf 0x19 ], deviceToken=[ 0x88 0x37 ], rxAdvertisement=null }, address=null, psm=0, deviceToken=<ByteString@39ec218 size=2 contents="\2107">, isSecondProfile=false, rxInstantConnectionAdv=null} (with EndpointId FRCB and EndpointInfo [ 0x32 0xa6 0x09 0x88 0x0e 0x95 0x64 0xac 0xac 0x03 0x3a 0x10 0x17 0x50 0xed 0xaf 0x19 ]).
2024-06-03 19:17:22.641 16339-4992  NearbySharing           com.google.android.gms.persistent    I  (REDACTED) Ignoring lost endpoint %s because we don't have an associated ShareTarget
2024-06-03 19:17:22.642 16339-4991  NearbySharing           com.google.android.gms.persistent    I  (REDACTED) Endpoint %s lost over Nearby Connections
2024-06-03 19:17:22.643 16339-3847  NearbyConnections       com.google.android.gms.persistent    I  ClientProxy(29104412) reporting onEndpointLost(FRCB)
2024-06-03 19:17:24.277 16339-4991  NearbyConnections       com.google.android.gms.persistent    I  Found BleAdvertisement [ 0x23 0x46 0x52 0x43 0x42 0x11 0x32 0xa6 0x09 0x88 0x0e 0x95 0x64 0xac 0xac 0x03 0x3a 0x10 0x17 0x50 0xed 0xaf 0x19 ] (with EndpointId FRCB and EndpointInfo [ 0x32 0xa6 0x09 0x88 0x0e 0x95 0x64 0xac 0xac 0x03 0x3a 0x10 0x17 0x50 0xed 0xaf 0x19 ])
2024-06-03 19:17:24.281 16339-4991  NearbyConnections       com.google.android.gms.persistent    I  ClientProxy(29104412) reporting onEndpointFound(FRCB)
2024-06-03 19:17:24.282 16339-4991  NearbyConnections       com.google.android.gms.persistent    I  ClientProxy(29104412) reporting onEndpointDistanceChanged(FRCB, UNKNOWN)
2024-06-03 19:17:24.283 16339-4991  NearbyConnections       com.google.android.gms.persistent    I  No Bluetooth Classic MAC address found in [ 0x23 0x46 0x52 0x43 0x42 0x11 0x32 0xa6 0x09 0x88 0x0e 0x95 0x64 0xac 0xac 0x03 0x3a 0x10 0x17 0x50 0xed 0xaf 0x19 ]
2024-06-03 19:17:24.284 16339-3847  NearbySharing           com.google.android.gms.persistent    I  (REDACTED) Endpoint %s discovered over Nearby Connections
2024-06-03 19:17:24.294 16339-4992  NearbySharing           com.google.android.gms.persistent    I  (REDACTED) Failed to convert given advertisement (%s)%s to share target from discovered advertisement. Ignoring.
2024-06-03 19:17:24.295 16339-4992  NearbySharing           com.google.android.gms.persistent    I  (REDACTED) Report endpoint %s as lost, since we cannot recognize this endpoint anymore
2024-06-03 19:17:24.296 16339-4992  NearbySharing           com.google.android.gms.persistent    I  (REDACTED) Ignoring lost endpoint %s because we don't have an associated ShareTarget
2024-06-03 19:17:27.639 16339-4992  NearbyMediums           com.google.android.gms.persistent    I  Found Fast Ble Advertisements :
                                                                                                    BleAdvertisement { version=2, socketVersion=2, isFast=true, serviceIdHash=null, data=[ 0x23 0x46 0x52 0x43 0x42 0x11 0x32 0xa6 0x09 0x88 0x0e 0x95 0x64 0xac 0xac 0x03 0x3a 0x10 0x17 0x50 0xed 0xaf 0x19 ], deviceToken=[ 0x88 0x37 ], rxAdvertisement=null } : 1 times.
                                                                                                    BleAdvertisement { version=2, socketVersion=2, isFast=true, serviceIdHash=null, data=[ 0x23 0x49 0x36 0x34 0x41 0x11 0x32 0xe3 0xa3 0x9a 0xba 0xcf 0x16 0x88 0xa7 0xbf 0x6a 0x21 0xd4 0xbf 0xa9 0xcf 0xc9 ], deviceToken=[ 0x43 0x5b ], rxAdvertisement=null } : 1 times.
                                                                                                    Total 2 fast advertisements.
2024-06-03 19:17:29.617  4640-4640  MainActivity            com.polar.androidblesdk              D  DISCONNECTED: 83E7B82B