polarofficial/polar-ble-sdk #232: HTML Javascript utility for combined H10 and Sense streams

cjs30, 2022-02-03:
I thought it might be useful to share where I have got to so far in trying to collect H10 and Sense data streams for a small project I am working on. The program uses Javascript within HTML and runs on Chrome and uses the Web Bluetooth interface together with Plotly to display ECG, PPG and accelerometer data (reduced using a Ramer-Douglas-Peucker algorithm). It took some effort to get the processor load low enough to work on relatively slow machines. The program lacks meaningful documentation and was written for a student project. I have not put any effort into making it user friendly and Bluetooth disconnects etc can result in the program falling-over. The Chrome console provides some feedback. As I keep saying, I am not a programmer - but, I am always open to suggestions for additions and improvements.

The program is a single file and can be found here: https://github.com/cjs30/FingerPulseLatency

samulimaa, 2023-10-12:
Nice program.

closed 2023-10-12 by samulimaa

brauliobo, 2025-02-03:
another one is still in development, straightforward and easy to read H10 code at https://github.com/brauliobo/fitron/blob/main/src/platforms/web_bluetooth/devices/polar/h10.js