polarofficial/polar-ble-sdk #233: ACC & ECG stream failed: notificationNotEnabled

hwixley, 2022-02-03:
***Platform your question concerns:*** 
- [ ] Android 
- [x ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Specs:**
SDK version: 3.2.5
Firmware version: 3.1.1
Xcode version: 13.2.1
Swift version: 5.5.2

**Description:**
I have created a SwiftUI app with my own PolarBleSdkManager class that uses allFeatures. After successfully connecting to my device I get the following output in the console:
```
ECG stream failed: notificationNotEnabled
ACC stream failed: notificationNotEnabled
```

I have successfully integrated this SDK into a Storyboard app before that didn't produce these errors when trying to stream ACC and ECG data so am quite confused as to why this may be happening. The only difference between my classes across these apps is that I removed some unneeded functions (such as the ppgToggle, ppiToggle, setTime and all the recording methods).

Do you know why this error is occuring? Or how I could go about fixing it?

Any help would be greatly appreciated!

JOikarinen, 2022-02-04:
Hi @hwixley,

from you explanation I understand that you are familiar with the [sample app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-ios/iosBleSdkTestApp), without any modifications is the sample app working for you? I tried your steps with the  [sample app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-ios/iosBleSdkTestApp), but I couldn't reproduce the problem. 

`notificationNotEnabled` may indicate that connected device is not having streaming service for ACC and ECG, or the initialisation of those services has failed for some reason. I must admit, the error description is not the most self explanatory in this case.

hwixley, 2022-02-04:
I am using the same PolarBleSdkManager class used in my Storyboard app (that worked) which only has a few small differences than the sample app (however these should not affect the data streams).

As soon as I connect to the H10 device I try to start streaming these features, and this error displays. Thus it seems more like an initialization error as you mentioned. It is also particularly weird as the `streamingFeaturesReady` function always outputs the following lines a bit after the stream failure logs:
```
Feature ecg is ready.
Feature acc is ready.
```

Does this suggest that these streams failed initially but are then sorted out?

hwixley, 2022-02-10:
Okay I managed to fix the issue!

I ended up calling `accToggle()` and `ecgToggle()` immediately after my app connected to my device. However due to the fact this feature had not been marked as 'ready' yet, these stream requests failed. If you just toggle these streams after the given feature is classified as 'ready' then it should work

closed 2022-02-10 by hwixley