polarofficial/polar-ble-sdk #235: Stopper: FATAL EXCEPTION: RxNewThreadScheduler-3

KennethEvans, 2022-02-09:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

I made a small change to an output file format, and when I went to run it I got a crash.  The details are in the attached log (includes API logging). The exception is:

`W/System.err: io.reactivex.rxjava3.exceptions.UndeliverableException: The exception could not be delivered to the consumer because it has already canceled/disposed the flow or the exception has nowhere to go to begin with. Further reading: https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling | java.lang.NoClassDefFoundError: Failed resolution of: Lorg/apache/commons/io/IOUtils;`

It was not to a point where the changes would matter, and they were very minor. It was working yesterday.

The other thing I had done was upgrade to Android Studio (Bumblebee 2021.1.1) Patch 1 (Feb 4) (when it prompted to do so.).

I have since tried reverting to Android Studio (Bumblebee 2021.1.1)  (Jan 25).
That did not help.

I have now uninstalled the degraded AS version, and reinstalled from a download for Bumblebee 2021.1.1) Patch 1.
I have rebooted the computer and the phone.
It is still crashing.

I ran another Bluetooth app that uses the H10 but not the SDK. It works fine.

I am dead in the water. I need this app.

[Crash-2022-02-08.txt](https://github.com/polarofficial/polar-ble-sdk/files/8029041/Crash-2022-02-08.txt)

JOikarinen, 2022-02-09:
Do you have `commons-io` in dependencies?

```
dependencies {
   ...
    implementation 'commons-io:commons-io:2.10.0'
  ...
}
```

JOikarinen, 2022-02-09:
I noticed that latest Android Studio tries to force the commons-io to version [20030203.000550/](https://repo1.maven.org/maven2/commons-io/commons-io/20030203.000550/). That version is something very old. I tried the `20030203.000550` version and I get the same error you got. The bug is in the Android Studio or Gradle. The problem may also be the existing old version in Maven Central which should be removed by Apache.org maintainers.

closed 2022-02-09 by JOikarinen

KennethEvans, 2022-02-09:
> Do you have `commons-io` in dependencies?
> 
> ```
> dependencies {
>    ...
>     implementation 'commons-io:commons-io:2.10.0'
>   ...
> }
> ```

No.  I did, but the new Android Studio, probably the Gradle upgrade part, apparently changed it to:
`    implementation 'commons-io:commons-io:20030203.000550'
`

Thanks for finding it. I came up on this late last night and didn't do everything I should have. In particular, I should have diff'ed the Gradle files. It would have been obvious then.

KennethEvans, 2022-02-09:
I am now getting the Lint warning.

`Warning:(35, 20) A newer version of commons-io:commons-io than 2.10.0 is available: 20030203.000550`

It also says that in Project Structure.

Definitely an Android Studio bug.

It looks like the latest version is actually 2.11.0. I am using that now.  It is still saying 20030203.000550 is newer.