polarofficial/polar-ble-sdk #238: Polar H10 HR Monitor: Occasional slow exercise data fetch

phadenz, 2022-02-14:
***Platform your question concerns:*** 
- [ ] Android 

***Device:***
- [ ] Polar H10

**Description:**
Each semester, the University of Otago Exercise Sciences group uses your Polar H10 HR Monitors for a student data collection project. Each of 300 students wears a monitor for 24 hours, while keeping an activity diary. After this period, the stored data are downloaded from the monitor and used to teach students various data analysis techniques.

To facilitate the high volume of data retrievals from H10 units, we (University of Otago Scientific Programming) have built a small Android app that connects to the monitors via bluetooth, fetches the raw HR data, and formats it appropriately for student analysis. We use the Polar Mobile SDK, and perform this specific action with the api.fetchExercise hook.

Most of the time, data fetch for a single device takes 2 to 3 minutes. However, on occasion, the fetch appears to hang (or at least spins until the technician runs out of patience). When this occurs, the technician kills the app, and retries the download. Usually, the second (or occasionally third) attempt is successful and the entire data file downloads atomically in the expected two minutes. Between these repeated attempts, there is no movement of the H10 monitor, change of controlling device (an Android tablet), etc. It just works if you keep trying.

We have explored possible software solutions to the problem without success, and are wondering if you are aware of any possible hardware issues that could produce such behaviour? And if so, how best to address them?

JOikarinen, 2022-02-15:
Hi @phadenz, thank you for the good explanation.

If the problem is hardware problem, I think it shall happen always with the same particular H10 device. With your explanation I assume the problem happening randomly with any H10 device you have, which leads my thinking the problem being a software issue. 

Is there any chance you could log the situation and share the logs of the moment the bug happens? 
I would need the logs from the Polar BLE SDK. Here is how to enable the logs in:

 Java
```java
api.setApiLogger(s -> Log.d("API_LOGGER", s));
```
Kotlin

```kotlin
api.setApiLogger { s: String -> Log.d(API_LOGGER_TAG, s) }
```

phadenz, 2022-02-15:
Thanks very much for the suggestion. We will try this and see what turns up.

closed 2023-10-12 by samulimaa