polarofficial/polar-ble-sdk #239: How to get ${polarBroadcastData.hr} work in the background

khangdang4vn, 2022-02-15:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I'm trying to get the heart rate data in my application from this example code:
```
broadcastButton.setOnClickListener {
            if (!this::broadcastDisposable.isInitialized || broadcastDisposable.isDisposed) {
                toggleButtonDown(broadcastButton, R.string.listening_broadcast)
                broadcastDisposable = api.startListenForPolarHrBroadcasts(null)
                    .subscribe(
                        { polarBroadcastData: PolarHrBroadcastData ->
                            Log.d(TAG, "HR BROADCAST ${polarBroadcastData.polarDeviceInfo.deviceId} HR: ${polarBroadcastData.hr} batt: ${polarBroadcastData.batteryStatus}")
                        },
                        { error: Throwable ->
                            toggleButtonUp(broadcastButton, R.string.listen_broadcast)
                            Log.e(TAG, "Broadcast listener failed. Reason $error")
                        },
                        { Log.d(TAG, "complete") }
                    )
            } else {
                toggleButtonUp(broadcastButton, R.string.listen_broadcast)
                broadcastDisposable.dispose()
            }
        }
```
It works well in the foreground, but when the phone screen is off, it doesn't collect heart rate data in the background. When the phone screen is off, I cannot see the log in the logcat showing `HR: ${polarBroadcastData.hr}` anymore.
Meanwhile, the code below still works when the phone screen is off:
```
api.setApiCallback(object : PolarBleApiCallback() {
            override fun hrNotificationReceived(identifier: String, data: PolarHrData) {
                Log.d(TAG, "HR value: ${data.hr} rrsMs: ${data.rrsMs} rr: ${data.rrs} contact: ${data.contactStatus} , ${data.contactStatusSupported}")
            }
        })
```
I still can see ${data.hr} in the logcat. I have some questions:

1. How can I get ${polarBroadcastData.hr} work in the background?
2. What is the sampling rate of heart rate data? Is the sampling rate constant? I assume that it is 60 samples per minute but when I test it is always more than 60 samples collected per minute.
3. Is there any way that I can get the timestamps for the heart rate samples collected?

Thank you & best regards,
Khang Dang

JOikarinen, 2022-02-15:
Hi @khangdang4vn, 

good questions. I start the answering for 2) and 3). 

> * What is the sampling rate of heart rate data? Is the sampling rate constant? I assume that it is 60 samples per minute but when I test it is always more than 60 samples collected per minute.
> * Is there any way that I can get the timestamps for the heart rate samples collected?

I am not exact sure what do you mean by "sample rate" in your question, so let me give you few alternative answers:

**Answer A)** heart rate unit is beats per minute (bpm). When talking about heart rate the "sample rate" is difficult to define,  as the heart rate is not the type of data which is sampled. It is number of beats your heart does in one minute. Behind scenes the Verity Sense does PPG signal sampling to detect individual heartbeat and then lots of calculations to say what is the current heart rate in bpm.  Here is couple links which may be helpful: 
    - https://www.polar.com/blog/optical-heart-rate-tracking-polar/
    - https://www.polar.com/blog/optical-heart-rate-monitoring-polar-precision-prime/

**Answer B)**   `hrNotificationReceived` is the callback function which is called when new heart rate value is received from BLE heart rate service of the connected device, from Verity Sense in your case. This callback is roughly called once per second. That is because the latest heart rate value is sent by BLE heart rate service around once per second. That is by the definition of [BLE Heart Service specification](https://www.bluetooth.com/specifications/specs/heart-rate-service-1-0/): `In typical applications, the Heart Rate Measurement characteristic is notified approximately 1 time per second`

**Answer C)** `api.startListenForPolarHrBroadcasts.subscribe()` the observer parameter you give for the `subscribe()` function is called every time BLE broadcast data is received and the data includes the heart rate data. There is filtering happening behind scenes so that not all the BLE broadcast data received by your phone is delivered for the observer, only heart rate data from Polar sensors. Polar Verity Sense sends broadcast very fast, for example every 50ms (varying of the state of the Verity Sensor). So as you noticed, you may observe high frequency of calls to observer. Even more if you have multiple Polar sensors around doing BLE broadcasting.

JOikarinen, 2022-02-15:
> It works well in the foreground, but when the phone screen is off
- this is related to power save functionality of the Android operating system. See the Android documentation in [BluetoothLeScanner](https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner#startScan(android.bluetooth.le.ScanCallback)): `For unfiltered scans, scanning is stopped on screen off to save power`. So when the screen is off the scanning is stopped, if no filters are given. In Polar SDK there are helper methods you may use, see [backgroundEntered()](https://github.com/polarofficial/polar-ble-sdk/blob/4bd8808dc2eb3ef7de8131d206855b2c9071e600/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L165) and [foregroundEntered()](https://github.com/polarofficial/polar-ble-sdk/blob/4bd8808dc2eb3ef7de8131d206855b2c9071e600/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L170). If you test the [Android example app ](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp) you can notice that even the screen is off the broadcast is working. See  the onPause and onResume lifecycle callbacks in the example app [MainActivity](https://github.com/polarofficial/polar-ble-sdk/blob/master/examples/example-android/androidBleSdkTestApp/app/src/main/java/com/polar/androidblesdk/MainActivity.kt).

closed 2023-09-08 by jimmyzumthurm