polarofficial/polar-ble-sdk #240: Is there any way to add ScanFilters when searching for devices

thaiphandinh, 2022-02-15:
***Platform your question concerns:*** 
- [ x ] Android 

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I want to filter the search, something likes `setServiceUuid` so I would not receive unwanted results. But there is only one api to perform the search `searchForDevice` from `PolarBleApi`, and it provides no options to customise the search. The returned data is a flow of `PolarDeviceInfo` also doesn't have much information to do further filtering.

I took a look at `BDBleApiImpl` (concrete implementation of `PolarBleApi`) and found these implement:
```
private void enableAndroidScanFilter() {
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
            List<ScanFilter> scanFilter = new ArrayList<>();
            scanFilter.add(new ScanFilter.Builder().setServiceUuid(
                    ParcelUuid.fromString(BleHrClient.HR_SERVICE.toString())).build());
            scanFilter.add(new ScanFilter.Builder().setServiceUuid(
                    ParcelUuid.fromString(BlePsFtpUtils.RFC77_PFTP_SERVICE.toString())).build());
            listener.setScanFilters(scanFilter);
        }
    }

```
This method only be called from  
```
@Override
    public void backgroundEntered() {
        enableAndroidScanFilter();
    }
```
That begs me a question why do you do this Android filtering only when enter the background. May be a bit explanation would make thing clearer. (I also took a look at https://github.com/polarofficial/polar-ble-sdk/issues/11 but didn't help much)
Even though the `enableAndroidScanFilter` is not exactly what I want, so may I add a **feature request** to add `ScanFilter`

(I know it is a bit odd to ask this Polar-SDK to probably do something more than it is, but since this SDK also allows to search for devices other than polar and event connect and collect data from them.)

JOikarinen, 2022-02-16:
Hi @thaiphandinh,

let's start with explanation of currently available API's: 

**[searchForDevice()](https://github.com/polarofficial/polar-ble-sdk/blob/4bd8808dc2eb3ef7de8131d206855b2c9071e600/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L351):**
- when subscribed starts the BLE scan. The found BLE device is emit as [PolarDeviceInfo](https://github.com/polarofficial/polar-ble-sdk/blob/4bd8808dc2eb3ef7de8131d206855b2c9071e600/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarDeviceInfo.java#L7). BLE Scanning continues as long the observable is subscribed. The device found is emitted only once for subscriber.  

**[setPolarFilter()](https://github.com/polarofficial/polar-ble-sdk/blob/4bd8808dc2eb3ef7de8131d206855b2c9071e600/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L150):**

-  [searchForDevice()](https://github.com/polarofficial/polar-ble-sdk/blob/4bd8808dc2eb3ef7de8131d206855b2c9071e600/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L351) either emits only Polar devices or any BLE devices. This can be controlled by [setPolarFilter()](https://github.com/polarofficial/polar-ble-sdk/blob/4bd8808dc2eb3ef7de8131d206855b2c9071e600/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L150)
- if [setPolarFilter()](https://github.com/polarofficial/polar-ble-sdk/blob/4bd8808dc2eb3ef7de8131d206855b2c9071e600/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L150) is called with parameter true then searchForDevice() emits only Polar devices. 
- the default setting is true

**[foregroundEntered()](https://github.com/polarofficial/polar-ble-sdk/blob/4bd8808dc2eb3ef7de8131d206855b2c9071e600/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L170) and [backgroundEntered()](https://github.com/polarofficial/polar-ble-sdk/blob/4bd8808dc2eb3ef7de8131d206855b2c9071e600/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L165):** 
- these are related to power save functionality of the Android operating system. See the Android documentation in [BluetoothLeScanner](https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner#startScan(android.bluetooth.le.ScanCallback)): `For unfiltered scans, scanning is stopped on screen off to save power`. So when the screen is off the scanning is stopped, if no filters are given. 
- as you noticed ```backgroundEntered()``` adds the filter for scanner and that is done to keep scanning enabled even the screen is turned off.

thaiphandinh, 2022-02-16:
Hi @JOikarinen ,
Thanks for your clear explanation. Now it's clear to me why do you add `enableAndroidScanFilter` when `backgroundEntered`. However, some of my question are still remained: 

- You didn't answer my main question : 

> Is it possible to control search process by adding scanFilter? Or it should be a feature request?

- 

> as you noticed backgroundEntered() adds the filter for scanner and that is done to keep scanning enabled even the screen is turned off.

To me, it sounds like a HACK. It makes sense that you wanted to keep scanning even when the app is in the background. But by calling `enableAndroidScanFilter` it would make the scanning results different in comparison to when the app is in foreground, isn't it?

JOikarinen, 2022-02-17:
Sorry, for missing the main question. Let's get back to that. 

> Is it possible to control search process by adding scanFilter? Or it should be a feature request?
- at the moment that's not possible by SDK API functions. 
- what you can do, is that you filter search results with Rx filter, if that approach match with your use case

```kotlin
api.setPolarFilter(false)
api.searchForDevice()
      .filter { polarDeviceInfo ->
          polarDeviceInfo.name.contains("Polar")
      }
      .observeOn(AndroidSchedulers.mainThread())
      .subscribe(
          { polarDeviceInfo: PolarDeviceInfo ->
              Log.d(TAG, "polar device found id: " + polarDeviceInfo.deviceId + " address: " + polarDeviceInfo.address + " rssi: " + polarDeviceInfo.rssi + " name: " + polarDeviceInfo.name + " isConnectable: " + polarDeviceInfo.isConnectable)
          },
          { error: Throwable ->
              toggleButtonUp(scanButton, "Scan devices")
              Log.e(TAG, "Device scan failed. Reason $error")
          },
          { Log.d(TAG, "complete") }
      )

```

 
> To me, it sounds like a HACK. It makes sense that you wanted to keep scanning even when the app is in the background. But by calling `enableAndroidScanFilter` it would make the scanning results [...]
- yes and no. If we consider only Polar devices the search result is consistent. For non Polar devices yes.

commit dea4b1df (2022-02-21): Merge pull request #241 from polarofficial/#240-API-docs-updated-and-scanning-fix

#240 documentation enhanced and deprecated backgroundEntered function
closed 2022-02-21 by JOikarinen