polarofficial/polar-ble-sdk #243: Git repository history too big

tien, 2022-02-24:
Currently the repo size is roughly 590MB because a lot of Pods, Carthage & frameworks dependencies are committed into the history. This result in really slow clone time & affecting CI/CD times & cost.

I suggest trimming those in the repo history with the following command using [git-filter-repo](https://github.com/newren/git-filter-repo/blob/main/INSTALL.md):

```sh
git filter-repo --path-glob '*/Pods' --path-glob '*/Carthage/Build' --path-glob '*/*.framework' --path-glob '*/*.xcframework'  --invert-paths
```

JOikarinen, 2022-02-25:
Good proposal. However, I am not familiar with `filter_repo`, I noticed the command you suggested rewrites the history of the whole repository. I think,  it is not the best idea to push rewritten history, might confusing for the many users not that familiar with git.  Or did I miss some options in `filter-repo`?

tien, 2022-02-25:
Unfortunately the only way to remove the big files from history is to re-write them :( those files are not supposed to be committed in the first place anyway. 590MB is a lot though and I know some build service can't support repo that big.

tien, 2022-02-28:
@JOikarinen 3.2.9 is working now. Yeah, JitPack doesn't work reliably with repo bigger than 590MB. It will retry a couple of times before it's able to build the repo. Don't know why though

tien, 2022-02-28:
You would think the people at JitPack would use shallow clone instead of a full clone :))

JOikarinen, 2022-02-28:
> You would think the people at JitPack would use shallow clone instead of a full clone :))

Exactly.🤷🏻‍♂️ 😀

JOikarinen, 2022-04-20:
Finally decided to fix this issue. Above I mentioned separating the iOS and Android libraries to own repositories, but I didn't do that yet. For now I choose to remove big files from repository history with [git-filter-repo](https://github.com/newren/git-filter-repo/blob/main/INSTALL.md). 

BIG NOTE  the whole repository history is now rewritten. If you have cloned https://github.com/polarofficial/polar-ble-sdk and want to refresh you local clone history:
1. Easy case - you don't have local changes (or you are ok to delete your local changes):  you can delete you current clone and just take a new fresh clone. Or you can run `git reset --hard @{u}` on your current clone to reset your clone. 
2. Advanced case - you have local changes and you want to keep them and you want update the local clone with new history from https://github.com/polarofficial/polar-ble-sdk: Please follow the instructions given in [RECOVERING FROM UPSTREAM REBASE](https://htmlpreview.github.io/?https://raw.githubusercontent.com/newren/git-filter-repo/docs/html/git-rebase.html)

closed 2022-04-20 by JOikarinen