polarofficial/polar-ble-sdk #247: [Noob?] Package Resolution Failed for polar-ble-sdk

Entilzar, 2022-03-05:
***Platform your question concerns:*** 
- [ X] iOS

***Device:***
- [X ] Device is not relevant for this question

**Description:**
Ok, I am a noob with MacOS and XCode, but I usually manage to follow a tutorial.
Using the XCode package manager, I successfully added RxSwift and Swift Protobuf dependencies to my project.
However, when I try to add polar-ble-sdk, using https://github.com/polarofficial/polar-ble-sdk.git (3.2.9 or 3.2.7), I got the following error : Failed to resolve dependencies
> because polar-ble-sdk >= 3.2.7 contains incompatible tools version and root depends on polar-ble-sdk 3.2.9..<4.0.0, version solving failed.

Do you have any idea on what's happening there and how to fix it ?

Thank you.

JOikarinen, 2022-03-06:
Hi @Entilzar,

first I wanted to double check that nothing is broken in Polar SDK when using as Swift Package Manager dependency. What I did:

1. Open my own test application in Xcode
2. In Xcode I followed the [Apple instructions](https://developer.apple.com/documentation/swift_packages/adding_package_dependencies_to_your_app). When url was asked I just passed the `https://github.com/polarofficial/polar-ble-sdk`

and in my experience swift package manager was able correctly resolve Polar BLE SDK. 

In your question you mention that `I successfully added RxSwift and Swift Protobuf dependencies to my project`. You might not need to add them separately to your project, as those are defined as needed dependencies by Polar SDK Swift package definition and are automatically defined as dependencies to your project when ever you take the Polar BLE SDK as decency via Swift Package Manager.

Entilzar, 2022-03-06:
Hi @JOikarinen ,
I don't understand what's happening.
I've just created two new projects : iOS/Game in Objective-C and iOS/Game in Swift.
Then tried to add the dependency with URL https://github.com/polarofficial/polar-ble-sdk.
And still have the same error. For both new projects, nothing else added before.
I am using XCode 12.4 on macOS Catalina v10.15.7

JOikarinen, 2022-03-07:
> I am using XCode 12.4 on macOS Catalina v10.15.7
okay, I suspect this might be the problem. My environment is the latest available XCode 13.2.1 and macOS Monterey 12.2.1. The intention is to keep Polar SDK backwards compatible with older XCode versions, but I started to think that chain may have got broken. Do you have possibility to update your enviroment?

JOikarinen, 2022-03-07:
@Entilzar, I think I know the root cause now. Currently the [Package.swift](https://github.com/polarofficial/polar-ble-sdk/blob/master/Package.swift) defines the swift tools version `// swift-tools-version:5.5`. 5.5 version is not available in your environment because XCode is in version 12.4. If you run the command line `swift package --version` you shall get something like "Swift Package Manager - Swift 5.3.x"

It shouldn't be a problem to downgrade the Polar BLE SDK from `// swift-tools-version:5.5` to `// swift-tools-version:5.3.2`. However, I didn't want to make that change straight to the `master` branch. So I created a branch [247-package-resolution-failed-for-polar-ble-sdk](https://github.com/polarofficial/polar-ble-sdk/tree/247-package-resolution-failed-for-polar-ble-sdk). You may test that branch if you define the `247-package-resolution-failed-for-polar-ble-sdk` as dependency rule for the Polar BLE SDK package. Something like this:

[image]

Entilzar, 2022-03-07:
@JOikarinen 
Ok. Thank you very much !
I will try to update my environment (a old Macbook which does not belong to me, so not sure I can do it) and if I cannot, I'll try the branch.
I'll let you know if I manage to get it working.

Entilzar, 2022-03-08:
Well...
Actually, I first tried the branch, and it was not working because my Swift Tools version was 5.3.0 instead of 5.3.2.
I uninstalled/reinstalled XCode but it didn't work.

So, I finally decided to update my MacOS. It's now 12.2.1 Monterey.
And I could add Polar BLE SDK master branch package to my project with no problem.

However (yes, sorry), while the project built fine, it crashed on my iOS device.
It's an iPad with iOS 12.4, and I got the following error
> dyld: Symbol not found: ___chkstk_darwin
I had also the following warning during the build :
> was built for newer iOS version (13.0) than being linked (12.4)

If I remove the Polar SDK from the "Framework, Libraries, Embedded Content", the warning and the crash disappear.
I've tried to update my iPad, but the maximum version it supports is 12.5.5.

So, I completely removed Polar SDK as dependency from my project, and added it back using the special branch.
I don't have the build warning, but have an import warning : 
> found 2 files which are unhandled; explicitly declare them as resources or exclude them from the target
>  .../polar-ble-sdk/sources/iOS/ios-communications/Sources/iOSCommunications/info/plist
>  .../polar-ble-sdk/sources/iOS/ios-communications/Sources/PolarBleSdk/info/plist

I did nothing about it, just build and run.
It built fine (no warning), but crashed at startup with the same error as before 
> dyld symbol not found : __chkstk_darwin

I've also noted that the RxSwift package version that was automatically added after importing the Polar SDK (branch version) was 6.5.0. Is it what it was supposed to be ?

That's all for today. 
I've run out of ideas for that matter, and cannot update more any of my devices.

JOikarinen, 2022-03-08:
@Entilzar thanks for the complete explanation. 

> I've tried to update my iPad, but the maximum version it supports is 12.5.5.
This might be an issue. The minimum target of the SDK is iOS13.

> I've also noted that the RxSwift package version that was automatically added after importing the Polar SDK (branch version) was 6.5.0. Is it what it was supposed to be ?
Yes, that is correct version, i.e. the latest [RxSwift](https://github.com/ReactiveX/RxSwift) version. 

> I've run out of ideas for that matter, and cannot update more any of my devices.
I am afraid that we are stuck. I don't have good ideas either what could we try next.

JOikarinen, 2022-03-08:
> So, I completely removed Polar SDK as dependency from my project, and added it back using the special branch. I don't have the build warning, but have an import warning :

This one is now fixed in master (commit 8be8c1fbc387c840b3a4754a9be52a978cf89ade), but I don't believe it has any effect on this case.

Entilzar, 2022-03-18:
Just for your information, I got a new device, with iOS version >= 13, so this is no longer an issue for me.

JOikarinen, 2022-03-18:
@Entilzar thanks for the update, I will close this issue for now. Let me know if more questions comes to your mind.

closed 2022-03-18 by JOikarinen