polarofficial/polar-ble-sdk #249: Getting PPI data using PDM onto my esp32

lalit-29, 2022-03-10:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Hi, 
I am trying to stream the PPI data from my polar verity sense using an ESP32 microntroller. I had success with getting the HR data using the HR service. But, I am having trouble getting the PPI data. Based on reading GATT documentation and seeing some of the code posted by other people, I understand that I need to subscribe to the PMD service, I am able to do that and send commands to the PMD control point characteristic but I get no response and no data coming out from the PMD data characteristic.

The requests I sent are `[0x01 0x03]` and `[0x02 0x03]` to Pmd control point and register a notify callback on the PMD data service. 

I used a modified version of the code from the following sources:
https://stackoverflow.com/questions/68092080/i-tried-to-do-ble-with-esp32-but-the-notifycallback-doesnt-work

```
//===== header file & define & global variable ===== 
#include"BLEDevice.h"
char val1 = 0;
char incomingByte;
boolean doConnect = false;
volatile boolean isConnected = false;
boolean doScan = false;

BLEUUID pmd_serviceUUID ("FB005C80-02E7-F387-1CAD-8ACD2D8DF0C8");
BLEUUID pmd_dataUUID ("FB005C82-02E7-F387-1CAD-8ACD2D8DF0C8");
BLEUUID pmd_ctrlUUID ("FB005C81-02E7-F387-1CAD-8ACD2D8DF0C8");
BLEAdvertisedDevice* myDevice;
BLEClient* pClient;
//---------------------------------------------------------------
static BLERemoteCharacteristic* pDataCharacteristic;
static BLERemoteCharacteristic* pControlCharacteristic;

String SensorName = "Polar Sense 91102B26"; // SDから読み取る
//===========================================

//===== class & function ====================
class MyClientCallback: public BLEClientCallbacks{
  void onConnect(BLEClient* pclient){ }
  void onDisconnect(BLEClient* pclient){
    isConnected = false;
    Serial.println("onDisconnetct");  
  }
};

void notifyCallback(BLERemoteCharacteristic* pBLERemoteCharacteristic, uint8_t* pData, size_t length, bool isNotify){
  Serial.print("Notify callback for characteristic ");
  Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
  Serial.print(" of data length ");
  Serial.println(length);
  Serial.print("data: ");
  for (int i = 0; i <= length - 1; i++) {
    Serial.print(String(*(pData + i), HEX));
    Serial.print(" ");
  }
  Serial.println();
}

bool connectToServer(){
  Serial.print("connection to : ");
  Serial.println(myDevice->getAddress().toString().c_str());
  pClient = BLEDevice::createClient();
  Serial.println(" - Created client");
  pClient->setClientCallbacks(new MyClientCallback() );
  pClient->connect(myDevice);
  Serial.println(" - Created to server");

  BLERemoteService* pRemoteService = pClient->getService(pmd_serviceUUID);
  if (pRemoteService == nullptr) {
     Serial.println("Failed to find our service UUID: ");
     Serial.println(pmd_serviceUUID.toString().c_str());
     pClient->disconnect();
     return false;
  }
  Serial.print(" - Found service ( ");
  Serial.print(pmd_serviceUUID.toString().c_str());
  Serial.println(" )");

  
  pControlCharacteristic = pRemoteService->getCharacteristic(pmd_ctrlUUID);
  if( pControlCharacteristic == nullptr ){
    Serial.print("Failed to find out characteristic UUID : ");
    Serial.println(pmd_ctrlUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }

  Serial.println("characteristics");
  std::map<uint16_t, BLERemoteCharacteristic*>* mapCharacteristics = pRemoteService->getCharacteristicsByHandle();
  for (std::map<uint16_t, BLERemoteCharacteristic*>::iterator i = mapCharacteristics->begin(); i != mapCharacteristics->end(); ++i) {
    Serial.print(" - characteristic UUID : ");
    Serial.print(i->second->getUUID().toString().c_str());
    Serial.print(" Broadcast:");
    Serial.print(i->second->canBroadcast()?'O':'X');
    Serial.print(" Read:");
    Serial.print(i->second->canRead()?'O':'X');
    Serial.print(" WriteNoResponse:");
    Serial.print(i->second->canWriteNoResponse()?'O':'X');
    Serial.print(" Write:");
    Serial.print(i->second->canWrite()?'O':'X');
    Serial.print(" Notify:");
    Serial.print(i->second->canNotify()?'O':'X');
    Serial.print(" Indicate:");
    Serial.print(i->second->canIndicate()?'O':'X');
    Serial.println();
  }

  
  pDataCharacteristic = pRemoteService->getCharacteristic(pmd_dataUUID);
    if( pDataCharacteristic == nullptr ){
    Serial.print("Failed to find out characteristic UUID : ");
    Serial.println(pmd_dataUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  Serial.print(" - Add Notify ( ");
  Serial.print(pmd_dataUUID.toString().c_str());
  Serial.println(" )");
  pDataCharacteristic->registerForNotify(notifyCallback);
  pControlCharacteristic->registerForNotify(notifyCallback);
  if(pDataCharacteristic->canNotify()){
    //std::string value = pControlCharacteristic->readValue();
    
    const uint8_t data[2] = {0x02,0x03};
    pControlCharacteristic->writeValue((uint8_t*)data,2,false);
    Serial.println(" - Set value");
    
    Serial.println(" - Can Notify");
    
  }
  
  isConnected = true;
  return true;
}
```
I have also tried using the NRF connect app to send data but I get no response. My guess is my data format is wrong, any help would be appreciated

Thank you

JOikarinen, 2022-03-11:
Hi @lalit-29 

> The requests I sent are `[0x01 0x03]` and `[0x02 0x03]` to Pmd control point and register a notify callback on the PMD data service.

The request `[0x01 0x03]` looks good to me: 
    `0x01`: Get Measurement settings
    `0x03`: PPI data. 

For the request you shall get the response like this `[0xF0 0x01 0x03 0x00 0x00]`:
  `0xF0`: Response code, 
  `0x01`: OP Code (Get Measurement settings)
  `0x03`: Measurement Type (PPI data)
  `0x00`: Error Code (success) 
  `0x00`: More (last packet)

The request `[0x02 0x03]` looks good to me: 
`0x02`: Request Measurement start, 
`0x03`: PPI data. 
For the request you shall get the response like this `[0xF0 0x02 0x03 0x00 0x00]`:
 `0xF0`: Response code
`0x02`: OP Code (Request Measurement start)
`0x03`: Measurement Type (PPI data)
`0x00`: Error Code (success)
`0x00`: More (last packet)

 After the Measurement start request and successful response you shall see PPI data to be started notified on `pmd_dataUUID `
 
 Based on the explanation you don't see the responses?

lalit-29, 2022-03-11:
The only thing I get back from the PMD control service regardless of what I send is: F-6E-2-0-0-0-0-0-0-0-0-0-0-0-0-0-0
[image]

I've tried sending `[0x01 0x03]` and `[0x02 0x03]`  but still the response is the same. 
This is the snippet where I send the data:

```
const uint8_t data[2] = {0x02,0x03};
pControlCharacteristic->writeValue((uint8_t*)data,2,false);
```
Is there specific format I need to send them in?

lalit-29, 2022-03-14:
HI @JOikarinen 

Any ideas on what might be going on? I could really use some help. 

I even tried using the NRF connect app. i am able to see the service, but, I get no response back when I send the requests.

JOikarinen, 2022-03-16:
Hi @lalit-29, 

here are the steps you shall get the nRF Connect working:
1. Search and connect with Verity Sense using nRF Connect
2. Set MTU to 232 (Found under option Request MTU)
3. Allow Indications for characteristics `FB005C81-02E7-F387-1CAD-8ACD2D8DF0C8`
[image]
4. Write `[0x01 0x03]`
[image]

lalit-29, 2022-03-17:
Hi @JOikarinen 

Setting the MTU value to 232 did the trick!  I am able to get notifications and stream the data in both the nrf connect app and in my ESP32.
`pClient->setMTU(232);`
Thank you, I really appreciate the help!

JOikarinen, 2022-03-18:
@lalit-29 I will close this issue for now, don't hesitate to open again for more questions.

closed 2022-03-18 by JOikarinen