polarofficial/polar-ble-sdk #250: com.polar.sdk.api.errors.PolarDeviceDisconnected

KennethEvans, 2022-03-11:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

I am getting PolarDeviceDisconnected errors in my ECG app.  This is something new. It recovers and continues getting ECG values, however.

Log attached.

[KE.Net-ECG-Exception2022-03-11.txt](https://github.com/polarofficial/polar-ble-sdk/files/8234740/KE.Net-ECG-Exception2022-03-11.txt)

KennethEvans, 2022-03-20:
It hasn't happened again. May be a sometime thing.

jimmyzumthurm, 2023-09-08:
Hi @KennethEvans 
Have you noticed this behavior since your original report ?

dmoonfire, 2023-09-08:
Sadly, @KennethEvans  passed on July 30 due to heart complications. I'm monitoring his account (you can verify on his profile) if you have questions, but I'm not able to verify anything related to his Polar device so this issue is safe to close.

jimmyzumthurm, 2023-09-11:
Hi @dmoonfire , thank you for letting us know. We're sorry to hear about Kenneth's passing. If there are any questions or concerns in the future, please feel free to contact us.

closed 2023-09-11 by jimmyzumthurm