polarofficial/polar-ble-sdk #251: H10 ECG "lead" type

cdswan, 2022-03-13:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

I've noticed that the ECG from the H10 doesn't really resemble an actual ECG, although it still has very workable characteristics. My app is able to read and display the ECG and is able to reliably detect peaks (using a slightly modified Pan-Tompkins algorithm). But, (question 1) which peak in the following picture is the actual R point of the QRS complex?

[image]

My second question is what ECG lead does the H10 most closely resemble?

My third question is what is the internal sample rate of the H10? I know the external sample rate is 130Hz which relates to a resolution of around 7ms, but the H10 clearly does better internally because the RRs available through through the heart rate service do way better than this.

Thanks in advance!

JOikarinen, 2022-03-16:
Hi @cdswan,

good questions. However I have bit of hard time give you good answers. For your question 1 and 2, I must answer the same answer given in earlier issue that ECG signal is given as raw signal but at the moment we don't have additional support to help with ECG data analysis. 

For the question 3 I cannot give more than this https://www.polar.com/sites/default/files/static/science/white-papers/polar-h10-heart-rate-sensor-white-paper.pdf

cdswan, 2022-03-16:
Thanks Jukka, I might need to talk to a cardiologist!

closed 2022-03-16 by cdswan

ikhwanperwira, 2024-11-19:
@cdswan So, what lead type is corresponded?