polarofficial/polar-ble-sdk #252: ECG with Verity Sense ?

Entilzar, 2022-03-16:
***Platform your question concerns:*** 
- [ X] iOS

***Device:***
- [X ] Polar Verity Sense

**Description:**
I am trying to get ECG from Verity Sense device, but according to the SDK results, ECG is not supported by Polar Verity Sense.
Can you confirm ?

keturn, 2022-03-16:
The Verity Sense only has an optical sensor (and some movement stuff), it has no ECG.

JOikarinen, 2022-03-16:
@Entilzar as @keturn already confirmed ECG data is impossible from the devices which has only optical sensors. Polar H10 has ECG feature. In Polar Verity Sense there is PPG and PPI. PPG is the raw data from the optical sensor.  PPI (Pulse-to-pulse interval) is the data which is calculated in Polar Verity Sense sensor using the PPG.

closed 2022-03-16 by JOikarinen