polarofficial/polar-ble-sdk #253: SDK Framework / Static Library

Entilzar, 2022-03-18:
***Platform your question concerns:*** 
- [ X] iOS

***Device:***
- [X ] Device is not relevant for this question

**Description:**
Hello again.
I need to embed the polar-ble-sdk into an iOS app built by a game engine.
To do so, I need the SDK to be compiled as a framework, then zipped.

I think I've managed to get the framework built, by building the "sources/iOS/ios-communications" project, but I got multiple ones :
- iOSCommunications
- Pods_iOSCommunications
- Pods_PolarBleSdk
- PolarBleSdk
- RxSwift-iOS/RxSwift
- SwiftProtobuf-iOS/SwiftProtobuf

I suppose that RxSwift and SwiftProtobuf are dependencies frameworks that are not embedded into yours to avoid conflicts.
=> Is that correct ?

Then, I don't know exactly what the Pods_ frameworks are, and what to do with them.
Can you explain that part ?

Alternatively, it's possible to use a Static Library (.a file) to do the trick.
Is there a way to build the Sdk as a Static Library ?

Thank you.

JOikarinen, 2022-03-18:
So, in your project it is not possible to take Polar BLE SDK via traditional dependencies Swift Package manager or CocoaPods?

About the XCFramework:
- there is a problem related XCFramework distribution which I haven't been able to resolve. Problem is related to the fact that PolarBleSdk depends on the [RxSwift](https://github.com/ReactiveX/RxSwift) and [SwiftProtobuf](https://github.com/apple/swift-protobuf). The problem appears when you try to use the PolarBleSdk.xcframework in your application. Xcframework dynamically links any dependent packages (i.e.  RxSwift and SwiftProtobuf), so those are not embedded in to the PolarBleSdk.xcframework and as a consequence the client app won't build.   I am quite sure this problem can be resolved somehow I just don't know how. Anyways if you want to play around:

    - you can build the XCFramework, the  script in https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/iOS/ios-communications/scripts/build_sdk_xcframework.sh which builds the XCFramework of PolarBleSDK
    ```bash
    scripts $ ./build_sdk_xcframework.sh  
    ```
    - the build output can be found in project [root]( https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/iOS/ios-communications/) in folder sdkXcBuild

Entilzar, 2022-03-18:
> So, in your project it is not possible to take Polar BLE SDK via traditional dependencies Swift Package manager or CocoaPods?
No, it's not. The game engine has its own Build tool that is not well documented. And the two working ways to work with it are either the Static Library or the embedded framework.

> About the XCFramework:
>
>     * there is a problem related XCFramework distribution which I haven't been able to resolve. Problem is related to the fact that PolarBleSdk depends on the [...]

I had quite the same problems with my tests. I tried to build a Static Library of a project that had PolarBleSdk as a dependency. It compiles, but when I try to use it in the game engine, either references are not found or there are a tons of Duplicate symbols in RxSwift when building. 

That's why I am trying another approach and see if I can get the framework from the sources.

Entilzar, 2022-03-19:
@JOikarinen I used the script to generate the XCFramework.
I managed to compile and link against it, but as expected, the app crashes at startup.

First, to make it compile, I had to add Core Bluetooth (import "CoreBluetooth/CoreBluetooth.h") to the Header ("PolarBleSdk-Swift.h").
Now, the problem is that the app cannot find the PolarBleSdk Dynamic Library. It is expecting a dylib file in order to load the PolarBleSdk, which is not included in the XCFramework.

Is there a way to modify the script to generate that file ?
Or may be add a Dynamic Library Target to the source workspace ?

Entilzar, 2022-03-23:
@JOikarinen 
I think I managed to get something on my side : I have a framework with a working "PolarBleSdk-Swift.h".

However, I am under the impression (but I may be wrong) that I cannot call any class from the PolarSDK using Objective-C, as the Swift classes have not been declared with @objc.

Is it possible for you to provide an equivalent to PolarBleSdkManager.Swift in Objective-C ?

JOikarinen, 2022-03-28:
Hi @Entilzar, sorry for a delayed response. 

I am afraid that I cannot fulfil your wish. The reason being that Polar BLE SDK API's are using [RxSwift](https://github.com/ReactiveX/RxSwift), which in turn is not Objective-C compatible.

Entilzar, 2022-03-30:
Hi @JOikarinen 

Fortunately, I finally managed to get something working.
I made a Static Library of a Objective-C project embedding the Polar SDK and the PolarSdkManager.swift file from the sample app, which I modified to be able to call from Objective-C.

So now, I just have 2 more questions, in order to get and send ECG data to another API :
- How many leads does the Polar H10 uses for the ECG stream ? (i suppose 2, since there are 2 leads on the device)
- What is the Resolution parameter ? And does it relate to the Resolution parameter of another API => "Example: resolution=1000 ; Resolution in microvolts." ?

JOikarinen, 2022-03-31:
Hi @Entilzar,

awesome news! 👍

> * How many leads does the Polar H10 uses for the ECG stream ? (i suppose 2, since there are 2 leads on the device)
- Yes, it is two. 

> What is the Resolution parameter ? And does it relate to the Resolution parameter of another API => "Example: resolution=1000 ; Resolution in microvolts." ?
- When [requesting settings](https://github.com/polarofficial/polar-ble-sdk/blob/8be8c1fbc387c840b3a4754a9be52a978cf89ade/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApi.swift#L260) from the H10 the response is always the Sampling Rate 130Hz and Resolution 14 wrapped in [PolarSensorSettings](https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/model/PolarSensorSetting.swift)
- When requesting the ECG stream from H10 using [startEcgStreaming](https://github.com/polarofficial/polar-ble-sdk/blob/8be8c1fbc387c840b3a4754a9be52a978cf89ade/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApi.swift#L282), you shall use the same settings received by [requestStreamSettings](https://github.com/polarofficial/polar-ble-sdk/blob/8be8c1fbc387c840b3a4754a9be52a978cf89ade/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApi.swift#L260)
- In practice you can hard code the settings as they are never changing for H10 and it is not possible change settings by the SDK.  By this I mean, if you wish, you don't need to request the settings from H10 before starting the stream. You may just hard code settings in startEcgStreaming request. 
- sampling rate is the sampling rate H10 uses internally when taking samples from ECG 
- resolution is the resolution how many bits H10 uses per sample. This is slightly irrelevant since the sampling range is not available, however it is still given in settings request.

Entilzar, 2022-03-31:
Thank you for these info !

JOikarinen, 2022-04-01:
@Entilzar , I close this issue for now. Please open again if more issues rises.

closed 2022-04-01 by JOikarinen