polarofficial/polar-ble-sdk #254: absolute newbie needs help to build (1st) app

marexel, 2022-03-19:
***Platform your question concerns:*** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [X] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Hi there,

I have never build an android-app before and was hoping not to have to do so because of my age. But:
to use the H10 without having to use the official "flow-app" I think I have to. I was looking for a way to read the data the H10 provides without sending it to an unknown server. I want to keep it on my own.
So far, I tried to build the app with the help of "Getting started". But I failed with

- On your application you must request for the [permissions](https://developer.android.com/guide/topics/permissions)...
- Load the default api implementation and add callback.

Can anyone help my finding more info on how to do that? Do I have to look for a "polar-ble-sdk" special way or is it a common way on how to build an app?
Or is there a better place to ask these questions (no Issue listed seems to have a basic question like mine)?
Thanks for your patience, kr,

marexel

KennethEvans, 2022-03-20:
I have an app that gets most of the useful data from the H10, including HR, RR, and ECGs. You could look at that. 

https://github.com/KennethEvans/KE.Net-ECG

If you've never built an app, be aware there is an extensive learning curve. Maybe you could just use this one, or I've written others that don't use the SDK. I sympathize with you on Beat and Flow. They are part of the reason I've written my own.

There is also the demo app in the SDK, but it is in Kotlin. It does about everything the SDK can do.

jimmyzumthurm, 2023-09-08:
Hi @marexel , I hope you were able to build your own app using Polar SDK. If you still need help with that, you can open a new issue as I'm closing this one as it has been stale for quite a long time.

closed 2023-09-08 by jimmyzumthurm