polarofficial/polar-ble-sdk #257: Different measurements when both iOS and Android is connected and in background.

JasurSalimov, 2022-03-23:
***Platform your question concerns:*** 
- [✅ ] Android  - 
- [ ✅] iOS

***Device:***
- [✅] Polar H10

**Description:**
Our company develops OU7 Training App, you can see it in the AppStore as well. The problem is when we start training on both Android device and iOS device with Polar H10 connected to both. In foreground everything works fine, however when we go to background on both devices connected, the iOS device does not receive the updates as frequently as it should be. Every 3-5 seconds one broadcast from polar. But android in the other hand works fine. So the question is, can you please suggest the way to resolve the problem and did it work in the same way when your team connected both devices? Does this problem related to iOS hardware and software capabilities and restrictions ?

JOikarinen, 2022-03-28:
Hi @JasurSalimov, 

I just played around with OU7 Training App, very nice app! About the issue, can you confirm I got the steps correctly:

STEPS:

1. You have enabled the [dual BLE connection option](https://support.polar.com/en/support/can_i_pair_h10_with_several_devices) in H10. 
2.   Android app creates connection to H10 and starts receiving heart rate updates via [hrNotificationReceived](https://github.com/polarofficial/polar-ble-sdk/blob/8be8c1fbc387c840b3a4754a9be52a978cf89ade/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApiCallback.java#L115)
3. iOS app creates connection to the same H10 and starts receiving heart rate updates via [PolarBleApiDeviceHrObserver](https://github.com/polarofficial/polar-ble-sdk/blob/8be8c1fbc387c840b3a4754a9be52a978cf89ade/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApiObservers.swift#L53)
4. move iOS app to background

OBSERVED BEHAVIOR:
- After step 4 the [PolarBleApiDeviceHrObserver](https://github.com/polarofficial/polar-ble-sdk/blob/8be8c1fbc387c840b3a4754a9be52a978cf89ade/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApiObservers.swift#L53) is called only every 3-5 seconds on iOS app

QUESTIONS:
- when iOS app is brought foreground again, is [PolarBleApiDeviceHrObserver](https://github.com/polarofficial/polar-ble-sdk/blob/8be8c1fbc387c840b3a4754a9be52a978cf89ade/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApiObservers.swift#L53) called roughly once a second again?

JOikarinen, 2022-04-26:
Hi @JasurSalimov,

today I came back on this issue. In your original description you mentioned "broadcast", so it make me think that maybe the problem you see is not related to BLE connection but BLE broadcasting. 

In Polar BLE SDK you have option to listen the broadcast data from Polar sensors, in this scenario you are not actually creating a BLE connection to the device. The Polar BLE SDK just scans for the advertisement packets. In this scenario you use the startListenForPolarHrBroadcasts ([iOS](https://github.com/polarofficial/polar-ble-sdk/blob/610d5e1a4cb38d05d873794055e60a44616bcbe5/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApi.swift#L247) / [Android](https://github.com/polarofficial/polar-ble-sdk/blob/610d5e1a4cb38d05d873794055e60a44616bcbe5/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L380)) API.

If you are actually listening the broadcast then I may know the reason for you issue. The reason is how the iOS operating system works when screen is turned off or the app is moved away from foreground. In this case the iOS operating system starts filter the data it gives. This filtering is seen as slower updates on heart rate data.

JasurSalimov, 2022-04-29:
Hi @JOikarinen, 
yes the problem was exactly because of the broadcast. Actually changing the API helped. Thanks for response. All the best.

closed 2022-04-29 by JasurSalimov