polarofficial/polar-ble-sdk #260: Getting data via GATT service, several questions

Simchev38, 2022-03-25:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ X] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [X ] Polar Verity Sense
- [X ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Hello 

I am trying to retrieve data from H10 and verity sense in a .NET Core software.

I'm using the "Polar measurement data specification" document here : https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf

I can receive  "live data" successfully

I have several questions : 

- I could not find any information on how , using the GATT service, I could  : 
              - Set the time of the sensors
              - Download "offline" data from memory
              - clear the memory 
             
-I also discovered the "SDK mode",  is it possible to enable it via  the GATT service ? if yes how ?

-Finally,  I understood that only HR and RR data can be recorded and then downloaded from memory ?  Did I understand right or is it possible to record and then download ACC, GYRO, MAGN data ? 

Thanks in advance for your answers

JOikarinen, 2022-03-26:
Hi @Simchev38, good questions. Let me try answer. 

First I must say the traditional disclaimer: Unfortunately, we are only able to support Android and iOS. The SDK includes source code and you can feel free to utilize / port the code in other platforms too but that is something we are not able to provide support for.

> * I could not find any information on how , using the GATT service, I could  :
>   - Set the time of the sensors
>   - Download "offline" data from memory
>   - clear the memory
That is because only the Polar measurement data (PMD) service is [documented](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf). And the features you mention are not supported by PMD service. 

> -I also discovered the "SDK mode", is it possible to enable it via the GATT service ? if yes how ?
Yes, that is possible. You may enable/disable SDK mode by similar commands you start the stream and stop the stream. To enable you sent [0x02, 0x09] and to disable may shall sent [0x02, 0x09] command on "FB005C81-02E7-F387-1CAD-8ACD2D8DF0C8" characteristics. 

> -Finally, I understood that only HR and RR data can be recorded and then downloaded from memory ? Did I understand right or is it possible to record and then download ACC, GYRO, MAGN data ?

Only the HR and RR data can be recorded and the downloaded. ACC, GYRO and MAGN recording is not supported by the  devices.

Simchev38, 2022-03-28:
Hello 

Thanks for your answers.

JOikarinen, 2022-03-28:
@Simchev38 I am closing this question for now, but don't hesitate open more questions.

closed 2022-03-28 by JOikarinen