polarofficial/polar-ble-sdk #261: iOS crash with GPBAutocreatedArray

srongch, 2022-03-29:
***Platform your question concerns:*** 
- [ ] Android 
- [ x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Hello, I sometimes get this crash when trying to discover  and connect to Polar device. Can anyone help?

Crashed: CBDeviceListenerImplQueue0
0  libswiftFoundation.dylib       0x45c28 Data._Representation.subscript.getter + 1044
1  libswiftFoundation.dylib       0x11eca4 Data.subdata(in:) + 120
2  PolarBleSdk                    0x60940 -[GPBAutocreatedArray enumerateObjectsWithOptions:usingBlock:] + 66872
3  PolarBleSdk                    0x60ad0 -[GPBAutocreatedArray enumerateObjectsWithOptions:usingBlock:] + 67272
4  PolarBleSdk                    0x53bf4 -[GPBAutocreatedArray enumerateObjectsWithOptions:usingBlock:] + 14316
5  PolarBleSdk                    0x95408 block_destroy_helper + 2004
6  PolarBleSdk                    0x93d90 block_destroy_helper.84 + 16464
7  CoreBluetooth                  0x293dc -[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] + 188
8  CoreBluetooth                  0x29580 -[CBPeripheral handleCharacteristicEvent:characteristicSelector:delegateSelector:delegateFlag:] + 124
9  CoreBluetooth                  0x2584c -[CBPeripheral handleMsg:args:] + 348
10 CoreBluetooth                  0x12e20 -[CBCentralManager handleMsg:args:] + 188
11 CoreBluetooth                  0x4b7b0 -[CBManager xpcConnectionDidReceiveMsg:args:] + 208
12 CoreBluetooth                  0x3ca7c __30-[CBXpcConnection _handleMsg:]_block_invoke + 68
13 libdispatch.dylib              0x2924 _dispatch_call_block_and_release + 32
14 libdispatch.dylib              0x4670 _dispatch_client_callout + 20
15 libdispatch.dylib              0xbdf4 _dispatch_lane_serial_drain + 672
16 libdispatch.dylib              0xc99c _dispatch_lane_invoke + 444
17 libdispatch.dylib              0xbccc _dispatch_lane_serial_drain + 376
18 libdispatch.dylib              0xc968 _dispatch_lane_invoke + 392
19 libdispatch.dylib              0x171b8 _dispatch_workloop_worker_thread + 656
20 libsystem_pthread.dylib        0x10f4 _pthread_wqthread + 288
21 libsystem_pthread.dylib        0xe94 start_wqthread + 8

JOikarinen, 2022-03-30:
Hi @srongch, interesting finding. Is it possible to add logging on for the Polar BLE SDK library and provide the logs of the situation? Logging can be enabled by implementing the [PolarBleApiLogger](https://github.com/polarofficial/polar-ble-sdk/blob/8be8c1fbc387c840b3a4754a9be52a978cf89ade/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/PolarBleApiObservers.swift#L94).

srongch, 2022-03-30:
Hello @JOikarinen. Thank you for your response. Can you instruct me further what kind of data should I log. 
This issue happened sometimes in production app. I could not able to reproduce it. It will take a while to receive that log data.

JOikarinen, 2022-03-31:
Okay, so you got the problem report via crash analytics tools from production and not able reproduce on development environment? The SDK logs maybe better to use only in development but preferably not in production. I have done some investigation on the code side, but yet I haven't get understanding of possible root cause.

srongch, 2022-04-08:
I use .searchForDevice() function to search for devices and automatically connect the first one on the list. I suspect that issue happens when multiple devices and multiple apps are near each other, and they all try to connect. I am requesting another Polar device to test this theory.

closed 2023-10-12 by samulimaa